emsesp/EMS-ESP32 #500: Unknown: unknown (DeviceID:0x53, ProductID:248, Version:05.02)

scat703, 2022-05-14:
**Is your feature request related to a problem? Please describe.**
*A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]*
Buderus Hybrid Manager HM200 not known so far by ems-esp

**Describe the solution you'd like**
*A clear and concise description of what you want to happen.*
In my Buderus Logasys SL213 environment I see the boiler, RC310 and KM200. DeviceID 0x53 is unknown to ems-esp so far. From my perspective this should be the Buderus Hybrid Manager HM200, as this is the only additional device connected to the ems bus. The heat pump is connected to HM200 by canbus.

**Describe alternatives you've considered**
*A clear and concise description of any alternative solutions or features you've considered.*
Checked KM200 for HM200 values, but KM200 does not discover/recognize HM200 so far.

**Additional context**
*Add any other context or screenshots about the feature request here.*

MichaelDvP, 2022-05-15:
It is a first step to add the HM200, i think best to heatpump-class, [i've done it here](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest). But you also want to have the measurements and settings from this module.
To determine the measurements, go to terminal and set `watch 53` to see all telegrams from this module. Write down names and values of all measured values you can see on the thermostat-display and we can search these values in the broadcasted telegrams. (`heatpump(0x53) -> All (0x00), unkown(0x..), data:...`), if a value changes while logging, not both values, we can better identify it.
For the settings also set the `watch 53` and change each HM200 setting on the thermostat, not only show up, really change the setting, confirm,, and then change it back. Note name and setting value. This should result in short broadcast telegrams with single value and offset.  (`heatpump(0x53) ->All(0x00) unknown(0x..), data: xx (offset: xy)`

MichaelDvP, 2022-05-15:
Correction: for the settings we expect a message from thermostat to H-module `Thermostat(0x10) -> heatpump(0x53) unknown(0x..), data: xx (offset: xy)`. If these messages are missing, it's also possible that the values are stored in thermostat, than we have a broadcast from thermostat and need to `watch 10` to fetch them.

scat703, 2022-05-15:
As the upcoming two weeks will be quite busy, I plan to gather all the information by the end of May or early June. I will upload them asap!

Joern-W, 2022-09-27:
Hi, 
I'm new to the EMS-ESP32. I recently got a HM200.2 device and I'm highly interested in filling the EMS-ESP with measurements.
Attached there is a text file with the output of the watch53 command and a spreadsheet with the values I read from the display. (also including a German translation for the value names) I read these values 4 times: once while off and 3 times while heating.

For now I did not change any settings. Is this data helpful? If so, I can change some settings next time.

[watch53.txt](https://github.com/emsesp/EMS-ESP32/files/9659804/watch53.txt)
[Watch53Values.ods](https://github.com/emsesp/EMS-ESP32/files/9659805/Watch53Values.ods)

proddy, 2022-09-28:
This is great, thanks. We can certainly add these. The list shows the readings of the entities, do you know which ones should be 'writeable'?

Joern-W, 2022-09-28:
I have no idea. However I just went through all the hybrid settings and all the hybrid function tests.
The log file includes a description of what I did.
[Wacht53_Settings.txt](https://github.com/emsesp/EMS-ESP32/files/9667935/Wacht53_Settings.txt)

MichaelDvP, 2022-09-29:
This is exactly the right way. I've reduced the file to the relevant data:
[Wacht53_Settings.txt](https://github.com/emsesp/EMS-ESP32/files/9672373/Wacht53_Settings.txt)
The settings are in telegrans 0x998 and 0x999, there is no broadcast, so please do a `read 53 998` and `read 53 999` to see the whole telegram.
In your first log we see the broadcasted (measurement) telegrams. 0x99A - 0x9A8 and a not changing 0x4AA. 
We see a lot of changing 2byte values, mostly temperatures if starting with 00, or 01, or 02. But i have not matched to the ods file yet.

Joern-W, 2022-10-09:
Hi, I had some busy days, but here finally the results of my investigation:
the values of `read 53 998` are (including the german translation):
`
Heatpump(0x53) -> Me(0x0B), ?(0x0998), data: 00 00 0B 00 00 1F 01 00 01 01 16 06 00 04 02 FF 00 01 7C 01
                                     offset: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19

offset 00: "Control strategy" (German: "Regelungsstrategie")
            00 -> "cost-optimized" ("Kostenoptimiert")
            01 -> "Environment-optimised" ("Umweltoptimiert")
            02 -> "Outside temp. alternative" ("Außentemp. alternativ")
            03 -> "Outside temp. parallel" ("Außentemp. parallel")
            04 -> "Heat pump preferred" ("Wärmepumpe bevorzugt")
            05 -> "Boiler only" ("Nur Kessel")

offset 01: "Hybrid" - "DHW" (German: "Hybrid" - "Warmwasser")
            00 -> "Eco" ("Eco")
            01 -> "Comfort" ("Comfort")

offset 02: "Energy Prices" - "Gas Costs" in cent (German: "Energiepreise" - "Gaskosten")

offset 03: unknown - value 0
offset 04: unknown - value 0

offset 05: "Energy Prices" - "Electricity Costs" in cent (German: "Energiepreise" - "Stromkosten")

offset 06: "Energy Prices" - "Feed-in tariff PV" in cent (German: "Energiepreise" - "Einspeisetarif PV")

offset 07: unknown - value 0
offset 08: unknown - value 1

offset 09:  "low noise operation" - "operating mode" (German: "Geräuscharmer Betrieb" - "Betriebsart")
            00 -> "Off" ("Aus")
            01 -> "Reduced Output" ("Reduzierte Leistung")
            02 -> "Switch Off HP" ("WP ausschalten")
            03 -> "perm. reduced" ("perm. reduziert")
            
offset 10: "low noise operation" - "Start Time" (German: "Geräuscharmer Betrieb" - "Startzeit")
            allowable values: 0 to 23
            
offset 11: "low noise operation" - "End Time" (German: "Geräuscharmer Betrieb" - "Endzeit")
            allowable values: 0 to 23
            
offset 12: unknown - value 0
offset 13: unknown - value 4

offset 14: "dual-fuel switch-over point" (German: "Bivalenztemperatur")

offset 15: unknown - value -1 (or 255)
offset 16: unknown - value 0
offset 17: unknown - value 1
offset 18: unknown - value 124
offset 19: unknown - value 1
`
More details in the attached text file
[Watch53_Settings_0x0998.txt](https://github.com/emsesp/EMS-ESP32/files/9742609/Watch53_Settings_0x0998.txt)

The values in `read 53 999` is basically function test stuff.

About the measurement telegrams: What format are the temperatures stored? (So I can match them to the ods file.)

proddy, 2022-11-13:
@MichaelDvP shall we try and add this to v3.5.0 or wait?

MichaelDvP, 2022-11-13:
The values as far as documented are in the dev since a month. But have never be confirmed.

Joern-W, 2022-11-13:
That's great news. I'm using the release firmware but I will upgrade to beta tomorrow and check the values.
Thanks Jörn

Joern-W, 2022-11-13:
Hi, I just checked the values.
First: only the settable values are shown, not the temperature readings etc.
The hybrid control strategy is shown as "co2 optimized" while my hybrid manager is actually set to "cost optimized". All other values appear to be correct.

MichaelDvP, 2022-11-14:
> The hybrid control strategy is shown as "co2 optimized" while my hybrid manager is actually set to "cost optimized". 

This ia a typo, the hybrid heatpump at boiler-id 8  have nearly the same setting, but in different order. I'll fix it.
> First: only the settable values are shown, not the temperature readings etc.

Yes, I've implentented the settings [here](https://github.com/emsesp/EMS-ESP32/issues/500#issuecomment-1262048980) and there was no feedback, so i did not work on other settings. 
For the settings we need a more exact match for telegram and current value change. I can not identify them in the log.
Maybe the changing percentage values can be identified, but a setting off, staying off, is one of the thousand zeros. 

Temperatures changinga lot, i'm not 100% sure, but i think the mentioned temperature are in telegram 0x099B:
`Heatpump(0x53) -> All(0x00), ?(0x099B), data: 80 00 80 00 01 3C 01 38 80 00 80 00 80 00 01 37 00 00 00 00 64`
flowtemp 4, retTemp 6, sysRetTemp 14
and the others in 0x09A0:
`Heatpump(0x53) -> All(0x00), ?(0x09A0), data: 02 23 01 3E 01 39 00 5D 01 DE 01 38 00 40 00 5E 00 58 00 3F 01 34 00 02`
Tc3: 2, Tr1:, 8, Tr3: 10, Tr4: 12, Tr5: 14, Tr6:, 0, Tl2: 6, Ta4: 16, Jr0: 18, Jr1: 20
There is another temperature at offset 4.

I'll implement this, but has to be checked. Please try: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

MichaelDvP, 2022-11-14:
"dont match" does not help. You have to say what temperatures these are. But i have not found other values matching the temperatures you documented in the ods. 0x099B offset 18 changes in the log from 0 to 1 and after 10 minutes back to 0, offset 19 goes in the same time from 0->28->100->28->0 seems a percentage. Could be heating pump on/off and modulation, but does not fit to your ods.
Please log again and note **exact time and value** when a specific temperature/value changes.

Joern-W, 2023-02-17:
Sorry for the Delay, but finally i could figure out some more values:

heatpump(0x53) -B-> All(0x00), HPFlowTemp(0x099B), data: 80 00 80 00 01 4E 01 39 80 00 80 00 80 00 01 31 00 00 01 5D 0F
01 4E (offset 4) = TH3 current flow temperature (Vorlauftemperatur)
01 39 (offset 6) = TH2 return temperature (Rücklauftemperatur)
01 31 (offset 14) = TH1 system return temperature (Rücklauftemperatur System)
5D (offset 19) =Heating Pump Modulation in %

heatpump(0x53) -B-> All(0x00), ?(0x099C), data: 41 04 00 00 00 00 00 00 00 00 00 00 00 0D 7E 2B 00 10 5E 00 02 76 00 05
2B (offset 51) = ER1 Kompressordrehzahl (in %), (rotational speed of compressor in %

MichaelDvP, 2023-02-18:
> 01 4E (offset 4) = TH3 current flow temperature (Vorlauftemperatur)
01 39 (offset 6) = TH2 return temperature (Rücklauftemperatur)
01 31 (offset 14) = TH1 system return temperature (Rücklauftemperatur System)

Hmm, these are the same as i've already added and you've said they *"don't match"*?

I'll add the other to my 3.5.1-dev0

Joern-W, 2023-02-18:
Yes, the values in the ems-esp-dashboard appear to be completly random. I'm attaching pictures to show you the difference.
[image]
[image]

MichaelDvP, 2023-02-18:
Ah, ok, i'll fix it.

Joern-W, 2023-02-18:
Great, thanks

closed 2023-04-23 by proddy