emsesp/EMS-ESP32 #501: All secrets are shown in api call in b17

tp1de, 2022-05-15:
@Proddy

Hi,  I recognized that in version v3.4.0b17  and api call _ems-esp/api/system/settings_  all secrets (passwords) are shown without encryption under "Security". 
I think that this should be changed !

proddy, 2022-05-15:
It's by design, part of https://github.com/emsesp/EMS-ESP32/issues/256#issuecomment-1120366697

This API call only works with an authenticated admin user. It's used to download and make a copy of all the system settings. So everything is exposed.

tp1de, 2022-05-15:
I use _Bypass Access Token authorization on API calls_ within settings - like other users of my ioBroker Adapter as well. 
I am not happy to expose the credentials while using api-interface for integration purposes. (in my case ioBroker).

Even with use of access token, the creditials should not be shown on api calls. It's a security risk.

proddy, 2022-05-15:
I see. The options then are
 - to remove all the sensitive data from the API and JSON file it creates. This could cause problems if someone uploads a settings file and find it no longer connects to their wifi since the password is missing. Need to think about those edge cases.
 -  encrypt all the sensitive data with a hash key. The question is then what should the key be? And if you know it you could still potentially decipher the passwords
 - make the whole JSON file encrypted so it's only used with #256. I'd prefer that, but then it would break your iobroker stuff.

tp1de, 2022-05-15:
Why not api calls w.o. secrets and a seperate JSON-File (encrypted) for saving and restoring settings? 
These are two seperate things .. isn't it?

proddy, 2022-05-15:
having two API calls was the initial idea [here](https://github.com/emsesp/EMS-ESP32/issues/256#issuecomment-1120376744) but then I didn't see a real use case for the `api/settings` call other than to make backup copies. What kind of information do you extract from the settings call?

tp1de, 2022-05-15:
no special info. The adapter just reads all api data provided. I could certainly exclude the settings data. But passwords (wifi, users) should not be exposed without encryption at all by manual web commands.

MichaelDvP, 2022-05-15:
@tp1de we have 2 api calls, the api call without secrets is now the system/info containing all relevant settings. You can skip reading the system/settings api, it creates a lot of double entities and your adapter doesnot make use of the settings. (i have a fork with removed setting and added analog sensors).
If you bypass the security, you don't have security, that's the logic! Why you use the bypass setting? Your adapter works fine with token (for POST), only reading the settings now throws an error because you use GET without the token.

@proddy Maybe dont allow the `bypass token` for the settings-file? But it needs a lot of refactoring in WebAPIService, Command::process and Command::call.

tp1de, 2022-05-15:
Thanks for feedback Michael. I was not aware that token is used for read access of settings as well. I need to adjust my adapter anyhow, since I got errors with b17. I will check to exclude the settings then.

tp1de, 2022-05-15:
@MichaelDvP Do I need to use the token for read as well?

MichaelDvP, 2022-05-15:
For now only system/settings needs the token to read. But you really dont need to read this in your adapter. Reading system/info (without token) should do.

tp1de, 2022-05-16:
@MichaelDvP  Thanks for your pr on my ioBroker Adapter. I implemented the changes within version 1.12.0.

@proddy without _bypass token_  the settings are save - with _bypass token_  all credentials are displayed. 
If this can not be changed within short notice I may recommend to change documentation and make a remark on security issues. (In the moment the token is only mentioned to be needed for http post changes)

proddy, 2022-05-16:
I'll just remove the `ems-esp/api/system/settings` API call. It's not needed anymore.

tp1de, 2022-05-16:
Fine.

closed 2022-05-17 by proddy