emsesp/EMS-ESP32 #503: Set summer mode empty/missing when setting thermostat to "always cooling"

tschamm, 2022-05-17:
**Bug description**
I'm using v3.4.0b18
Using EMS-ESP, I cannot set the heat pump to always cooling. Only options provided are
* auto
* summer
* winter

When setting the heat pump to `always cooling` using the heat pump controller, summer mode will be empty. 

**Steps to reproduce**
*Steps to reproduce the behavior.*
Set the heat pump to `always cooling` in the controller, hc1 set summer mode is empty.

Output of `show` command on telnet, hc1 set summer mode is missing:
```
Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:73.03) (63)
  date/time: 17.05.2022 09:23
  floor drying: off
  damped outdoor temperature: 17.6 °C
  floor drying temperature: 0.0 °C
  building type: medium
  minimal external temperature: -10.0 °C
  dhw mode: low
  dhw circulation pump mode: on
  dhw charge duration: 60.0
  dhw charge: off
  dhw circuit 1 extra: 0.0 °C
  dhw disinfecting: off
  dhw disinfection day: tu
  dhw disinfection time: 120.0
  hc2 selected room temperature: 21.0 °C
  hc2 current room temperature: 22.2 °C
  hc2 mode: manual
  hc2 mode type: comfort
  hc2 eco temperature: 17.0 °C
  hc2 manual temperature: 20.0 °C
  hc2 comfort temperature: 22.0 °C
  hc2 summer temperature: 18.0 °C
  hc2 design temperature: 38.0 °C
  hc2 offset temperature: 0.0 °C
  hc2 min flow temperature: 25.0 °C
  hc2 max flow temperature: 55.0 °C
  hc2 room influence: 0.0 °C
  hc2 room influence factor: 4.0
  hc2 current room influence: 0.0 °C
  hc2 nofrost temperature: 5.0 °C
  hc2 target flow temperature: 22.0 °C
  hc2 heating type: floor
  hc2 summer mode: winter
  hc2 control mode: optimized
  hc2 program: prog_1
  hc2 temporary set temperature automode: -1.0 °C
  hc2 fast heatup: 0 %
  hc1 selected room temperature: 21.0 °C
  hc1 mode: manual
  hc1 mode type: comfort
  hc1 eco temperature: 17.0 °C
  hc1 manual temperature: 21.0 °C
  hc1 comfort temperature: 22.0 °C
  hc1 summer temperature: 18.0 °C
  hc1 design temperature: 38.0 °C
  hc1 offset temperature: 0.0 °C
  hc1 min flow temperature: 25.0 °C
  hc1 max flow temperature: 55.0 °C
  hc1 room influence: 0.0 °C
  hc1 room influence factor: 4.0
  hc1 current room influence: 0.0 °C
  hc1 nofrost temperature: 5.0 °C
  hc1 target flow temperature: 90.0 °C
  hc1 heating type: floor
  hc1 summer mode: winter
  hc1 control mode: optimized
  hc1 program: prog_1
  hc1 temporary set temperature automode: -1.0 °C
  hc1 fast heatup: 0 %
```

**Expected behavior**
*A clear and concise description of what you expected to happen.*
Provide an option to set summer mode to `cooling`.

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

[image]
[image]

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api?device=system&cmd=info*
```
{
  "System Status": {
    "version": "3.4.0b18",
    "uptime": "000+00:31:37.339",
    "freemem": 105,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -61,
    "IPv4 address": "192.168.1.78/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 1188,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Status": {
    "API calls": 1,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 9471,
    "bus reads (tx)": 730,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 5,
    "tx gpio": 17,
    "dallas gpio": 4,
    "pbutton gpio": 33,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx",
      "device id": "0x08",
      "product id": 172,
      "version": "01.20",
      "entities": 102,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x04AA 0x04AB 0x04AD 0x049F 0x04A0 0x04A2 0x04A3 0x04A8 0x04A9 0x04AC 0x04C0 0x04C9 0xF9 0x04A7 0x0499 0xF7 0x0488"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "73.03",
      "entities": 63,
      "handlers received": "0x06 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x031B 0x031E",
      "handlers ignored": "0x02EC 0x04F4 0x0508 0xF9 0x0509 0x050A 0x0507 0x023E 0x0292 0x0239 0xBF 0xF7 0xF6"
    },
    {
      "type": "Mixer",
      "name": "MM100",
      "device id": "0x21",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D8",
      "handlers ignored": "0x0255 0xBF"
    },
    {
      "type": "Heatpump",
      "name": "HP Module",
      "device id": "0x39",
      "product id": 200,
      "version": "40.07",
      "entities": 0,
      "handlers pending": "0x042B 0x047B",
      "handlers ignored": "0xBF"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0
    }
  ]
}
```

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2022-05-17:
Please read (terminal `read 10 471`) or fetch from log (log all) the telegram 471 for hc1 when in cooling mode. And also the telegram 2AF.

tschamm, 2022-05-17:
```
000+03:07:03.906 I 0: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer2(0x0471), data: 03 12 04 01 01 18 04 02
000+03:07:25.714 I 1: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x02AF), data: 00 28 00 00 3C 26 00 00 19 0F 00
```

MichaelDvP, 2022-05-17:
Thanks, so it's value 03 in offset 0 in 471 and offset 6 in 2AF stays zero. To avoid conflicts with gas boilers and the extra option i've added a new entity `hpsummersetmode` with options `off, auto, heating, cooling` and leave the `summersetmode` to gasboilers with option `summer, auto, winter`.  Is this ok? 
[Check here](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest), if all is ok, i'll make a PR.

tschamm, 2022-05-17:
Looks good, thanks for your fast response!

MichaelDvP, 2022-05-17:
If you want to suggest other/better names for entity or options, this is the best time before i make the PR.

proddy, 2022-05-23:
@MichaelDvP shall we change, then close this one?

commit 60726069 (2022-05-20): heatpump summer mode, fix #503
closed 2022-05-23 by MichaelDvP