emsesp/EMS-ESP32 #504: 3.4 b17 - very very small observations

harvey637, 2022-05-17:
Hi @ Proddy,
yupp, I am byk into internet world after offline holiday :-)

Running your b17 is very good, no problems so far. Something very minor, depends on my usage:

I reduce the entities to get a smaller set of api content, adopted for my boiler and use of info in iobroker graphs and stuff. So it is about REDUCING. With the filter I can reduce the entries eg. "show only favourites". But when I REFINE displayed/api entries I work on the inverse filter, like "show only all entries, that still have api/displayed attribute set", and then work on them, eg. giving the favourites attribute to some of them. In b17 I can only HIDE the api/displayed entries, I can not select "SHOW only all api/displayed" entries. SHOW_ONLY_x is then exclusive to HIDE_ ALL_x. The new extra iconscould be alterations of the mqtt/api and the displayed icon, but now WITHOUT the negating diagonal lines on it.

The wording of "mark it as favourite to be listed at the top of the dashboard" is not true anymore (it is not on top), all the wordings in the filter icons are somewhat describing the action of the icon and not the effect of using the filter.
Hide faourites, Hide Read_Only, Hide MQTT/API, Hide visible on dashboard, [new], Show MQTT/APi, Show only visable on dashboard [New]

The "Hide Read_Only" works on the customized Read_Only attribute. I can understand the fully, but the effect looks a bit strange. I might expext to filter all read_only entries. Also the inverse filter looks more natural to me, as I want to play with read/write entries to pollible mark them as erad_only, with filtered read_only I can only work with the other attributes.

Width of table columns (ENTITY-NAMES | VALUE) should be e.g. 50/50% or better 60/40%, as names are often longer than values.

Not shure wether I wrote it understandable: in the top level dashboard the number of SHOWN entries is displayed, I have no clue about the number of hidden entries. So an information like "43 of 115 displayed" might be handy. Also the word "ENTITIES" is fully displayed without opened device, with opened device it is truncated to "ENTITI...". At least with my FHD monitor there is ample space for the full word and more.

Thank you so much, cu
Harvey

proddy, 2022-05-20:
Thanks @harvey637 for the feedback. Hope you had a good holiday

> I reduce the entities to get a smaller set of api content, adopted for my boiler and use of info in iobroker graphs and stuff. So it is about REDUCING. With the customization filter I can reduce the entries eg. "show only favourites". But when I REFINE displayed/api entries I work on the inverse filter, like "show only all entries, that still have api/displayed attribute SET", and then work on them, eg. giving the favourites attribute to some of them or hide some of them from api. In b17 I can only HIDE the api/displayed entries, I can not select "SHOW only all api/displayed" entries. SHOW_ONLY_x is then exclusive to HIDE_ ALL_x. The new extra icons could be alterations of the mqtt/api and the displayed icon, but now WITHOUT the negating diagonal lines on it.

This will add extra code and complexity I'm afraid. I'm trying not to over engineer the customization feature as I don't think many people will use it (and if they do it's set once and forgotten) plus it takes time away from the core of what EMS-ESP is, to read EMS values. Are you able to set all your settings using the current functionality? There is also a way now to save the customization to a file, edit it and then upload it back if that is easier.

> The wording of "mark it as favourite to be listed at the top of the dashboard" is not true anymore (it is not on top), all the wordings in the filter icons are somewhat describing the action of the [...]

very good point, I'll change it to describe the filter action.

> The "Hide Read_Only" works on the customized Read_Only attribute. I can understand the fully, but the effect looks a bit strange. I might expext to filter all read_only entries. Also the inverse [...]

I wanted this too buts it hard. Ideally the icon should show two states and toggle between them with different icons (one solid and one with a diagonal line through it) but I couldn't find an easier way to do this using the native MUI library.

> 
> Width of table columns (ENTITY-NAMES | VALUE) should be e.g. 50/50% or better 60/40%, as names are often longer than values.

Agree, but its really hard. I spent hours trying to get the formatting right and its not easy. Any help looking at the CSS is appreciated! 
 
> Not shure wether I wrote it understandable: in the top level dashboard the number of SHOWN entries is displayed, I have no clue about the number of hidden entries. So an information like "43 of 115 displayed" might be handy.

Here I'm assuming the end user is only interested in seeing the values he/she choose, and what is displayed when you click on an entity (boiler, thermostat etc..). Showing the # hidden entries would just confuse the user.

> Also the word "ENTITIES" is fully displayed without opened device, with opened device it is truncated to "ENTITI...". At least with my FHD monitor there is ample space for the full word and more.

That happens if you resize the window. It's a bug in the library which I'm trying to fix to make sure the table resizes itself when the browser window resizes.

proddy, 2022-05-27:
the rest of the comments we'll take forward in 3.4.x

closed 2022-05-27 by proddy