emsesp/EMS-ESP32 #507: Upgrading firmware looses board profile setting E32 - Ethernet becomes unavailable

Oggerschummer, 2022-05-17:
I just upgraded version 3.4.0b8 to b18 on a freshly received E32 gateway using the web interface while connected to ethernet.

After the upgrade the device was no longer accessible using ethernet which was the case before (no Wifi configured).
Investigations showed that the Wifi access point was available, but the board profile setting for E32 was gone,
causing the ethernet port to be unavailable. 
Just found this by accident, setting the profile to E32 and rebooting the device ethernet was back.

Expectation is that after a firmware update this setting is preserved - "end users" should not have to figure out how to re-enable to ethernet port. At least this should be shown in the documentation for firmware updates. Or did I miss it?

MichaelDvP, 2022-05-17:
The same issue was reported before in #345. But there is no idea why this happened sometimes, what causes it, and how to reproduce it. The settings are stored in a persistent file system in a different partition then OTA. And afaik only the settings file is affected, mqtt, network, AP, etc., stored in the same partition with different filenames are not affected. Proddy and i are updating often and have never seen this.

proddy, 2022-05-17:
I'm 99% sure it was an old bug in 3.4.0b7/b8. I'm surprised we're putting beta versions on gateways anyway - which means we need to get 3.4 out pretty soon (we keep talking about it)

closed 2022-05-17 by proddy