emsesp/EMS-ESP32 #509: How to get organized for pull-request with VSC?

tp1de, 2022-05-18:
I learned how to do code changes and adding new entities / fields into the ems-esp code. #508 
There is a guide "contributing.md" how to make the PR. I just ask myself if my set-up is correct for contributing to pull-request.

I use VSC on a Windows 11 Desktop PC. Within a new folder I issued the command: _git clone https://github.com/emsesp/EMS-ESP32.git_

While using VSC with PlatformIO I could then build new Firmware and could do code changes. So far so good for testing.
All changes are then maintained in VSC:
[image]

Can I just commit this changes into dev version? Or is there another way to do pr's?

MichaelDvP, 2022-05-18:
I always push to my github and start the PR from my github webpage. Maybe there are other ways, but this works.

tp1de, 2022-05-18:
OK. I will try the same using my fork. 

I understood one push for every change (value / entity) or can I do a couple of changes within one push?

So I made my first PR for ems-esp :)

tp1de, 2022-05-20:
@MichaelDvP @proddy May I get your help please:

A you have seen I made my first PR using my fork. I have seen that you merged with other changes.
So I issued and "fetch upstream" in GH on my fork to get the actual version.

When I try to synchronize in VSC I get the following GH error:

> git pull --tags origin dev
From https://github.com/tp1de/EMS-ESP32
 * branch              dev        -> FETCH_HEAD
 ! [rejected]          latest     -> latest  (would clobber existing tag)
 
 Any I idea what I did wrong and how to correct it?

tp1de, 2022-05-20:
... I found a VSC settings to disable git pull tags. But now I struggle with changes on my forked repository and changes I have done in the meantime in VSC to test a new entity.

What would be the right behavior? 
Wait for any new changes until you have done the merge? ... or any other way to do it?

MichaelDvP, 2022-05-20:
You can merge the remote to your local repo if there are differences. First commit your changes to local repo, then fetch and merge the remote. If you have Git-Gui instaled, this is sometimes easier as commandline or VSC to manage commits and branches.

tp1de, 2022-05-20:
Thanks. I was not clever enough to commit my changes to local repro before fetch and merge the remote. Now I have in VSC conflicts.

Any proposal what to do?

MichaelDvP, 2022-05-20:
Conflicts are shown in VSC like below with the options how to resolve. Resolve the conflicts, save files, stage (+ behinde filename), then you can commit the merge.
[image]

tp1de, 2022-05-20:
Yes that's what I have done. After committing the merge I had double entries in boiler.h . Deleting these double entries, the build works again but I now have 3 commits on my fork. 1 for the changes for a new entity (wwmaxtemp) and one for the Merge and one for the correction of boilers.h.

This does not look correct. Any Idea how to get this corrected?

[image]

tp1de, 2022-05-21:
@MichaelDvP @proddy 

Please advice what to do. Can I continue with these 3 commits or do I have to reset something?

I would like to continue with other changes / additions but I am not sure if everything is right or might I loose further changes / commits when I just go on?

proddy, 2022-05-21:
just do a PR and we'll clean it up centrally.

The way I work is a combination between VSC, GitHub CLI and GitHub web as I don't always trust the CLI. I do
 - go to my fork of EMS-ESP32 on github.com, select the dev branch and click on Fetch upstream.  
 - in VSC refresh dev branch but clicking the 'synchronize changes' circle icon bottom left
 - work on the local branch, then commit locally 
 - I use WSL2 within Windows11 as running VSC and platformio under Ubuntu is 2x faster, specially compiling. It also allows me to build and test standalone without ever uploading to an ESP32 chip.
 - then if I want a PR I also use github.com's web site and click Contribute

you can do all of this within VSC too (just "Pull, Push" -> "Pull from..." -> upstream) but sometimes I run into merge conflicts. If you prefer command lines you can do `git ftech upstream; git pul upstream dev` which does the same. If you have issues with tags no appearing in VSC open a terminal window and type `git pull —tags -f`

MichaelDvP, 2022-05-21:
Yes this happens sometimes, git shows the commits you have done, there is nothing wrong. But you need to do one commit more to add the formatting back (edit boiler.h, press Alt-Shift-F). Or you do an "Undo last commit", open boiler.h in editor and press Alt-Shift-F for formatting, stage and commit again, but this will throw errors when trying to push to githube, because the commits are different now.
Best to avoid this is to take more care when solving conflicts.
In Git-Gui you can show the branch history, reset branch to proddys last commit in mixed mode, and then stage and commit your changes. If you want to avoid losing changes by accident, create a new branch first and try out with this new branch. If all is ok, you can delete the dev-branch and rename the current branch to dev.

tp1de, 2022-05-21:
Thanks @proddy for your fast reply. Understood and I will make a PR now.
I was just not sure if I do something wrong ... I am not an IT guy.

tp1de, 2022-05-21:
@MichaelDvP Thanks as well ..... I still struggle to fully understand the GH concept of committing, merging, undoing commits and staging.
Within ioBroker it is easy since I am the only one to do changes and just (seldom) get a PR.

But the ems-esp software is quite an excellent masterpiece of software. Congratulations to you both.

closed 2023-02-19 by proddy