emsesp/EMS-ESP32 #517: Problem with connecting to Junkers Cerapur Modul

matigda123, 2022-05-22:
Can read information from Junkers Cerapur Modul, but not able to make any changes via tx.
[image]

matigda123, 2022-05-22:
[image]
[image]

matigda123, 2022-05-22:
[log.txt](https://github.com/emsesp/EMS-ESP32/files/8750008/log.txt)

MichaelDvP, 2022-05-23:
Please update to latest dev-build (reload page in brower after update) and include the debug information you'll find under "help". 
This looks like you have selected tx-mode off, but we'll see in the debug info.

matigda123, 2022-05-24:
I've updated to the latest dev-build and still can't get any result. The log attached.
[ems log.txt](https://github.com/emsesp/EMS-ESP32/files/8765054/ems.log.txt)
[image]
[image]

matigda123, 2022-05-24:
000+00:04:02.403 DEBUG 843: [emsesp] No telegram type handler found for ID 0x05 (src 0x08)
000+00:04:03.918 DEBUG 844: [telegram] Sending read Tx [#65], telegram: 8B 90 02 00 1B
000+00:04:05.167 DEBUG 845: [telegram] Sending read Tx [#192], telegram: 8B 90 02 00 1B
000+00:04:06.413 DEBUG 846: [telegram] Sending read Tx [#193], telegram: 8B 90 02 00 1B
000+00:04:07.660 DEBUG 847: [telegram] Sending read Tx [#194], telegram: 8B 90 02 00 1B
000+00:04:07.692 ERROR 848: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 02 00 1B
000+00:04:07.692 WARNING 849: [emsesp] Unrecognized EMS device (device ID 0x10, no product ID). Please report on GitHub.
000+00:04:08.908 DEBUG 850: [telegram] Sending read Tx [#77], telegram: 8B 89 02 00 1B
000+00:04:10.082 DEBUG 851: [mqtt] Publishing topic ems-esp/boiler_data (#275, retain=0, retry=1, size=399, pid=1)
000+00:04:10.155 DEBUG 852: [telegram] Sending read Tx [#195], telegram: 8B 89 02 00 1B
000+00:04:10.183 DEBUG 853: [mqtt] Publishing topic ems-esp/boiler_data_ww (#276, retain=0, retry=1, size=283, pid=1)
000+00:04:10.284 DEBUG 854: [mqtt] Publishing topic ems-esp/dallassensor_data (#277, retain=0, retry=1, size=62, pid=1)
000+00:04:10.749 DEBUG 855: [emsesp] No telegram type handler found for ID 0xA2 (src 0x10)
000+00:04:10.749 DEBUG 856: [telegram] Tx read request to device 0x10 for type ID 0x02
000+00:04:11.652 DEBUG 857: [telegram] Sending read Tx [#196], telegram: 8B 89 02 00 1B
000+00:04:12.344 DEBUG 858: [emsesp] No telegram type handler found for ID 0x05 (src 0x08)
000+00:04:12.647 DEBUG 859: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
000+00:04:12.647 DEBUG 860: [telegram] Tx read request to device 0x09 for type ID 0x02
000+00:04:12.647 DEBUG 861: [emsesp] New EMS device detected with ID 0x10. Requesting version information.
000+00:04:12.647 DEBUG 862: [telegram] Tx read request to device 0x10 for type ID 0x02
000+00:04:14.147 DEBUG 863: [telegram] Sending read Tx [#198], telegram: 8B 89 02 00 1B
000+00:04:14.179 ERROR 864: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 89 02 00 1B
000+00:04:14.179 WARNING 865: [emsesp] Unrecognized EMS device (device ID 0x09, no product ID). Please report on GitHub.
000+00:04:15.395 DEBUG 866: [telegram] Sending read Tx [#78], telegram: 8B 90 02 00 1B
000+00:04:16.642 DEBUG 867: [telegram] Sending read Tx [#201], telegram: 8B 90 02 00 1B
000+00:04:17.889 DEBUG 868: [telegram] Sending read Tx [#202], telegram: 8B 90 02 00 1B
000+00:04:19.137 DEBUG 869: [telegram] Sending read Tx [#203], telegram: 8B 90 02 00 1B
000+00:04:19.168 ERROR 870: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 02 00 1B
000+00:04:19.168 WARNING 871: [emsesp] Unrecognized EMS device (device ID 0x10, no product ID). Please report on GitHub.
000+00:04:20.093 DEBUG 872: [mqtt] Publishing topic ems-esp/boiler_data (#278, retain=0, retry=1, size=397, pid=1)
000+00:04:20.194 DEBUG 873: [mqtt] Publishing topic ems-esp/boiler_data_ww (#279, retain=0, retry=1, size=283, pid=1)
000+00:04:20.295 DEBUG 874: [mqtt] Publishing topic ems-esp/dallassensor_data (#280, retain=0, retry=1, size=62, pid=1)
000+00:04:20.384 DEBUG 875: [telegram] Sending read Tx [#91], telegram: 8B 90 02 00 1B
000+00:04:21.632 DEBUG 876: [telegram] Sending read Tx [#204], telegram: 8B 90 02 00 1B
000+00:04:22.573 DEBUG 877: [emsesp] No telegram type handler found for ID 0x05 (src 0x08)
000+00:04:23.628 DEBUG 878: [telegram] Sending read Tx [#205], telegram: 8B 90 02 00 1B
000+00:04:24.875 DEBUG 879: [telegram] Sending read Tx [#206], telegram: 8B 90 02 00 1B
000+00:04:24.906 ERROR 880: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 02 00 1B
000+00:04:24.906 WARNING 881: [emsesp] Unrecognized EMS device (device ID 0x10, no product ID). Please report on GitHub.
000+00:04:26.122 DEBUG 882: [telegram] Sending read Tx [#92], telegram: 8B 90 02 00 1B
000+00:04:27.370 DEBUG 883: [telegram] Sending read Tx [#207], telegram: 8B 90 02 00 1B
000+00:04:28.617 DEBUG 884: [telegram] Sending read Tx [#208], telegram: 8B 90 02 00 1B
000+00:04:29.865 DEBUG 885: [telegram] Sending read Tx [#209], telegram: 8B 90 02 00 1B
000+00:04:29.896 ERROR 886: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 02 00 1B
000+00:04:29.896 WARNING 887: [emsesp] Unrecognized EMS device (device ID 0x10, no product ID). Please report on GitHub.
000+00:04:30.092 DEBUG 888: [mqtt] Publishing topic ems-esp/boiler_data (#281, retain=0, retry=1, size=399, pid=1)
000+00:04:30.193 DEBUG 889: [mqtt] Publishing topic ems-esp/boiler_data_ww (#282, retain=0, retry=1, size=283, pid=1)
000+00:04:30.294 DEBUG 890: [mqtt] Publishing topic ems-esp/dallassensor_data (#283, retain=0, retry=1, size=62, pid=1)
000+00:04:31.112 DEBUG 891: [telegram] Sending read Tx [#102], telegram: 8B 90 02 00 1B
000+00:04:32.303 DEBUG 892: [emsesp] No telegram type handler found for ID 0x05 (src 0x08)
000+00:04:33.108 DEBUG 893: [telegram] Sending read Tx [#210], telegram: 8B 90 02 00 1B
000+00:04:34.355 DEBUG 894: [telegram] Sending read Tx [#211], telegram: 8B 90 02 00 1B
000+00:04:35.449 DEBUG 895: [emsesp] Process setting 0x1A for device 0x08
000+00:04:35.507 DEBUG 896: [emsesp] No telegram type handler found for ID 0x267 (src 0x10)
000+00:04:35.507 DEBUG 897: [telegram] Tx read request to device 0x10 for type ID 0x02
[... cut ...]
000+00:04:39.003 DEBUG 908: [telegram] Sending read Tx [#217], telegram: 8B 90 02 00 1B
000+00:04:40.037 DEBUG 909: [mqtt] Publishing topic ems-esp/boiler_data (#284, retain=0, retry=1, size=399, pid=1)
000+00:04:40.138 DEBUG 910: [mqtt] Publishing topic ems-esp/boiler_data_ww (#285, retain=0, retry=1, size=283, pid=1)
000+00:04:40.239 DEBUG 911: [mqtt] Publishing topic ems-esp/dallassensor_data (#286, retain=0, retry=1, size=62, pid=1)
000+00:04:40.244 DEBUG 912: [telegram] Sending read Tx [#218], telegram: 8B 90 02 00 1B

matigda123, 2022-05-24:
[image]

MichaelDvP, 2022-05-24:
Yes, there is something not working with tx, i don't  know what and can't remember how to analyse in old version. 
I've ask for this information.:
[image]
Have you alreadytried a different tx-mode as suggested in EMS Status?

matigda123, 2022-05-25:
I've tried a different tx mode with no results. In the log file I also found this information, can this cause an issue?
" [emsesp] Unrecognized EMS device (device ID 0x10, no product ID). Please report on GitHub. "

Debug information attached
[EMS ESP INFO.zip](https://github.com/emsesp/EMS-ESP32/files/8768314/EMS.ESP.INFO.zip)

MichaelDvP, 2022-05-25:
You should not post the settings-file, it contains all passwords, i suggest to change your passwords now.

You are using a custom board profile, is the hardware homemade? Are you sure tx pin is selected right?
The setting with tx mode 2 (ht3) should work with Junkers models, so i think something is wrong with tx hardware.

matigda123, 2022-06-01:
The board is home made, I'll try to make another one. However, what does it mean " [emsesp] Unrecognized EMS device (device ID 0x10, no product ID). Please report on GitHub. " on the log file?

proddy, 2022-07-10:
close due to inactivity

closed 2023-02-19 by proddy