emsesp/EMS-ESP32 #522: Controller showing in WebUI customization

proddy, 2022-05-26:
@MichaelDvP 

with the addition of 0x06/RCTime being added to the Controller device object everyone will now see this pop up in the Customization page like:

[image]

although it will never show in up the Web Dashboard, MQTT/API or Console.

Any ideas on how to prevent it showing? e.g. just ignore the Controller type in the Customization or only add this device entity if the boiler type is IVT ?

MichaelDvP, 2022-05-26:
It is shown in dashoard for IVT, if it have a value [https://github.com/emsesp/EMS-ESP32/blob/60714b0b0b2cfa913064cb5346847a956f0b6ad4/src/web/WebDataService.cpp#L81](https://github.com/emsesp/EMS-ESP32/blob/60714b0b0b2cfa913064cb5346847a956f0b6ad4/src/web/WebDataService.cpp#L81)

If you don't like it in customizations, maybe dont show devices with no active entities. But the customization shows all registered values, it's consequent to show the controller value too.

proddy, 2022-05-26:
There is logic in boiler and thermostat to only register the entities that are supported on that particular hardware. This is what the device flags are used for. I think we should do the same for the Controller.

MichaelDvP, 2022-05-26:
Yes, it's only the IVT controller, i'll add a model-flag.

proddy, 2022-05-27:
fixed in 3.4.1

commit 77a8857e (2022-05-26): fix #522, device-flag for IVT controller
commit 9d25f804 (2022-05-26): Merge pull request #525 from MichaelDvP/dev

fixes #522, #523, #524
closed 2023-02-19 by proddy