emsesp/EMS-ESP32 #523: hidden entity hc->climate showing up in WebUI customization

proddy, 2022-05-26:
@MichaelDvP 

The hc->climate which I think is used to automatically fall back to the Thermostats roomTemp if there is no selTemp is appearing in the WebUI customization page.

[image]

It's an internal entity and never show in the console/mqtt/api/dashboard etc...

Question: how best to exclude it from the list?

MichaelDvP, 2022-05-26:
It's used to generate the HA climate config, not shown in dashboard, but exclude from mqtt is used to disable the climate config.

proddy, 2022-05-26:
yes, but how to exclude it from showing up in the Customization list?

MichaelDvP, 2022-05-26:
If you exclude it from list, you have no way to disable it.

proddy, 2022-05-26:
ok, I looked at the code and understand now. It's not a real EMS device entity but a derived one and used to determine if the HA's climate component should be created. It's never shown in the API/MQTT/Console/Web output as it's a internal hidden value. If you disable it from Customizations it will prevent the HA climate discovery topic from being created.

It may confuse people (like me!) so best to rename to ha_climate and give it a full-name

MichaelDvP, 2022-05-26:
Yes, it's a bit confusing, but it's hard to create or not create the right climate config, it takes a lot of tries to find this. If you add a full-name it will show up on dashboard and mqtt, maybe use a "short-name" like `ha_climate_config_creation`, since it it not shown in mqtt, the name will not blow up the mqtt.

SpaceTeddy, 2022-05-27:
for me the same. customization menu shows items, which i can not see in the device overview.
i.E.

[image]

[image]

commit d2609e42 (2022-05-26): fix #523, rename 'climate' to more explaning name
closed 2022-05-27 by proddy