emsesp/EMS-ESP32 #524: software reset in ems-esp32 v3.4.1b0

harvey637, 2022-05-26:
Hi,
as you know I am running the dev versions, mostly ok, some observations, but now (getting more often): Systemreset

Background:
I use ems-esp32 mainly for fetching lots of infos into infuxdb, mostly with iobroker-plugin, but some values (around 8) are fetched with wget directly from api. This happens every 5 minutes.
Second use is to manipulate the thermostat.seltemp to substract/add 2°C from the roomtemparature, as the thermostat is only able to do 2 actions/per day. So I reduce temperature while I am out of home working and switch back before I get home.

When testing manually with lots (5~10 api requests over 2~3 minutes) sometimes I get reboots. Yesterday I got reboot during normal dayly routine, meaning 120s fetch interval for iobroker-plugin, 5 min interval raspberymatich wget fetching, and two! writings of seltemp.

The first in the morning no problem, but from 17:05 (setting temp plus 2°C) until 21:35 I get errors on the wget-script, every 5 minutes. 

This is the fetching call (with later getting the values and putting them into variables):
```
    set url "http://ems-esp/api/boiler"
    exec /usr/bin/wget --no-check-certificate -q -O /usr/local/addons/solaredge/bosch-boiler.xml $url

    set f [open "/usr/local/addons/solaredge/bosch-boiler.xml"]
    set input [read $f]
    close $f
```
and the writing call:
```
new_temp=$(echo "$1" | sed -e"s/00000$//")
curl http://ems-esp.fritz.box/api/thermostat/hc1/seltemp \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer eyJ...._my_api_key_.........................dDg' \
  -d "{ \"value\" : $new_temp }"

  sleep 4

/usr/bin/wget --no-check-certificate -q http://ems-esp/api/thermostat/hc1/seltemp/value -O -
echo
```

Interesting story is, that the iobroker-plugin seems to read values during the timeout, only the wget... from the api is broken. At the end, after some hours, ems-esp32 crashes/reboots and works again (but customizations are lost :-)

sadly no no log before the reboot.

I can reproduce crashes/reboots with lots of wget-calls.
```
2022-05-26 11:00:04.795   DEBUG 178:    [command]    Calling command 'thermostat/seltemp' (selected room temperature) with value 21
2022-05-26 11:00:05.073   DEBUG 179:    [telegram]   Sending write Tx [#205], telegram: 8B 10 FF 08 01 B9 2A
2022-05-26 11:00:05.073   DEBUG 180:    [emsesp]     Last Tx write successful
2022-05-26 11:00:05.073   DEBUG 181:    [telegram]   Sending post validate read, type ID 0x2A5 to dest 0x10
2022-05-26 11:00:05.073   DEBUG 182:    [emsesp]     Process setting 0x2B9 for device 0x10
2022-05-26 11:00:05.073   DEBUG 182:    [emsesp]     Process setting 0x2B9 for device 0x10
2022-05-26 11:00:07.285   DEBUG 183:    [telegram]   Sending read Tx [#206], telegram: 8B 90 FF 00 19 01 A5
2022-05-26 11:00:07.378   DEBUG 184:    [emsesp]     Last Tx read successful
2022-05-26 11:00:07.649   DEBUG 185:    [telegram]   Sending read Tx [#207], telegram: 8B 90 FF 19 20 01 A5
2022-05-26 11:00:07.715   DEBUG 186:    [emsesp]     Last Tx read successful
2022-05-26 11:00:08.117   DEBUG 187:    [emsesp]     This telegram (RC300Monitor) is not recognized by the EMS bus
2022-05-26 11:00:08.117   DEBUG 188:    [emsesp]     No telegram type handler found for ID 0x2A6 (src 0x10)
2022-05-26 11:00:08.553   DEBUG 189:    [emsesp]     No telegram type handler found for ID 0x42B (src 0x38)
2022-05-26 11:00:08.586   DEBUG 190:    [emsesp]     No telegram type handler found for ID 0x47B (src 0x38)
2022-05-26 11:00:08.891   ERROR 191:    [emsesp]     Call failed with error code (Error)
2022-05-26 11:00:11.106   DEBUG 192:    [emsesp]     No telegram type handler found for ID 0xF9 (src 0x10)
2022-05-26 11:00:12.218   DEBUG 193:    [emsesp]     No telegram type handler found for ID 0xF9 (src 0x10)
2022-05-26 11:00:12.357   DEBUG 194:    [emsesp]     No telegram type handler found for ID 0x49F (src 0x08)
```
Now I can try to read values, not working:
```
  /usr/bin/wget --no-check-certificate -q http://ems-esp/api/boiler -O -
null
```

```
2022-05-26 11:03:53.257   DEBUG 629:    [command]    Calling command 'system/info' (show system status)
2022-05-26 11:03:56.839   DEBUG 630:    [command]    Calling command 'boiler/values' ()
2022-05-26 11:03:58.012   DEBUG 631:    [emsesp]     No telegram type handler found for ID 0x2F6 (src 0x10)
2022-05-26 11:04:00.012   DEBUG 632:    [emsesp]     No telegram type handler found for ID 0x2F6 (src 0x10)

```
Same problem in normal browser call!
After reboot of emsesp32 v3.4.1b0 everything works again!!!!!

If you need any extra info please ask. I can easy reproduce the problem with several (3~5) api call to set seltemp.

have a nice day (special day for fathers in germany)
cu
Harvey

harvey637, 2022-05-26:
Hi, 
some additional info:
when reading the full status of an entry everything works, even after fetching 20 times in short time.
```
/usr/bin/wget --no-check-certificate -q http://ems-esp/api/thermostat/hc1/seltemp/ -O -
{
  "name": "seltemp",
  "fullname": "hc1 selected room temperature",
  "circuit": "hc1",
  "value": 21,
  "type": "number",
  "min": 0,
  "max": 29,
  "uom": "°C",
  "readable": true,
  "writeable": true,
  "visible": true
root@homematic-raspi:

```

PS: looks like a missing closing parenthesis?

This call crashs after 3 to 5 repetitions:
```
 /usr/bin/wget --no-check-certificate -q http://ems-esp/api/thermostat/hc1/seltemp -O -
nullroot@homematic-raspi:/usr/local/addons/solaredge#
root@homematic-raspi:/usr/local/addons/solaredge#  /usr/bin/wget --no-check-certificate -q http://ems-esp/api/thermostat/hc1/seltemp/value -O -
root@homematic-raspi:/usr/local/addons/solaredge#
root@homematic-raspi:/usr/local/addons/solaredge#
```

After reboot:
```
root@homematic-raspi:/usr/local/addons/solaredge#  /usr/bin/wget --no-check-certificate -q http://ems-esp/api/thermostat/hc1/seltemp/value -O -
21root@homematic-raspi:/usr/local/addons/solaredge#
```

good luck hunting!
cu
Harvey

MichaelDvP, 2022-05-26:
Thanks for the finding, in api the json response is not deleted, and heap is getting low until crash. Sadly this is now also in the master build. For the dev i'll make a PR. If you compile youre own, add before this line: [https://github.com/emsesp/EMS-ESP32/blob/60714b0b0b2cfa913064cb5346847a956f0b6ad4/src/web/WebAPIService.cpp#L135](https://github.com/emsesp/EMS-ESP32/blob/60714b0b0b2cfa913064cb5346847a956f0b6ad4/src/web/WebAPIService.cpp#L135)
`delete response;`

proddy, 2022-05-26:
good catch, I should have thought about that. We can make a 3.4.1 to the main build with these fix and it's a major bug.

MichaelDvP, 2022-05-27:
No, your adapter only uses the json output from `api/thermostat`and `api/thermostat/seltemp`. Since v3.4.0b16 we have also the possibility to read a single item as plain text like `api/thermostat/seltemp/max`. In this case the value is send as plain text and json memory keeps allocated. I'm mainly using iobroker-mqtt with single values on change, but your adapter is also running and i have no reboots aand memory is stable.

proddy, 2022-05-27:
fixed in latest beta

commit 34cb3ad3 (2022-05-26): fix #524, free memory of json response
closed 2022-05-27 by proddy