emsesp/EMS-ESP32 #535: Elektrisch Element toevoegen ?

willemsenior, 2022-05-30:
I would like to monitor the electrical element
can you add those sensors
they could be seen at the Husdate box
namely in 3 steps of 33/66 and 100 percent
so you can see if the electric element is switched on during a defrost or legionella run
BEST REGARDS

**Describe the solution you'd like**
elektrisch element in drie stappen zichtbaar met uw EMS-ESP32 kastje
33
66
100 procent vermogen

picture Husdate box
[image]

proddy, 2022-06-01:
where does the data come from? Is it in the EMS telegrams from the Boiler? I don't know what Husdate is

proddy, 2022-06-01:
you'll need to find out where the data is yourself, in which telegram and which position. And then we can add to EMS-ESP. The website says the H60 is opensource in GitHub so that might give you some pointers.

willemsenior, 2022-06-01:
with Home Assistant i use this code in Yaml file to monitor the electriche element 

- platform: mqtt
    state_topic: "fcf5c49281ee/HP/3104"
    name: "Additional electric heater"
    unit_of_measurement: "%"
    icon: mdi:flash

maybe that helps you

proddy, 2022-06-01:
> okay unfortunately I don't understand that so I won't find that code or telegram thats why i ask it to add it to the EMS-ESP32 as standard sensor/code for the EMS-EPS32 box

me neither. I don't own an Enviline or a Husdate so there's no way to find out where the data is stored in the EMS telegrams.

closed 2022-08-27 by willemsenior