emsesp/EMS-ESP32 #536: FR100 - thermostat time correction from ntp

odelcourt, 2022-06-02:
Since update to v3.4.1 (coming from 3.3.1), every second I have this message in console:
     000+00:54:48.710 I 2158: [thermostat] thermostat time correction from ntp

[image]

it seems to "overwhelm" the thermostat because it respond much slower to command from MQTT (changing mode, tuning one wwonetime,...). I have deactivated the NTP sync on the EMS and I have no more the message in the console and the thermostat react to command "as usual" (fast). It seems that the "thermostat time correction from ntp" occupies the EMS bus.

My Thermostat is a BOSCH/JUNKERS FR100 . Do you need other information(s) ? Can I actually do something else than deactivate the NTP sync ?

[image]

odelcourt, 2022-06-02:
> That's one option, second is to set the clock correctly, than it needs no correction.

I have checked / tried that with no success :-(
I had not that problem before, is it something that has been changed with the version 3.4.x ?

MichaelDvP, 2022-06-02:
v3.4 checks the clock and if thermostat clock differs much from NTP it writes the corrected time to the clock, see #386. But for some thermostats the clock is not writeable via ems-bus. Sad that this was not detected in beta tests.

odelcourt, 2022-06-02:
> v3.4 checks the clock and if thermostat clock differs much from NTP it writes the corrected time to the clock, see #386. 

OK, this explains this.

>But for some thermostats the clock is not writeable via ems-bus. Sad that this was not detected in beta tests.

Bah. You can't test all with all hardware.... and the workaround was easy. Thank you for the follow of the tickets.

odelcourt, 2022-06-08:
> Can we close this?

How can I test this  ? I guess I have to install a dev version? But mine EMS-ESP32 is used in "production", thus have you a dev build "safe" ? Otherwise, yes you can close this but I will only be able to test it with the next official release. Many thanks for your support and development.

MichaelDvP, 2022-06-08:
I've made a test branch based on main build and only add this patch. You can [check here](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest), [source here](https://github.com/MichaelDvP/EMS-ESP32/tree/main_patched), and after test you can also go back to previous builds.

odelcourt, 2022-06-09:
> I've made a test branch based on main build and only add this patch. You can [check here](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest), [source here](https://github.com/MichaelDvP/EMS-ESP32/tree/main_patched), and after test you can also go back to previous builds.

Tested successfully. I'll keep that version running for now. Many thanks !

commit 794b3c04 (2022-06-03): Merge pull request #538 from MichaelDvP/dev

fix #536 and uart task priority
commit e7bcc380 (2022-06-03): fix #536, FR100 datetime not writable
closed 2022-06-09 by odelcourt