emsesp/EMS-ESP32 #541: Rest command to change dhw mode, not recognize argument.

Arkaqius, 2022-06-04:
**Bug description**
*I am sending REST command to change  dhw mode on thermostat. As result i got _Call failed with error: callback function failed_ . (Error). On previous software thre was another error description that argument is invalid.
REST command:
url: http://[REDACTED]/api/thermostat/wwmode
    method: POST
    headers:
      authorization: !secret bearer_ems_esp_token
    payload: "{'value': 'off' }"
    content_type:  'application/json'
    verify_ssl: true
Changing mode via ems-esp page is working corretly, i saw that it's send same arguments (off/high).*

**Steps to reproduce**
*1. Send REST command as listed in descritpion.
 2. Check dhw mode - > Shall be off -> System returns that _call failed with error callback function failed_*

**Expected behavior**
*DHW mode sets to off.*

**Screenshots**
I am attaching 2 screenshosts from system logs.
[image]
[image]

**Device information**
*{
  "System Status": {
    "version": "3.4.1",
    "uptime": "004+23:10:33.357",
    "freemem": 151,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -68,
    "IPv4 address": "192.168.0.4/255.255.255.0",
    "IPv4 gateway": "192.168.0.1",
    "IPv4 nameserver": "192.168.0.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": true,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 1301683,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 10,
    "clean session": true,
    "base": "ems",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": true,
    "publish time boiler": 1,
    "publish time thermostat": 1,
    "publish time solar": 1,
    "publish time mixer": 1,
    "publish time other": 1,
    "publish time sensor": 1,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Status": {
    "API calls": 16,
    "API fails": 6
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 413668,
    "bus reads (tx)": 183,
    "bus writes (tx)": 5,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": true,
    "shower alert coldshot": 10,
    "shower alert trigger": 7,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i",
      "device id": "0x08",
      "product id": 123,
      "version": "06.03",
      "entities": 66,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA",
      "handlers ignored": "0x04 0x25 0x27"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "18.05",
      "entities": 41,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x031E",
      "handlers ignored": "0xBF 0xF7"
    },
    {
      "type": "Controller",
      "name": "Controller",
      "device id": "0x09",
      "product id": 152,
      "version": "03.01",
      "entities": 0
    }
  ]
}*

proddy, 2022-06-04:
Thanks for reporting this. We'll check.

proddy, 2022-06-04:
I can't get it to fail within the test scripts which simulate an RC3x0. Does a CLI call like

```
% curl http://ems-esp.local/api/thermostat/wwmode \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ8.eyJ1c2VybmFtZSI6ImFkbWluIiwiYWRtaW4iOnRydWUsInZlcnNpb24iOiIzLjEuMWIwIn0.qeGT53Aom4rDYeIT1Pr4BSMdeWyf4_zN9ue2c51ZnM0' \
  -d '{ "value" : "off" }'
```

also give the same error?

Arkaqius, 2022-06-05:
[image]
Hello, it's works! :) 
I was sending from HomeAssistant only key from json (off/high).
Example from guys who use REST from HA integration:
It's better to create whole json if using templates like:
"{% if is_state('input_boolean.ww_state','off') %}{ 'value' : 'off' }{% else %}{ 'value' : 'high' }{% endif %}"

Ticket can be closed, thanks for fast help and example that it's really works :)
Thanks a lot 👍

closed 2022-06-05 by Arkaqius

proddy, 2022-06-05:
glad it works now. This is a good example to add to the wiki