emsesp/EMS-ESP32 #543: No automatic MQTT reconnect after software restart in 3.4.1b2

gaszu, 2022-06-06:
After software restart don't start automatic mqtt connection - I must to go to mqtt settings tab, click save, than mqtt connects to my ha server. On previous versions - no problem - only in 3.4.1b2. No hardware changes beetween present and previous version.

proddy, 2022-06-06:
This is Ethernet correct? And you can confirm this happens all the time, so from the EMS-ESP web page click on 'Restart' it will not connect to the MQTT server and you'll see these errors in the Console log?

gaszu, 2022-06-06:
No errors on ethernet. The same situation when i'm connected via wifi. This time only ethernet. faster and more stable, and more bandwith :)

proddy, 2022-06-11:
I can't reproduce. You will need to compile the code yourself and use the debug flags (see pio_local.ini_example) and watch the MQTT client logs in the serial console.

proddy, 2022-07-19:
@mtc716 anything happen up to this point? i.e. wifi reconnected, MQTT broker restarted etc...

mtc716, 2022-07-19:
> @mtc716 anything happen up to this point? i.e. wifi reconnected, MQTT broker restarted etc...

Nothing at all. It happens just randomly during normal operation. Sometimes after 1 day, sometimes after 2-4 days.

proddy, 2022-07-19:
> > @mtc716 anything happen up to this point? i.e. wifi reconnected, MQTT broker restarted etc...
> 
> Nothing at all. It happens just randomly during normal operation. Sometimes after 1 day, sometimes after 2-4 days.

if the MQTT disconnects there should be an INFO log message. Do you see this in the system log? If it happens after a few days its easier to use a SysLog server to capture the logs. In the next build I'll change it to Warning to make it easier to spot.

mtc716, 2022-07-19:
> if the MQTT disconnects there should be an INFO log message. Do you see this in the system log? If it happens after a few days its easier to use a SysLog server to capture the logs. In the next [...]

Will check for it and I guess it should reoccure latest tomorrow....

mtc716, 2022-07-20:
> Will check for it and I guess it should reoccure latest tomorrow....

It happened again and just before the disconnect, the WiFi was disconnected according to the logs....

2022-07-20 09:26:38.109    INFO 18305:  [emsesp]     WiFi connection dropped, stopping NTP
2022-07-20 09:26:38.109    INFO 18306:  [system]     NTP disconnected
2022-07-20 09:26:38.109    INFO 18307:  [emsesp]     WiFi disconnected. Reason code=3
2022-07-20 09:26:38.109    INFO 18308:  [mqtt]       MQTT disconnected: TCP
2022-07-20 09:26:42.975    INFO 18309:  [emsesp]     Starting NTP
2022-07-20 09:26:42.978    INFO 18310:  [emsesp]     WiFi connected with IP=192.168.178.36, hostname=ems-esp-new
2022-07-20 09:26:43.019    INFO 18311:  [emsesp]     mDNS responder service started
2022-07-20 09:26:45.655    INFO 18312:  [system]     NTP connected

proddy, 2022-07-20:
thanks for checking. The reason for the wifi says "WIFI_REASON_AUTH_LEAVE" which doesn't tell us much. Your wifi is down for about 4 minutes. I'll run some tests here again to see if the MQTT reconnects when the wifi is off for a long period.

proddy, 2022-07-20:
I can't reproduce. I used connected the EMS-ESP to a Wifi AP, then switched off the AP and back on after a few minutes. As soon as the WiFi reconnected so did MQTT. @mtc716 could you share your MQTT settings in EMS-ESP? Are you using authentication, clean session....

mtc716, 2022-07-20:
> I can't reproduce. I used connected the EMS-ESP to a Wifi AP, then switched off the AP and back on after a few minutes. As soon as the WiFi reconnected so did MQTT. @mtc716 could you share your MQTT settings in EMS-ESP? Are you using authentication, clean session....

I even had it a second time today. Settings for MQTT are below: 
[image]

proddy, 2022-07-20:
Can you try the latest beta? Looks like you're using an older firmware. Not sure it'll make a difference but worth trying. The problem we need to fix is why EMS-ESP loses the WiFi connectivity. The system information we need you can download from the Help page.

mtc716, 2022-07-20:
The problem occured with the latest Beta 3.4.2b3. I switched today back to 3.4.1 to check If it happens as well...

Do you need the debug information?

proddy, 2022-07-20:
ah ok. The emsesp_info.txt file which you can download from the Help page (latest beta) would be useful as it shows the settings and also your WiFi strength (called RSSI). If the WiFi is unstable you can try playing with the settings like using a lower bandwidth or turning on nosleep.

tp1de, 2022-07-20:
could the instability come from the mqtt broker? Since I installed a new mqtt broker within homeassistant as add in it seem to be stable now.

mtc716, 2022-07-21:
> ah ok. The emsesp_info.txt file which you can download from the Help page (latest beta) would be useful as it shows the settings and also your WiFi strength (called RSSI). If the WiFi is unstable you can try playing with the settings like using a lower bandwidth or turning on nosleep.

the file can be found below. As you will see the RSSI is good and by the way, it worked since month at the same location. The difference is just the sw-version. If I can be of any further help, let me know.

[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/9156294/emsesp_info.txt)

mtc716, 2022-07-21:
> could the instability come from the mqtt broker? Since I installed a new mqtt broker within homeassistant as add in it seem to be stable now and reconnect works

I am using already the internal mqtt broker of homeassistant and there has been no change since month.