emsesp/EMS-ESP32 #547: Make Change Value dialog box left aligned for long text

tp1de, 2022-06-07:
Dialog box opens right aligned (end of text) for strings longer then dialog box. 
Left aligned (start of text( would be better.

tp1de, 2022-06-07:
Even better would be a multi line input within dialog box. 
(accepting "\n" line feeds within string or any other LF character.

proddy, 2022-06-11:
can you provide a screenshot and an example, so I can add test data

tp1de, 2022-06-11:
see #275 last entry.

I would like to use for hm1-hm5 e.g. 01.01.2023-01.02.2023  : {3,,2,[hc1,hc2,dhw1]}
and make a second line where the ":"  is ...

MichaelDvP, 2022-06-13:
The input line **is** left aligned, but it's an input and with focus on it, it shows the cursor position. If the focus changesto another element, it's left aligned again.
You can insert a line `multiline` [here after this line 538](https://github.com/emsesp/EMS-ESP32/blob/a02d4ab25983da88dbdb0000ed25f509351ceb60/interface/src/project/DashboardData.tsx#L538), than the input expands for long texts like this:

[image]     [image]

proddy, 2022-06-25:
I was buy re-doing the tables so added this multline, it'll be in the 3.4.2b3 which I'll push soon to dev

closed 2022-06-25 by proddy

MichaelDvP, 2022-06-25:
Please use `multiline={deviceValue.u ? false : true}` to keep the step-up/down-arrows for number fields.

tp1de, 2022-06-26:
> Please use `multiline={deviceValue.u ? false : true}` to keep the step-up/down-arrows for number fields.

@MichaelDvP what does this mean for formatting the lines?

proddy, 2022-06-26:
it means you press enter and it'll accept multiline strings. You can test it using the online demo https://ems-esp.derbyshire.nl/ems-esp/dashboard/data by selecting the Boiler and clicking on 'next maintenance date'

tp1de, 2022-06-26:
Ok understood. I am just testing multiline and I ask myself if I can use "\n" linefeed to build a second line or if I had to fill each line with " " .

tp1de, 2022-06-26:
Filling with blanks in multiline works so far:

[image]

tp1de, 2022-06-26:
Multiline is fine for me  ... thanks for adjusting

commit 30da267d (2022-06-25): fix #547