emsesp/EMS-ESP32 #548: Warm water problem - Junkers CERAPURMODUL + FW200 + IPM2

km4lin, 2022-06-10:
I've been using EMS-ESP (ESP8266) for 2 years and it worked fine, but it is not developed anymore.
For warm water I've been using script:
```
repeat:
  count: '60'
  sequence:
    - service: mqtt.publish
      data:
        topic: ems-esp/boiler
        payload: '{"cmd":"wwtemp","data":48}'
    - delay: '00:01:00'
```
Yesterday I've been updated ESP8266 to ESP32 thanks to article: https://bbqkees-electronics.nl/wiki/gateway/upgrade-the-gateway-to-an-esp32.html
And now I'm struggling with getting warm water activation.
Please help - how can I get warm water activated via telnet/mqtt?
On FW200 there is the button for warm water activation, is there some option to activate it via telnet/mqtt?
I'll be very grateful for any help.
Regards

MichaelDvP, 2022-06-10:
There are some renamings, this is now `wwseltemp`. Go to settings/customization and you can see the full-name and mqtt-name of each parameter. Or try `http://ems-esp.local/api/boiler/commands` or use terminal...

km4lin, 2022-06-10:
Thanks, it working now!

MichaelDvP, 2022-06-10:
BTW: Reason for renaming was sync of commands and value names. V2 was valuename `wwseltemp`, but command `wwtemp`.
With v3-mqtt you can use alternativ single value payload:
topic `emsesp/boiler/wwseltemp`, payload: `48`

closed 2023-02-19 by proddy