emsesp/EMS-ESP32 #550: hc1 summer mode shows always winter

pswid, 2022-06-13:
In my case (Buderus WSW196iT) `hc1 summer mode` always shows `winter`, regardless of current hp working mode (now it's summer mode of course).
Furthermore I noticed that `hc1 set summer mode` with options `summer`, `auto`, `winter` in [v3.4.1](https://github.com/emsesp/EMS-ESP32/releases/tag/v3.4.1) has been replaced by `hc1 heatpump operating mode` with options: `off`, `auto`, `heating`, `cooling`. Unfortunately my hp has no cooling functionality, so the previous was better (in my case). BTW, setting `cooling` switches mode to `auto`.
If there is no way to detect if hp has cooling functionality or not, maybe you could simply enable both `hc1 heatpump operating mode` and  `hc1 set summer mode`, so the user could disable irrelevant one by customization? What do you think?

MichaelDvP, 2022-06-13:
The `summer mode` has not changed since long time, i think it was not working on your heatpump all the time. Seems that heatpumps reports the state in another telegram/offset, but i dont know where. You can try to log while the mode changes, it should be in a broadcasted telegram.

The change to `operating mode` was introduced in #503, this is a special heatpump telegram 0x471, `summersetmode` is from telegram 0x2AF and not shown if telegram 0x471 exists.
Please read telegrams 0x2AF and 0x471.

pswid, 2022-06-13:
Yes, in previous EMS-ESP versions `summer mode` was not working as well. If I have some time, I'll try to find out telegram/offset with current summer mode status.

While changing `operating mode` (via KM200) telegrams 0x471 are:

change to forced (=winter/heating):
```
000+00:53:54.884 N 110: [emsesp] Gateway(0x48) -> Thermostat(0x10), RC300Summer2(0x0471), data: 02
000+00:53:57.313 N 111: [emsesp] Thermostat(0x10) -> All(0x00), RC300Summer2(0x0471), data: 02
000+00:54:04.213 N 112: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Summer2(0x0471), length: 0x19
000+00:54:04.247 N 113: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer2(0x0471), data: 02 0D 01 01 01 23 08 04
```

change to automatic:
```
000+00:54:14.516 N 114: [emsesp] Gateway(0x48) -> Thermostat(0x10), RC300Summer2(0x0471), data: 01
000+00:54:14.898 N 115: [emsesp] Thermostat(0x10) -> All(0x00), RC300Summer2(0x0471), data: 01
000+00:55:04.239 N 116: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Summer2(0x0471), length: 0x19
000+00:55:04.273 N 117: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer2(0x0471), data: 01 0D 01 01 01 23 08 04
```

change to off (=summer, doesn't mean cooling!)
```
000+00:55:14.922 N 118: [emsesp] Gateway(0x48) -> Thermostat(0x10), RC300Summer2(0x0471), data: 00
000+00:55:17.547 N 119: [emsesp] Thermostat(0x10) -> All(0x00), RC300Summer2(0x0471), data: 00
000+00:56:04.290 N 120: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Summer2(0x0471), length: 0x19
000+00:56:04.324 N 121: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer2(0x0471), data: 00 0D 01 01 01 23 08 04
```

telegram 0x2AF doesn't change while changing `operating mode `and is:
```000+01:08:02.566 N 145: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x02AF), data: 00 28 00 00 31 27 00 00 19 1E 05```

MichaelDvP, 2022-06-14:
Yes, that's the same as in #503, the `summersetmode` in 0x02AF stays zero. If we want to skip the cooling option we have to find out if there is a configuration setting about heatpump cooling. 

For the `summermode` flag this is reported in monitor-message 0x02A5 as single bit (bit 4 offset 2). I think the heatpumps will use a different position or telegram to show in auto-mode the states heating/off/cooling. 
If you switch the operatingmode from off to heating and log all, is there something changing in a broadcasted telegram (thermostat->all)? Maybe in a different position in 0x2A5 or an extra telegram.

pswid, 2022-06-21:
it seems that:
`Thermostat(0x10) -> All(0x00), RC300Monitor(0x02A5), data: 01 (offset 20)` means winter (or heating_on?)
and
`Thermostat(0x10) -> All(0x00), RC300Monitor(0x02A5), data: 02 (offset 20)` means summer (or heating_off or cooling?)

pswid, 2022-06-22:
Example 0x02A5 telegrams from my heatpump (with **no cooling support**) are:

for **heating on** (=forced):
```
RC300Monitor(0x02A5), data: 01 (offset 20)
RC300Monitor(0x02A5), data: 80 00 01 0A 09 00 0A 2A 05 A0 02 01 03 05 A0 05 A0 00 00 11 01 02 FF FF 00
RC300Monitor(0x02A5), data: 01 04 00 00 00 00 FF 64 2D 00 3C 01 FF (offset 25)
```

for **off**:
```
RC300Monitor(0x02A5), data: 02 (offset 20)
RC300Monitor(0x02A5), data: 80 00 41 0A 00 00 0A 2A 05 A0 02 01 03 05 A0 05 A0 00 00 11 02 02 FF FF 00
RC300Monitor(0x02A5), data: 01 08 00 00 00 00 FF 00 2D 00 3C 01 FF (offset 25)
```

If some owners of the hp **with** the cooling function could couch such telegrams for heating, cooling and off, maybe it would be possible to find a way to detect whether the hp supports cooling or not and info about current summer/winter mode.

MichaelDvP, 2022-06-22:
Can you check with operation mode auto and setting the summertemp below and above the current (damped) outdoortemp. Than we should see the summer-mode-status changing.
In offset 2 there is bit 6 (0x40) changing. Also offset 26 changes from 0x04 to 0x08 and offset 32 seems to be a percent value (pump?).  Boiler summer modes are in offset 2, bit 4, i think for heatpump the bits 6/7 are good candidates.

pswid, 2022-06-22:
automatic, heating on (summer temp set to 30°C)
```
RC300Monitor(0x02A5), data: 80 00 01 0A 00 00 0A 2A 05 A0 02 01 03 05 A0 05 A0 00 00 11 01 02 FF FF 00
RC300Monitor(0x02A5), data: 01 07 00 00 00 00 FF 00 2D 00 3C 01 FF (offset 25)
```

automatic, heating off (summer temp  set to 5°C)
```
RC300Monitor(0x02A5), data: 80 00 41 0A 00 00 0A 2A 05 A0 02 01 03 05 A0 05 A0 00 00 11 02 02 FF FF 00
RC300Monitor(0x02A5), data: 01 08 00 00 00 00 FF 00 2D 00 3C 01 FF (offset 25)
```

BTW, I've noticed that `hc1 summer temperature`  has not defined range (HA allows 0-255) and step is 0.5°C. My heat pump allows values 10-30°C and step is 1°C. It would be good to set range (10-30 seems to be good for all) and step...

MichaelDvP, 2022-06-22:
>  (summer temp set to 5°C)
>  My heat pump allows values 10-30°C 

One of these values seems to be a typo. 
In RC35 i have 9-25°C, but 9°C is "always summer", the RC35 does not have an extra operating mode. 

The summermode change is shown in offset2 bit 6 and offset 20. 

For the cooling we need a bit help: @tschamm Could you check with your heatpump the telegram 0x02A5 for heating/off/cooling and in auto mode when off/heating/cooling.

pswid, 2022-06-22:
Yes, my mistake. I tried to set 5°C via ems-esp dashboard and later realized that value was actually set to 10°C. 
Valid range I obtained directly from KM200 gateway (via proper REST call):
`{"id": "/heatingCircuits/hc1/suWiThreshold", "type": "floatValue", "writeable": 1, "recordable": 0, "value": 10.0, "unitOfMeasure": "C", "minValue ": 10.0," maxValue ": 30.0}`

MichaelDvP, 2022-06-25:
> BTW, offset 2, bit 0 is =1 in my case (no cooling support) and =0 for [tschamm](https://github.com/tschamm)'s hp (with cooling capabilty). Could we use it to determine if hp supports cooling? It would allow us to decide if to display hp operating mode with options: off, auto, heating, cooling or just set summer mode with options summer, auto, winter?

Normally monitor telegrams only show states, not configurations. Norbert lists this bit as "heating possible", but this makes no sense in this case, tschamms hp also set this bit while heating. Maybe it is "only heating possible", than we can use it the way you mentioned. I'll make a test buld with this logic, for checking: customizations will show summermode/summersetmode and hpoperatingmode/hpoperatingstate, but only one pair should have values. Please [test this](https://github.com/MichaelDvP/EMS-ESP32/releases).

pswid, 2022-06-26:
I've installed your version. `summer mode` now shows `winter`/`summer`. Unfortunately `set summer mode` (with options `summer`, `auto`, `winter`) in unavailable now.

[image]

tschamm, 2022-06-26:
I've installed it too. For my system, summer mode is not shown anymore (ok for me).
[image]

> BTW, offset 2, bit 0 is =1 in my case (no cooling support) and =0 for [tschamm](https://github.com/tschamm)'s hp (with cooling capabilty). Could we use it to determine if hp supports cooling?
Could it be that offset 2, bit 0 shows if heating is allowed currently? For my system, outdoor temp is too high currently, so that heating is disabled.

MichaelDvP, 2022-06-27:
@pswid Sorry i had a typo in summersetmode, summersetmode should be shown now, please try.

MichaelDvP, 2022-06-29:
@pswid Have you tried after the latest fix? Write command should work now.

pswid, 2022-06-30:
Yes, now it works fine.

commit 422657fb (2022-06-30): fix #550

typo summersetmode/summermode
commit 70fe08a3 (2022-06-30): fix #550 set_summermode
commit efa5eb27 (2022-06-30): show summermode or operatingmode, fix #550
closed 2022-07-09 by proddy