emsesp/EMS-ESP32 #553: boiler burner selected max power range incorrect

bluediamond1984, 2022-06-18:
I've updated firmware to the latest stable 3.4.1. And I like the new added features.
However in home assistant log I noticed errors while using the hot water.

ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 143 (range 0.0 - 130.0)

It seems that the upper limit is to low for my Nefit Topline/GB162.

proddy, 2022-06-19:
@MichaelDvP make it 150?

MichaelDvP, 2022-06-19:
Set it to 254 % for selBurnPow_, burnMaxPower_, and wwMaxPower_. For UINT this will always fit.
I'm not sure if the Bosch percentage should be renamed to `"max power (P.M.P.O)" ` or if this a simple trick to use the same burner for different power classes: 100kW boiler is 0-100%, 80kW boiler is 0-125% and 50kW is 0-200% power modulated with same burner.

commit 0e563e91 (2022-06-21): burner max power to 254%, fix #553
commit 1fab47ac (2022-06-21): Merge pull request #558 from MichaelDvP/dev

burner max power to 254%, fix #553
closed 2022-06-21 by proddy