emsesp/EMS-ESP32 #554: Vacations for warm water

OHacks, 2022-06-19:
I am using ems-esp32 EMS-ESP version 3.3.1  with Buderus GB142 and RC30:

I can set the vacation time for hc1 with ems-esp/thermostat/hc1/vacations "01.10.2022-10.10.2022".
If I do so the heating of the water is not turned off.

On RC30 I can manually set up the vacation time ("Urlaub" in german) for "Heizkreis1 " = HC1, "Warmwasser" = warm water and "Gesamtanlage" = complete system.

When I change the vacation for HC1 manual on the RC30 I can see the new values in MQTT as hc1 : vacations
When I change the vacation for warm-water manual on the RC30 I can not see the values in MQTT at all.

I also don't see a way to change the vacations for the warm-water via MQTT/emsesp.

Please add a way to modiy and read out the vacations for warm-water!

proddy, 2022-06-19:
@tp1de any thoughts on how best to address this?

OHacks, 2022-06-20:
> You need to have telegram examples

If you need any telegram examples please simply explain how I can generate them and I will do that.

MichaelDvP, 2022-06-20:
@OHacks Is your RC30 product id 67 or 78? I have RC35 which is mostly similar to RC30 product-id 67. Please post the respnse to `read 10 38` and `read 10 39`.

The RC35 have timerprograms, holiday/vacation, pause/party for each circuit:
dhw 0x38, circuation 0x39, hc1 0x3F, hc2 0x49, hc3 0x53, hc4 0x5D

For now we have in heatingcircuits values for holiday-dates (at home), vacation-dates (away from home), pause and party. 
I can add the same for dhw and circulation, but this are a lot of new entities. Maybe better to add only dhw holiday/vacation and send a new date to dhw and circulation, circulation without dhw preparation makes no sense.

Or we try to combine the values like the RC300 structure and use a single string with something like
`startDate-endDate-mode-assigns:(dhw,circ,hc1,hc2,hc3,hc4)` 
but this does not allow to set different dates to different hcs. And you can't set different dates for vacation and holiday.

Any suggestions?

OHacks, 2022-06-20:
```
ems-esp:$ read 10 38
004+15:02:25.136 N 0: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x38), data: 01 1E 00 8D 21 1E 20 8D 41 1E 40 8D 61 1E 60 8D 81 1E 80 8D A1 22 A0 8E C1 22 C0
004+15:02:25.640 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x38), data: 8E E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 27)
004+15:02:26.015 N 2: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x38), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 54)
004+15:02:26.369 N 3: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x38), data: 90 E7 90 00 00 00 16 06 16 17 06 16 01 01 00 01 01 00 (offset 81)
```

```
ems-esp:$ read 10 39
004+15:03:08.137 N 4: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x39), data: 01 21 00 84 21 21 20 84 41 21 40 84 61 21 60 84 81 21 80 8A A1 27 A0 8D C1 2A C0
004+15:03:08.891 N 5: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x39), data: 84 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 27)
004+15:03:09.766 N 6: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x39), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 54)
004+15:03:10.366 N 7: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x39), data: 90 E7 90 01 00 00 16 06 16 17 06 16 01 01 00 01 01 00 (offset 81)
```
The RC30 / GB142 allows to set different vacation times for HC1 and warm water. If someone does this manually on the RC30 then emsesp should be able to at least read out both vacation times. So combining hcx and warm water (dhw?) will not be so good.

OHacks, 2022-06-20:
> Is your RC30 product id 67 or 78?

Do you need an addition info or does the output of the read commands already tell you the product id? If not, how can I find out the id?

MichaelDvP, 2022-06-20:
> If not, how can I find out the id?

Best on ems-esp-help-page click the _"[download]() debug information"_  (always good to know and should attached to each new issue)

The telegrams are the same as RC35, You have set for dhw and circualation a vacation 22.06.2022-23.06.2022, pause/party is zero (cleared) and holiday is not set (01.01.2000-01.01.2000).

`00 00 16 06 16 17 06 16 01 01 00 01 01 00` (pause, party, vac::d.m.y-d.m.y, holiday:d.m.y-d.m.y

OHacks, 2022-06-20:
The vacation time is correct. circulation is disabled. The manual  tells that dhw and circulation are set by setting "warm water" vacation time.

MichaelDvP, 2022-06-20:
I've tested the dhw pause/party functions, they are writeable, but have no effect, we can skip.
So i'll add only `dhw vacations` and `dhw holiday`, reading the dates from 0x38 and writing to dhw and circulation.

OHacks, 2022-06-20:
I uploaded firmware EMS-ESP-3_4_1_patch1-ESP32.bin, but I do not see any vacation/holiday entries for dhw:

> Entity;Value;UoM
> last error code;A0(800) 18.03.2022 14:59 (7 min);
> date/time;20.06.2022 18:49;
> display;ww temperature;
> language;german;
> clock offset;1;second
> internal temperature offset;0;°C
> minimal external temperature;-14;°C
> damped outdoor temperature;18;°C
> building type;medium;
> dhw mode;auto;
> dhw circulation pump mode;auto;
> dhw circulation program;std_prog;
> dhw disinfecting;off;
> dhw disinfection day;all;
> dhw disinfection hour;16;
> dhw maximum temperature;62;°C
> dhw program switchtime;00 mo 05:00 on;
> dhw circulation program switchtime;00 mo 05:30 on;
> hc1 selected room temperature;0;°C
> hc1 mode;auto;
> hc1 mode type;day;
> hc1 day temperature;22,5;°C
> hc1 night temperature;18;°C
> hc1 design temperature;38;°C
> hc1 offset temperature;0;°C
> hc1 holiday temperature;15;°C
> hc1 target flow temperature;5;°C
> hc1 summer temperature;9;°C
> hc1 summer mode;summer;
> hc1 holiday mode;off;
> hc1 nofrost temperature;6;°C
> hc1 nofrost mode;room;
> hc1 room influence;3;°C
> hc1 min flow temperature;5;°C
> hc1 max flow temperature;42;°C
> hc1 flow temperature offset for mixer;5;°C
> hc1 heating type;radiator;
> hc1 reduce mode;outdoor;
> hc1 control device;off;
> hc1 holiday dates;01.01.2000-01.01.2000;
> hc1 vacation dates;20.06.2022-20.06.2022;
> hc1 program;own_1;
> hc1 pause time;0;hours
> hc1 party time;0;hours
> hc1 dhw priority;on;
> hc1 own1 program switchtime;00 mo 04:00 on;

MichaelDvP, 2022-06-20:
> I uploaded firmware EMS-ESP-3_4_1_patch1-ESP32.bin

The fix is only in the dev-build 3.4.2b2, look at the included commits. Additions always go only to the dev.

OHacks, 2022-06-20:
Sorry, I did not see firmware EMS-ESP-3_4_2b2-ESP32.bin.
I just upladed it:

**Some parts work:**
dhw vacation dates and holiday dates work in both way:
- when I change dhw vacation dates with emsesp then it is changed in RC30 warm water vacations
- when I change warm water vacations in RC30 the then the changes are visible in emsesp dhw vacation dates
- when I change dhw holiday dates with emsesp then it is changed in RC30 warm water holidays
- when I change dhw holiday dates in RC30 the then the changes are visible in emsesp dhw holiday dates

**BUT the follwing does not work correctly:**
- when I change hc1 vacation dates with emsesp then it is changed in RC30 warm water vacations and this is also read back in emsesp dhw vacation dates
- when I change hc1 holiday dates with emsesp then it is changed in RC30 warm water holidays and this is also read back in emsesp dhw holiday dates

In MQTT I see hc1:vacations, hc1:holidays and holidays and vacations. **The warm water holdays and vacations may be should be renamed to wwholidays and wwvacations.**

> {"lastcode":"A0(800) 18.03.2022 14:59 (7 min)","datetime":"20.06.2022 19:17","display":"ww temperature","language":"german","clockoffset":1,"intoffset":0,"minexttemp":-14,"dampedoutdoortemp":18,"building":"medium","wwmode":"auto","wwcircmode":"auto","wwcircprog":"std_prog","wwdisinfecting":"off","wwdisinfectday":"all","wwdisinfecthour":16,"wwmaxtemp":62,"wwswitchtime":"00 mo 05:00 on","wwcircswitchtime":"00 mo 05:30 on","holidays":"21.06.2022-21.06.2022","vacations":"20.06.2022-20.06.2022","hc1":{"seltemp":0,"mode":"auto","modetype":"day","daytemp":22.5,"nighttemp":18,"designtemp":38,"offsettemp":0,"holidaytemp":15,"targetflowtemp":5,"summertemp":9,"summermode":"summer","holidaymode":"off","nofrosttemp":6,"nofrostmode":"room","roominfluence":3,"minflowtemp":5,"maxflowtemp":42,"flowtempoffset":5,"heatingtype":"radiator","reducemode":"outdoor","control":"off","holidays":"21.06.2022-21.06.2022","vacations":"20.06.2022-20.06.2022","program":"own_1","pause":0,"party":0,"wwprio":"on","switchtime1":"00 mo 04:00 on"}}

MichaelDvP, 2022-06-20:
Ups, thanks for testing, i've only tested with dhw vacations. I've forgot to rename to wwvacation. Try again with new build.

OHacks, 2022-06-20:
Everything works fine now!

I changed hc1 vacation dates, hc1 holiday dates, dhw vacation dates and dhw holiday dates on emsesp and verified them on RC30.
I also changed all 4 values on RC30 and verified them on emsesp! 
Finally I changed wwholidays, wwvacations, hc1/holidays and hc1/vacations via MQTT and verified the values with emsesp output.

Really a cool project!
Thanks a lot - great job and an incredible fast solution!

commit 0e5cb2df (2022-06-20): fix #554, add dhw vacation/holiday
commit 8790d81e (2022-06-20): Merge pull request #556 from MichaelDvP/dev

fix #554, add dhw vacation/holiday
closed 2022-06-21 by OHacks