emsesp/EMS-ESP32 #555: Shorten "friendly names" in Home Assistant

pswid, 2022-06-20:
Curently each entity in HomeAssiatnt has a "friendly name" that includes prefix **System**, **Boiler** or **Thermostat**. This makes often such names too long to be fully displayed at the HA dashborad. So, name of each entity must be manually edited (shortened) in HA. Could you add an option that will enable/dissable device name (Boiler, Thermostat, etc.) in "frindly names" visible in Home Assitant or even disable it permanently, please? Entity IDs should remain unchanged (I mean still contain prefix Boiler, Thermostat, etc.).

[image]

pswid, 2022-06-21:
How to permanently disable entity prefixes in HA (boiler, thermostat etc) via customization page? I don't see such an option there.

Checkbox to disable/enable prefixes globally (for all entities) would be OK, although I personally think that checkbox is not even needed, because there should be no prefixes at all. 
Code modifiaction to shorten HA names should be rather easy. It would only require a slight modification of the HA mqtt discovery config messages:
1. change current `name` to `object_id`
2. add `name` in short verson (I mean without prefix)

e.g. change from:
```
{
  "~": "ems-esp",
  "uniq_id": "thermostat_floordry",
  "stat_t": "~/thermostat_data",
  "name": "Thermostat floor drying",
  "val_tpl": "{{value_json.floordry}}",
  "dev": {
    "ids": [
      "ems-esp-thermostat"
    ]
  }
}
```

to:

```
{
  "~": "ems-esp",
  "uniq_id": "thermostat_floordry",
  "stat_t": "~/thermostat_data",
  "object_id": "Thermostat floor drying",
  "name": "Floor drying",
  "val_tpl": "{{value_json.floordry}}",
  "dev": {
    "ids": [
      "ems-esp-thermostat"
    ]
  }
}
```

proddy, 2022-06-21:
I'll welcome a PR !

pswid, 2022-06-21:
I will try

proddy, 2022-06-24:
in b3, nice work!

closed 2022-06-24 by proddy