emsesp/EMS-ESP32 #559: Regular message: Last Tx Read operation failed after 3 retries

odelcourt, 2022-06-21:
I have no problem with my config (or I don't notice it) but in the Telnet console, I can see that kind of messages (low frequency):

000+06:16:48.939 W 0: [telegram] Incomplete Rx: 90 00 06 00 16 06 11
000+12:09:03.913 E 1: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 19 00 6F
001+05:50:33.788 W 2: [telegram] Incomplete Rx: 90 00 06 00 16 06 11
002+04:46:26.763 E 3: [telegram] Last Tx Write operation failed after 3 retries. Ignoring request: 8B 10 FF 04 00 79 01
002+20:54:13.355 E 4: [telegram] Last Tx Write operation failed after 3 retries. Ignoring request: 8B 10 FF 04 00 79 01
005+08:07:48.099 W 5: [telegram] Incomplete Rx: 90 88 00 18 00 00 02 16 64 00 00 12 00 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
006+20:57:04.700 E 6: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 19 00 79
007+18:36:13.762 E 7: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 19 00 6F
007+23:04:15.426 E 8: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 19 00 6F
009+17:36:17.017 E 9: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 19 00 79
010+14:57:18.684 E 10: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 19 00 6F

How can I help to debug this ? Is it something that could be improved on my system ?

[emsesp_info.json.txt](https://github.com/emsesp/EMS-ESP32/files/8947204/emsesp_info.json.txt)

proddy, 2022-06-21:
misses are normal. You've been running it for 10 days I see and only 10 Tx errors from 80000 requests which is impressive. I wouldn't worry about it.

odelcourt, 2022-06-21:
> misses are normal. You've been running it for 10 days I see and only 10 Tx errors from 80000 requests which is impressive. I wouldn't worry about it.

OK, perfect then :-) Thx so much

closed 2023-02-19 by proddy