emsesp/EMS-ESP32 #564: Allow JSON for API - different formats for dashboard, MQTT and API

tp1de, 2022-06-27:
The data content for holiday-modes and/or switch programs can be quite complex and can contain quite an number of datapoints and parameters. E.g. switch program for EMS+ (RC310) might contain up to 6 switch points per day - a maximum of 42 switch points per week. Each switch point is associated with a temp parameter (eco / comfort / temp).

The RC310 has 2 switch programs for each hc and one for ww and one for circulation pump.
For my quite standard heating system with 2 hc's and ww in total 6 switch programs with up to 42 switch points.
A bit similar but less complex are the holiday modes hm1 to hm5. 

I believe it makes sense to add JSON for the API-interface to cover this complex data structures within one JSON-structure similar to the km200 web-API interface. In this case it does not make sense to include this into the dashboard and as far as I understood HA MQTT can't process this either. 

Let me recommend to differentiate between dashboard / HA-MQTT and API in this case. 
Either different data-content / data-structure or just API-content not shown in dashboard.

proddy, 2022-06-27:
I'd prefer to create a special feature from the web where you can upload a JSON file for programs, rather than changing the API to support this. And the API already takes JSON as input, just not in the old km200 format

tp1de, 2022-06-28:
> I'd prefer to create a special feature from the web where you can upload a JSON file for programs, rather than changing the API to support this

I do not understands what this means?
My proposal was to register the telegram and then not creating individual variables but rather one JSON-structure for the whole telegram content.
E.g. switchProgram:

_[ { "dayOfWeek": "Mo", "setpoint": "comfort2", "time": 300},
  { "dayOfWeek": "Mo", "setpoint": "eco", "time": 1320},
  { "dayOfWeek": "Tu",  "setpoint": "comfort2", "time": 300 },
  ...... ]_

I do have the coding to extract this out the respective telegrams identical to km200 JSON-structure or any other JSON you like. 

Within API-call read/write it should show as: ems-esp/api/thermostat:

_{ .....
"hc1": {
    "seltemp": 22,
    "currtemp": 22.8,
    "mode": 1,
    "switchProgram_A": 
    [ { "dayOfWeek": "Mo", "setpoint": "comfort2", "time": 300},
  { "dayOfWeek": "Mo", "setpoint": "eco", "time": 1320},
  { "dayOfWeek": "Tu",  "setpoint": "comfort2", "time": 300 },
  ...... ],
"switchProgram_B": 
[ { "dayOfWeek": "Mo", "setpoint": "comfort2", "time": 300},
  { "dayOfWeek": "Mo", "setpoint": "eco", "time": 1320},
  { "dayOfWeek": "Tu",  "setpoint": "comfort2", "time": 300 },
  ...... ]_

... and swithProgram_A and switchProgram_B should be writable.

Within the actual ems-esp coding logic I need to create Individual variables as far as I understood. Within the existing Change Value dialog box this does not work either.

proddy, 2022-07-09:
yes, it's too complicated and will blow up the ESP32 this way. As I mentioed in an earlier comment https://github.com/emsesp/EMS-ESP32/issues/564#issuecomment-1167847703 for specific use cases like this the best forward is to create a new feature with a file import via the web UI.