emsesp/EMS-ESP32 #565: The normalized unit of sensor.boiler_burner_starts (None) does not match the unit of already compiled statistics (times)

swa72, 2022-06-28:
**Bug description**
 
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:492 
Integration: Sensor (documentation, issues) 
First occurred: 26 June 2022, 18:00:10 (1 occurrences) 
Last logged: 26 June 2022, 18:00:10

The normalized unit of sensor.boiler_burner_starts (None) does not match the unit of already compiled statistics (times). Generation of long term statistics will be suppressed unless the unit changes back to times. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this

**Steps to reproduce**

None

**Expected behavior**

No log entry 

**Screenshots**
N/A

**Device information**
Oddly, that URL gives me …
```
{
  "message": "unknown device"
}
```

**Additional context**

Upgraded from 3.3.1 to 3.4.1, I‘m on HA 2022.6

proddy, 2022-06-28:
This is because the state_class of some of the entities was changed earlier this year and HA can no longer collect the historical statistics for analysis.

e.g. `homeassistant/sensor/ems-esp/boiler_burnstarts/config` shows
```json
{"~":"ems-esp","uniq_id":"boiler_burnstarts","stat_t":"~/boiler_data","object_id":"boiler_burner starts","name":"Burner starts","val_tpl":"{{value_json.burnstarts}}","ic":"mdi:counter","state_class":"total_increasing","dev":{"ids":["ems-esp-boiler"]}}
```

I get the same warnings too. They are harmless and can be ignored unless you really want to use HA to graph how many times the boiler is restarted.

The solution is to reset the statistics in HA. I don't know how, and to be honest, didn't really look into it. If you find a way please post it here and I'll add to the Wiki.

There is nothing EMS-ESP can do here I'm afraid.

swa72, 2022-07-03:
Hi @proddy and thanks for the update. I'll ignore it as well ;-) 

It may be a related subject: since the upgrade I've lost the following sensors 
```
binary_sensor.solar_external_tank (Solar external tank)
binary_sensor.solar_tank_heated (Solar tank heated)
binary_sensor.thermostat_hc1_summer_mode (Thermostat hc1 summer mode)
binary_sensor.thermostat_hc2_summer_mode (Thermostat hc2 summer mode)
binary_sensor.thermostat_hc3_summer_mode (Thermostat hc3 summer mode)
binary_sensor.thermostat_ww_disinfection (Thermostat ww disinfection)
```
No big deal as I can ignore them as well. 

Then I read https://emsesp.github.io/docs/#/Troubleshooting?id=ha-is-showing-errors-like-quotreceived-message-on-illegal-discovery-topicquot-or-quottemplate-variable-warning-39dict-object39-has-no-attributequot and disabled "MQTT Discovery".
Waited 5min and enabled it again. Everything back to normal :-)