emsesp/EMS-ESP32 #570: Uploading of saved customisation

harvey637, 2022-07-11:
Hi, just a question:
I save settings and customizations, all in 3.4.2b3, ok.  Content looks like expected:
```
{
  "type": "customizations",
  "Customizations": {
    "sensors": [],
    "analogs": [],
    "masked_entities": [
      {
        "product_id": 172,
        "device_id": 8,
        "entity_ids": [
          "03heatingactive",
          "03tapwateractive",
          "03selflowtemp",
          "03selburnpow",
          "03heatingpumpmod",
          "03heatingpump2mod",
          "03switchtemp",
          "03burngas",
          "03burngas2",
          "03flamecurr",
          "03burnminperiod",
          "03burnminpower",
          "03burnmaxpower",
          "03setburnpow",
          "03curburnpow",
          "03burnstarts",
...
          "03hc1/switchonoptimization",
          "03hc1/reducemode",
          "03hc1/noreducetemp",
          "03hc1/reducetemp",
          "03hc1/wwprio"
        ]
      }
    ]
  }
}
```

After a restart ( I had some rtc-watchdog with no other info) I try to upload, select file, upload, then automatic restarting ems-esp32.

But NO customization is reloaded, I have to do so manually and compare the saved settings until satisfied.

This is not the intended behaviour, right? Is it personel carma are something else?
cu
Harvey

MichaelDvP, 2022-07-11:
@harvey637 iirc you use a lot of excludes. I think it's a size effect because it works for me with some excludes, With much more excludes the saved file is nearly empty and near 4k limit i can reproduce your issue. 
Could you please attach here the complete customization file. 

If you look at weblog, could you see a log message `[system] Applying new Customizations settings` as one of the first messages? Or is it `[system] Unrecognized file uploaded, not json`

harvey637, 2022-07-11:
Hi Michael,
yes size came into my mind also, I have to save my customization in chunks. Sorry for this problem, but I have NO WarmWater, NO Pool, No Burner, only Heatpump :-)

complete customization:

```
{
  "type": "customizations",
  "Customizations": {
    "sensors": [],
    "analogs": [],
    "masked_entities": [
      {
        "product_id": 172,
        "device_id": 8,
        "entity_ids": [
          "03heatingactive",
          "03tapwateractive",
          "03selflowtemp",
          "03selburnpow",
          "03heatingpumpmod",
          "03heatingpump2mod",
          "03switchtemp",
          "03burngas",
          "03burngas2",
          "03flamecurr",
          "03burnminperiod",
          "03burnminpower",
          "03burnmaxpower",
          "03setburnpow",
          "03curburnpow",
          "03burnstarts",
          "03burnworkmin",
          "03burn2workmin",
          "03uptimecompww",
          "03uptimecomppool",
          "03wwstarts2",
          "03poolstarts",
          "03nrgconscompww",
          "03nrgconscomppool",
          "03auxelecheatnrgconsww",
          "03auxelecheatnrgconspool",
          "03nrgsuppww",
          "03nrgsupppool",
          "03hpwwon",
          "03hppoolon",
          "03poolsettemp",
          "03wwtapactivated",
          "03wwsettemp",
          "03wwseltemp",
          "03wwseltemplow",
          "03wwseltempoff",
          "03wwseltempsingle",
          "03wwtype",
          "03wwcomfort",
          "03wwcomfort1",
          "03wwflowtempoffset",
          "03wwchargeoptimization",
          "03wwmaxpower",
          "03wwmaxtemp",
          "03wwcircpump",
          "03wwchargetype",
          "03wwhyston",
          "03wwhystoff",
          "03wwdisinfectiontemp",
          "03wwcircmode",
          "03wwcirc",
          "03wwcurtemp",
          "03wwcurtemp2",
          "03wwcurflow",
          "03wwstoragetemp1",
          "03wwstoragetemp2",
          "03wwactivated",
          "03wwonetime",
          "03wwdisinfecting",
          "03wwcharging",
          "03wwrecharging",
          "03wwtempok",
          "03wwactive",
          "03ww3wayvalve",
          "03wwsetpumppower",
          "03wwmixertemp",
          "03wwcylmiddletemp",
          "03wwstarts",
          "03wwworkm"
        ]
      },
      {
        "product_id": 200,
        "device_id": 56,
        "entity_ids": [
          "08airhumidity",
          "08dewtemperature"
        ]
      },
      {
        "product_id": 158,
        "device_id": 16,
        "entity_ids": [
          "02lastcode",
          "03floordry",
          "03floordrytemp",
          "03building",
          "03minexttemp",
          "03wwsettemp",
          "03wwmode",
          "03wwsettemplow",
          "03wwcircmode",
          "03wwchargeduration",
          "03wwcharge",
          "03wwextra1",
          "03wwextra2",
          "03wwdisinfecting",
          "03wwdisinfectday",
          "03wwdisinfecttime",
          "08hc1/seltemp",
          "08hc1/currtemp",
          "02hc1/HA climate config creation",
          "02hc1/manualtemp",
          "02hc1/comforttemp",
          "02hc1/summertemp",
          "02hc1/designtemp",
          "02hc1/offsettemp",
          "03hc1/minflowtemp",
          "03hc1/maxflowtemp",
          "03hc1/roominfluence",
          "03hc1/roominflfactor",
          "03hc1/curroominfl",
          "03hc1/nofrostmode",
          "02hc1/nofrosttemp",
          "03hc1/targetflowtemp",
          "03hc1/heatingtype",
          "03hc1/fastheatup",
          "03hc1/switchonoptimization",
          "03hc1/reducemode",
          "03hc1/noreducetemp",
          "03hc1/reducetemp",
          "03hc1/wwprio"
        ]
      }
    ]
  }
}
```

harvey637, 2022-07-11:
as you anticipated:

```
2022-07-09 11:46:59.000    INFO 0:      [emsesp]     Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2022-07-09 11:46:59.000   ERROR 1:      [system]     Unrecognized file uploaded, not json
2022-07-09 11:46:59.000    INFO 2:      [emsesp]     Loading board profile CUSTOM
2022-07-09 11:46:59.000    INFO 3:      [emsesp]     Starting EMS-ESP version 3.4.2b3 (hostname: ems-esp)
2022-07-09 11:46:59.000    INFO 4:      [analogsensor] Starting Analog sensor service
2022-07-09 11:46:59.000    INFO 5:      [emsesp]     Loaded EMS device library (100 records)
2022-07-09 11:46:59.000    INFO 6:      [emsesp]     Starting Access Point with captive portal on 192.168.4.1
2022-07-09 11:47:01.193    INFO 7:      [emsesp]     Starting NTP
2022-07-09 11:47:01.197    INFO 8:      [emsesp]     WiFi connected with IP=192.168.23.95, hostname=ems-esp
2022-07-09 11:47:01.225    INFO 9:      [emsesp]     mDNS responder service started
2022-07-09 11:47:01.780    INFO 10:     [system]     NTP connected
2022-07-09 11:47:03.933    INFO 11:     [emsesp]     Recognized new thermostat with deviceID 0x10
2022-07-09 11:47:04.725    INFO 12:     [emsesp]     Recognized new boiler with deviceID 0x08
2022-07-09 11:47:05.575    INFO 13:     [emsesp]     Recognized new heatpump with deviceID 0x38
2022-07-09 11:47:06.339    INFO 14:     [emsesp]     Recognized new gateway with deviceID 0x48
2022-07-09 11:47:09.000    INFO 15:     [emsesp]     Stopping Access Point
2022-07-09 11:49:12.036 WARNING 16:     [telegram]   Incomplete Rx: 38 00 B8 00 48 00
2022-07-09 18:38:53.144 WARNING 17:     [telegram]   Incomplete Rx: 90 48 F9 00 FF 01 6C 08 4F 00 00 00 01 00 00 00 02 00 06 00 03 C0 06 00 02
```

MichaelDvP, 2022-07-11:
It's the 4k file/json limit, as workaround try to remove 3-4 settings from the customization file and upload should work.

proddy, 2022-07-11:
we could msgpack (compress) the json but then you wouldn't be able to edit it offline as the file is binary and unreadable

harvey637, 2022-07-12:
Hi, thank you for looking into this one.
It is a small problem to remove ~ 5 settings and add them manually after restart. But will it survive (watchdog) reboot?

One question is, why is the customization not reloaded automatically after any reboot? Is this the same size limit? 

Background:
I see sporadically (around 2 times in a month) a reboot caused by "rtc-watchdog". After the reboot ems-esp32 works normal, the api sends ALL data, I see around 150 entities in my dashboard. The mass of data is no real problem, but unneccessary and at least confusing in dashboard an the creation of ~ 100 iobroker datapoints.  This large number is my first hint that a reboot has been occured. So I try to upload customization file, not working, so I add my customization manually an save the file.
1) should the customization be reloaded to my former status automatically? (config stays correct even after watchdog reboot)
2) is this (internal reload of old customizations) the same length problem as the manual upload via the upload funtction?

Than the length problem should be remedied (with low prio).

Some other thought about reduction: save in two versions, compressed or uncompressed, upload identifies both.

But the keeping of customisazion after watchdog reset should work without length problems please.
cu
Harvey

MichaelDvP, 2022-07-12:
@harvey637 With 5 settings less it will survive the reboot. This is also the limit of `FS_BUFFER_SIZE`. 
@proddy I think we should enlarge the buffers and keep the settings readable.

MichaelDvP, 2022-07-16:
@harvey637 Have you tried the latest build? I think this should work until you you have doubled the excluded entitiies.

closed 2022-07-22 by proddy