emsesp/EMS-ESP32 #573: Add Buderus AM200 Support

OlliHein, 2022-07-16:
Requirement:
Im using GB172 together with SM200, RC310 and AM200. Currently AM200 is not available and cannot be detected. 

System:
EMS-ESP Version: v3.4.1
Device (Platform / SDK): ESP32 / v3.3.5-1-g85c43024c
Heap (Free / Max Alloc): 118.816 / 62.172 bytes 
Flash Chip (Size / Speed): 4.194.304 bytes / 40 MHz
File System (Used / Total): 20.480 / 327.680 bytes (307.200 bytes free)

Devices and Sensors:
Type;Description;;Entries
Boiler;GBx72/Trendline/Cerapur/Greenstar Si/27i;67
Thermostat;RC300/RC310/Moduline 3000/1010H/CW400/Sense II;34
Solar;SM200/MS200;42
Gateway;KM200/MB LAN 2;0
Unknown;unknown;0
Sensors;Attached EMS-ESP Sensors;0

As you can see there is an unknown device detected, should be AM200

Environment:
Im using picture 23 from Buderus "Installation instructions for contractors". If you need PDF, I can give you a attachment. But I think you already have it or can download it.
As you can see in picture 23 I am using following sensors, pumps and valves:
Sensors: TA1, TR1, TR2, TB4, TB1, TB3
Valves: VR1, VB1
Pump: PR1

I can give you all the data and values of the running system to better identify the devices.

At first I attached the current logfile of EMS-PSP32:
[log_AM200.txt](https://github.com/emsesp/EMS-ESP32/files/9126502/log_AM200.txt)

and also picture 23 from Buderus installation guide:
[image]
  
Hope that will you give a first overview of my running system.

It will be great if AM200 can be integrated in your system, thanks for dooing that ;)

Thanks
Oliver

MichaelDvP, 2022-07-17:
Please go to help-page and download the debug info (not settings/customizations), and post it here. We need the product-id and telegrams.
The log shows 3 telegrams from device 0x60.
0x054D seems to be Temperatures 
'''
00F5 00F7 8000 00D5 010F 01E5 8000 01DD 0000 8000
24.5 24.7      21.3 27.1 48.5      47.7
''' 
0x054E could be a status message, all valves off: `00 00 00 00 00 00`
0x0550 could be settings: 00 FF 00 FF FF 00 0D 00 01 00 00 00 00 01 03 01 00 03 00 2D 19 C8 02 94 00 00 FF FF

For the temperatures note the temperatures shown in thermostat monitor screen and in terminal type `read 60 54d`
For the status telegrams you can use disgnostic menu, switch valves and check if and how telegram 0x54e changes.
 
For settings you have to change every setting for AM200 on the thermostat and check what changes in telegram 0x550.

With all this info we can add values/settings to emsesp.

@proddy : What device type? It is a special case of additional heating device, or in pos 10, as boiler replacement. I think best to add it as boiler and check for device-id 0x60? Like 0x70... for cascaded boilers.

PS:
@proddy Could you make the debug info on help page with a large button? Nobody uses it. And maybe move the settings/customization-download to system->upload (rename tab to up/download) to avoid confusion on the help page?

proddy, 2022-07-17:
> @proddy : What device type? It is a special case of additional heating device, or in pos 10, as boiler replacement. I think best to add it as boiler and check for device-id 0x60? Like 0x70... for cascaded boilers.

Agree, let's add it to boiler

> PS: @proddy Could you make the debug info on help page with a large button? Nobody uses it. And maybe move the settings/customization-download to system->upload (rename tab to up/download) to avoid confusion on the help page?

Both good ideas. I'll try and think of a name for the up/download tab

OlliHein, 2022-07-17:
@MichaelDvP 
At first I attached debug info file. You are right, it is hard to find.
BTW: I also put it into a zip file because .json is not supported as an attachment here ;) 

[emsesp_info.zip](https://github.com/emsesp/EMS-ESP32/files/9128347/emsesp_info.zip)

Tomorrow I will investigate more time to find out values and status informations as you descibed. Thanks for such fast support ;)

MichaelDvP, 2022-07-18:
> BTW: I also put it into a zip file because .json is not supported as an attachment here ;)

Good point, we should rename to `emsesp_info.json.txt`.

OlliHein, 2022-07-18:
@MichaelDvP 
Here are the temperature definitions:
```
[telegram] Rx: 60 00 FF 00 04 4D 0103 0108 8000 00C6 0127 0205 8000 0200 0000 8000 6C
                                                    TB4   TR2            TA1    TR1   TB1   TB2*  TB3
```

There are two 0x8000 I do not have connected. Should be TF1 and T1

Pump detection is following in % (I used 0x55 to test)
```
[telegram] Rx: 60 00 FF 00 04 4E 55 01 C6
                                                   PR1
[emsesp] No telegram type handler found for ID 0x54E (src 0x60)
```

Values of valves are not so easy to detect. I can only set On or Off in test mode. AM200 will adjust to any valve position depending on temperatures. 

Let us start with temperatures and pump. What do you think?

MichaelDvP, 2022-07-18:
Ok, first test here: [https://github.com/MichaelDvP/EMS-ESP32/releases](https://github.com/MichaelDvP/EMS-ESP32/releases)
We can do some iterations and renames and if all works i'll do a PR.

OlliHein, 2022-07-18:
Wow, that's fast. I will try it.

Mit freundlichen Grüßen
Oliver Heine

MichaelDvP, 2022-07-18:
Ups, forgot that boilers have a special handling when detecting. Hopefully it works now.

OlliHein, 2022-07-18:
Great,
It works and I can read temperatures. I will try to check pump and values later.
Where did you get the informations about valves

[image]
?

OlliHein, 2022-07-18:
I mark all needed entries as favourites and enable it for MQTT.  I have curflowtemp twice in both boilers abialable but with different temperature values.

MichaelDvP, 2022-07-19:
The valves are not correct at the moment, i expect them in telegram 54E like pump in offset 0, and read offset 1,2,3 as bool. But i think some are mixing valves an use percent-states. You have to ckeck the right values/positions.

The flow temp is TB4/TR2, the en-manual says "system flow temp" and "system return temp", i thought it is the same as main boiler curflowtemp, but it seems they are different. I'll rename and take the names from the manual.

For mqtt it we can use a tag "hsa" (heating source alternative) like cascaded boilers (hs1-hs16) and heating circuits (hc1-hc4).
This shows as json-nest in "boiler"-topic, or, if you choose  non-nested format, as extra "boiler_hsa"-topic.

If it helps we can add the input to names i.e. "system flow temperature (TB4)", and/or use the input as mqtt-shortname: "tb4" instead of "sysflowtemp"?

I thought telegram 550 is settings, but it is broadcasted (uncommon for settings) and there are some values in manual like "set temp return" (65°C), "Set flow temp AHS" (75°C) ,etc i can not see. Try in terminal `read 60 54F`, maybe there are some values. (try also 551).

I've updated my build.

OlliHein, 2022-07-20:
@MichaelDvP Currently it is hard to find out changes of status information in logfile. You already have a great method to post status information with MQTT available in EMS-ESP. I mean it should be easy to write telegram data from bus depending of the devices (i.e. 0x60)  to MQTT as a topic. The value should be the hex stream. If we have this, it will be easy to change data on a terminal and I can be seen directly as changes in MQTT explorer. Wat do you think?

MichaelDvP, 2022-07-20:
Emsesp has already a lot of tools to examine the bus, remember that emsesp is based on reverse engineering the bus.
Best to do from terminal with `watch` and `read` commands.
`watch` affects also syslog and weblog and can also be set by API or mqtt.
If you like to read a specific telegram via mqtt, use topic `system` with payload `{"cmd":"send","data":"0B E0 FF 00 1B 04 4E"}`
(same as `read 60 54E` or `call system send "0B E0 FF 1B 04 4E"` in terminal) and look at the topic `response` in your mqtt-broker.

OlliHein, 2022-07-22:
Yes, you are right, I hat to read documentation first. The terminal watch is a good way to examine bus data.
I will try it this weekend. Send you an update if I have the missing data / information's.

OlliHein, 2022-07-23:
@MichaelDvP Here are the values of both valves:

I set buffer bypass to **off** in diagnostic menu:
```
000+21:20:58.702 N 362: [emsesp] Thermostat(0x10) -> Boiler(0x60), ?(0x054F), data: 00 (offset 3)
000+21:21:14.321 N 363: [emsesp] Thermostat(0x10) <- Boiler(0x60), ?(0x054F), length: 0x01 (offset 3)
000+21:21:14.342 N 364: [emsesp] Boiler(0x60) -> Thermostat(0x10), ?(0x054F), data: 00 (offset 3)
000+21:21:25.569 N 366: [emsesp] Boiler(0x60) -> All(0x00), AmTemperatures(0x054D), data: 00 F1 00 F4 80 00 00 DD 01 0C 01 CF 80 00 01 C4 00 00 80 00
000+21:21:25.771 N 367: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 00 00 00 00 00 00
000+21:21:26.454 N 368: [emsesp] Boiler(0x60) -> All(0x00), AmSettings(0x0550), data: 00 FF FF FF FF FF FF 00 02 00 00 00 00 00 03 00 00 03 00 2D 19 C8 02 94 00
000+21:21:26.654 N 369: [emsesp] Boiler(0x60) -> All(0x00), AmSettings(0x0550), data: 00 FF FF (offset 25)
```

I set buffer bypass to **open** in diagnostic menu:
```
000+21:12:14.200 N 170: [emsesp] Thermostat(0x10) -> Boiler(0x60), ?(0x054F), data: 01 (offset 3)
000+21:12:15.413 N 171: [emsesp] Boiler(0x60) -> All(0x00), ?(0x054F), data: 01 (offset 3)
000+21:12:15.598 N 172: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 08 (offset 1)
000+21:12:17.514 N 173: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 03 (offset 5)
000+21:12:18.562 N 174: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 04 (offset 5)
000+21:12:20.153 N 176: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 06 (offset 5)
000+21:12:22.421 N 177: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 08 (offset 5)
000+21:12:24.176 N 178: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 0B (offset 5)
000+21:12:25.241 N 179: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 00 08 00 00 00 0C
000+21:12:25.486 N 180: [emsesp] Boiler(0x60) -> All(0x00), AmTemperatures(0x054D), data: 00 F1 00 F4 80 00 00 DD 01 0D 01 CF 80 00 01 C4 00 00 80 00
000+21:12:27.739 N 181: [emsesp] Boiler(0x60) -> All(0x00), AmSettings(0x0550), data: 00 FF FF FF FF FF FF 00 02 00 00 00 00 00 03 00 00 03 00 2D 19 C8 02 94 00
000+21:12:27.928 N 182: [emsesp] Boiler(0x60) -> All(0x00), AmSettings(0x0550), data: 00 FF FF (offset 25)
000+21:12:30.398 N 183: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 11 (offset 5)
000+21:12:32.761 N 184: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 13 (offset 5)
000+21:12:33.841 N 185: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 15 (offset 5)
000+21:12:35.862 N 187: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 17 (offset 5)
000+21:12:37.037 N 188: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 18 (offset 5)
000+21:12:37.896 N 189: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 19 (offset 5)
000+21:12:39.455 N 190: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 1B (offset 5)
000+21:12:41.199 N 191: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 1D (offset 5)
000+21:12:43.027 N 192: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 1F (offset 5)
000+21:12:44.426 N 193: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 21 (offset 5)
000+21:12:45.839 N 194: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 22 (offset 5)
000+21:12:47.296 N 195: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 24 (offset 5)
000+21:12:48.461 N 196: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 25 (offset 5)
000+21:12:49.649 N 197: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 26 (offset 5)
000+21:12:51.107 N 198: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 29 (offset 5)
000+21:12:51.884 N 199: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 29 (offset 5)
000+21:12:53.466 N 200: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 2B (offset 5)
000+21:12:55.201 N 201: [emsesp] Thermostat(0x10) <- Boiler(0x60), ?(0xF9), length: 0x11
000+21:12:55.275 N 202: [emsesp] Boiler(0x60) -> Thermostat(0x10), ?(0xF9), data: FF 04 4F 03 47 00 00 00 00 00 00 00 00 00 07 00 02 00 07 00 01
000+21:12:55.549 N 203: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 2D (offset 5)
000+21:12:55.732 N 204: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 2D (offset 5)
000+21:12:56.675 N 205: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 2E (offset 5)
000+21:12:59.593 N 206: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 31 (offset 5)
000+21:13:01.183 N 207: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 34 (offset 5)
000+21:13:02.016 N 208: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 34 (offset 5)
```
I set buffer bypass to **close** in diagnostic menu:
```
000+21:13:46.792 N 245: [emsesp] Thermostat(0x10) -> Boiler(0x60), ?(0x054F), data: 02 (offset 3)
000+21:13:47.463 N 246: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 62 (offset 5)
000+21:13:47.652 N 247: [emsesp] Boiler(0x60) -> All(0x00), ?(0x054F), data: 02 (offset 3)
000+21:13:48.648 N 248: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 10 (offset 1)
000+21:13:48.832 N 249: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 61 (offset 5)
000+21:13:50.598 N 250: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 5F (offset 5)
[... cut ...]
000+21:25:37.293 N 513: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 07 (offset 4)
000+21:25:39.872 N 514: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 05 (offset 4)
000+21:25:41.038 N 515: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 03 (offset 4)
000+21:25:42.008 N 516: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 03 (offset 4)
000+21:25:43.921 N 518: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 01 (offset 4)
000+21:25:44.778 N 519: [emsesp] Boiler(0x60) -> All(0x00), AmStatus(0x054E), data: 00 (offset 4)
```

MichaelDvP, 2022-07-25:
Thx, so we have a telgram 0x54F for settings, please use `read 60 54F` to get the whole telegram.

In statusMessage 0x54E we have now in offset:
0: pumpMod (percent)
1: bitfield: 01-pump on, 02 -VR1 opening, 04-VR1 closing, 08-VB1 opening, 10-VB1 closing
4: VR1 percent open
5: VB1 percent open
message 0x54E is only 6 bytes long, so only offsets 2/3 missing, i think for VR2 and OA

and in 0x54F commands for
2: VR1: 0-stop, 1-open, 2-close
3: VB1: 0-stop, 1-open. 2-close

I think the main information to show in ems-esp are the percent values for pump and valves. Is it usefull also to add entities for opening/closing and pump on/off?
Is there a usecase for setting the valves via ems-esp? I think in normal operation the valves are set by module depending on temperature settings and manual setting will be overwritten. 
For tweaking the heating it is imo more  usefull to find and manipulate the values of "set return temp", "boiler release wait time" and "Hyst".

OlliHein, 2022-07-25:
> Is it usefull also to add entities for opening/closing and pump on/off?

I think also the percent values are enought.  0=close; >1-99 = opening; 100=open
For the pump is also clear 0=off; >1 - 100 = running

> Is there a usecase for setting the valves via ems-esp?

I don't think so and I do not need it, AM200 should control valves.

Did you add the new valves informations to the new boiler device? I miss the data in last update v3.4.2b4?
There are also entries of:
- collector shutdown
- cyl heated 
[image]

MichaelDvP, 2022-07-26:
> Did you add the new valves informations to the new boiler device? I miss the data in last update v3.4.2b4?

I have not updated yet, i'm waiting for the complete 54F telegram.
> There are also entries of:
    collector shutdown
    cyl heated

Strange, these are from solar class, don't know howthey came to boiler, maybe something from browser cache?

OlliHein, 2022-07-26:
here is the settings telegram:

```
ems-esp:$ read 60 54f
000+22:29:03.384 I 1: [emsesp] Boiler(0x60) -> Me(0x0B), ?(0x054F), data: 00 00 00 00 00 00 00

```

MichaelDvP, 2022-07-30:
Nice findings, good work. So 54F is only testing mode and 54C are settings. It will take some days to implement all with commands. 
Im missing the positions of HH (maybe one of th 5A  in offset 11 or 18) and OO (default is 5 deg, so i think offset 14).
The mixer runtime default in manual is 120s in range 0..600, there must be a factor, or it is a 16 bit value: KK pos 8/9 and HH pos 10,11?  Your setting is 90s (5A) for HH and KK, right?

MichaelDvP, 2022-07-31:
First implementation with the mentioned assumptions is on my GH. Please check.

OlliHein, 2022-08-04:
Hi @MichaelDvP ,
Sorry for the delay, I was bussy this week.
I forgot to write positions for HH and OO
Here are a corrected output of values:

```
Boiler(0x60) -> Me(0x0B), ?(0x054C), data: 00 01 01 00 01 00 41 4B 00 5A 00 5A 00 01 05 3C 00 00 5A 00 01 23 00
                                           BB FF JJ DD CC    GG II    HH    KK AA    OO PP MM NN       LL EE

``` 

>  KK pos 8/9 and HH pos 10,11? Your setting is 90s (5A) for HH and KK, right?

-> I corrected settings of HH, KK and OO. I did a mistake. Yes, my setting of (HH and KK) are both 90s 
    
I  updated my ems-esp with new version and it look nice with your current implementation ;)
Let us coninue - thx

MichaelDvP, 2022-08-04:
>> KK pos 8/9 and HH pos 10,11? Your setting is 90s (5A) for HH and KK, right?

Now you have marked HH to (8/)9 and KK to (10/)11,  Check which is first.

OlliHein, 2022-08-04:
Yes KK is pos 8/9 and HH is pos 10/11 I cheked it twice. It was a mistake in my first documentation. 
Me settings are 90s (5A) for HH and KK

OlliHein, 2022-08-06:
@MichaelDvP I updated following post with lager data to show the overflow of a 255s value
https://github.com/emsesp/EMS-ESP32/issues/573#issuecomment-1205458141

OlliHein, 2022-08-19:
@MichaelDvP Do you have enough information to bring AM200 running? If you need more data I can update you.

proddy, 2022-08-20:
@OlliHein I've merged Michael's AM200 additions to the latest dev branch/release.

closed 2022-10-29 by proddy