emsesp/EMS-ESP32 #574: DWH flow temp offset can be > 40

proddy, 2022-07-17:
As reported in [Discord](https://discord.com/channels/816637840644505620/816958087160266775/998120269522673774) by vmonkey...

> Hi, in the latest beta, HA logs report this:

> Logger: homeassistant.components.mqtt.number
> Source: components/mqtt/number.py:211
> Integration: MQTT (documentation, issues)
> First occurred: 08:39:29 (65 occurrences)
> Last logged: 08:51:11

> Invalid value for number.boiler_dhw_flow_temperature_offset: 42 (range 0.0 - 40.0)

MichaelDvP, 2022-07-17:
For my ems-boiler (telegram 33, offset 5) it **is** limited to 40°C. Maybe telegram EA, offset 9 has a different limit, or it depends on boiler.
```
ems-esp:# call boiler wwflowtempoffset 38
000+03:47:08.879 N 3: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 26 (offset: 0x05)
000+03:47:09.198 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2F FB FF 26 FF 07 46 00 00
ems-esp:# call boiler wwflowtempoffset 40
000+03:47:13.346 N 6: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 28 (offset: 0x05)
000+03:47:14.202 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2F FB FF 28 FF 07 46 00 00
ems-esp:# call boiler wwflowtempoffset 42
000+03:47:34.652 N 15: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 2A (offset: 0x05)
000+03:47:35.365 N 17: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2F FB FF 28 FF 07 46 00 00
```

vmonkey, 2022-07-17:
{
  "System Status": {
    "version": "3.4.2b3",
    "uptime": "000+00:06:39.841",
    "freemem": 136,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -55,
    "IPv4 address": "192.168.68.113/255.255.255.0",
    "IPv4 gateway": "192.168.68.1",
    "IPv4 nameserver": "192.168.68.114",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "216.239.35.0",
    "tz label": "Europe/Prague"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 485,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 2,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 0,
    "publish time thermostat": 0,
    "publish time solar": 0,
    "publish time mixer": 0,
    "publish time other": 0,
    "publish time sensor": 0,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Status": {
    "API calls": 2,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 529,
    "bus reads (tx)": 153,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "18.14",
      "entities": 64,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "18.05",
      "entities": 76,
      "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02CE 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x02D0 0x02D2 0x031E",
      "handlers ignored": "0x0292 0x02E2 0x02EC"
    },
    {
      "type": "Mixer",
      "name": "MM100",
      "device id": "0x21",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D8",
      "handlers ignored": "0x0255 0xBF"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "18.14",
      "entities": 0
    }
  ]
}

vmonkey, 2022-07-17:
ems-esp:# call boiler wwflowtempoffset 42
000+00:20:14.689 D 12: [command] Calling command 'boiler/wwflowtempoffset' (flow temperature offset) with value 42
000+00:20:14.876 D 13: [telegram] Sending write Tx [#172], telegram: 8B 08 33 05 2A
000+00:20:14.876 T 14: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 2A (offset 5)
000+00:20:14.876 D 15: [emsesp] Process setting 0x33 for device 0x08
000+00:20:14.905 D 16: [emsesp] Last Tx write successful
000+00:20:14.905 D 17: [telegram] Sending post validate read, type ID 0x33 to dest 0x08
000+00:20:14.965 D 18: [mqtt] Publishing topic ems-esp/boiler_data_ww (#691, retain=0, retry=1, size=460, pid=1)
000+00:20:15.020 T 19: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 2B 00 00 2A 00 02 4B 00 FF FF

MichaelDvP, 2022-07-17:
So it's not depending on telegram, sad. What's the limit for your boiler?

vmonkey, 2022-07-17:
Sorry for a stupid question, but how to find that out? Should this be present somewhere in the service menu of my thermostat?

For clarification: I believe that a former value was probably in the preset range, and I may have changed it by lowering the boiler_dhw_selected_temperature using the boiler physical button.

MichaelDvP, 2022-07-17:
> Should this be present somewhere in the service menu of my thermostat?

Maybe.
The easiest way is to check what the boiler accepts.  In my test i have set `watch 33` and then:
```
ems-esp:# call boiler wwflowtempoffset 42
000+03:47:34.652 N 15: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 2A (offset: 0x05)
000+03:47:35.365 N 17: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2F FB FF 28 FF 07 46 00 00
```
as you see N15 sets offset 5 to 2A, but in automatic check the telegram N17 shows in offset 5: 28 (not changed).
You have to test the `call boiler wwflowtempoffset xx` one by one to get the maximum value, a to large value is ignored by boiler.

vmonkey, 2022-07-28:
Sorry for the late response, I was on vacation. I tried what you suggested; however, it seems that wwflowtempoffset is not writable. Regardless of what I send, the offset 5 stays at 2A (which corresponds to the value 42).

proddy, 2022-07-28:
can you try this with logging on so we can see what is sent and what is returned?

vmonkey, 2022-07-30:
Is this what you mean?

`ems-esp:# call boiler wwflowtempoffset 42
000+00:01:20.499 N 1: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 2A (offset 5)
000+00:01:21.003 N 2: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
000+00:01:21.373 N 3: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+00:01:21.409 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
ems-esp:# call boiler wwflowtempoffset 41
000+00:01:41.331 N 5: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 29 (offset 5)
000+00:01:41.394 N 6: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
000+00:01:41.763 N 7: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+00:01:41.800 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
ems-esp:# call boiler wwflowtempoffset 40
000+00:01:56.752 N 9: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 28 (offset 5)
000+00:01:56.825 N 10: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
000+00:01:57.189 N 11: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+00:01:57.225 N 12: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
ems-esp:# call boiler wwflowtempoffset 39
000+00:02:00.155 N 13: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 27 (offset 5)
000+00:02:00.223 N 14: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
000+00:02:01.498 N 15: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+00:02:01.534 N 16: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF
000+00:02:02.622 N 17: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+00:02:02.658 N 18: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 2A 00 00 2A 00 02 4B 00 FF FF`

proddy, 2022-07-30:
after the write do you see a Tx successful? If so, then maybe setting the temp offset is handled differently on an HT3 type boiler. We'll need to back to Norberts doc and figure it out.

vmonkey, 2022-07-31:
Yes, Tx write successful

vmonkey, 2022-08-11:
Today's message

Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:216
Integration: MQTT (documentation, issues)
First occurred: 10. srpna 2022 23:01:05 (1320 occurrences)
Last logged: 05:00:24

Invalid value for number.boiler_dhw_flow_temperature_offset: 70 (range 0.0 - 45.0)

What I do not understand is why the range is different to to the previous case.

proddy, 2022-08-11:
I think we should just change the max temp limit to 100

MichaelDvP, 2022-08-11:
> What I do not understand is why the range is different to to the previous case.

Simple: we have changed it because it seems that the value was fixed to 42 in your boiler. But now you have 70°C. 
Maybe the boiler always uses a high flowtemp for dhw preparation like 90° and calculate the offset to the current dhw setpoint. Do you have something like 48° dhw before and now reduced to 20°C? 
> I think we should just change the max temp limit to 100

agree. 
If this happpens with other values, maybe we should change the ha-config dynamically if the reported value is out of range.

vmonkey, 2022-08-11:
Yes, the boiler uses a fairly high temperature for hot water preparation, see the attached graph (this is with heating off indeed).
[image]

proddy, 2022-08-21:
This has been changed to 100 in 3.4.2b5

closed 2022-08-21 by proddy