emsesp/EMS-ESP32 #575: Unknown module: Buderus EM10 (Device-ID: 0x12, Product-ID: 74)

Oktonaut, 2022-07-18:
**Is your feature request related to a problem? Please describe.**
*I have a Buderus GB142 with BC10, RC35, MM10, SM10 and EM10, unfortunately the EM10 module (External Error Detection) is displayed as unknown in the dashboard.*

**Describe the solution you'd like**
*It would be nice if the module could be identified correctly and the status of the two outputs and the measurement data of the analog input could also be read out.*

**Describe alternatives you've considered**
*If the module does not deliver any data via EMS, at least the module designation: EM10 and the type: External should be displayed.*

**Additional context**
*Device ID: 0x12
Product ID: 74
Version: 01.03
Screenshot:* https://c.gmx.net/@326646499389735493/niC42mSPQSqIxB0Pf4NYzw
*Manual:* https://www.manualslib.de/manual/38125/Buderus-Em10-F%C3%BCr-Uba-3.html

MichaelDvP, 2022-07-18:
Adding the module is easy (try my testbuild [(https://github.com/MichaelDvP/EMS-ESP32/releases)](https://github.com/MichaelDvP/EMS-ESP32/releases)), but for the values you need to identify the telegrams and positions. Best way is to use the terminal and type `watch 12` to get all messages from/to this module. Then try to modulate the input and try to trigger an error to see what changes in telegrams.

MichaelDvP, 2022-07-19:
> I loaded the latest Development Build v3.4.2b3

From here or my (linked) repo?

Oktonaut, 2022-07-19:
Loaded it from the link in the version check. Thought it would be the same based on the number.

With the adapted version, however, it is now displayed. Thanks, great!

Unfortunately, I can't say when I'll be able to log the telegrams.

MichaelDvP, 2022-07-19:
> Unfortunately, I can't say when I'll be able to log the telegrams.

As first check you can go to system-log, select ' log all' , wait a bit more than a minute, export the log and attach it here.
With a bit luck there will be single byte messages from and to the module and current state means no error.

Oktonaut, 2022-07-19:
Here is the [system log](https://github.com/emsesp/EMS-ESP32/files/9141159/log.txt). It's a little longer than a minute... 0x12 is also included.

P.S: However, I don't think the type: "Connect" is appropriate for this. Since this is more or less a separate device category, you could actually name it "Interface" and create a new logo for it!?

MichaelDvP, 2022-07-19:
Message 0x12 is included, but nothing from/to device 0x12. Maybe it only listens to errormessages and sends only if there is a voltage. Or the sending period is much longer. 
> P.S: However, I don't think the type: "Connect" is appropriate for this. 

We have a category "switch", and "Gateway", both fits not 100%.  It's difficult, the 0-10V input is a control device, the error output a switch or something different.

Oktonaut, 2022-07-22:
I was able to log some data via telnet last night.

First I connected a 9V battery to the analog input and finally provoked a 6A error.

`ems-esp:$ watch 12`
`Watching incoming telegrams, displayed in decoded format`
`Filtering only telegrams that match a deviceID or telegram type of 0x12`
`002+03:57:14.632 N 0: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 50 (offset 1)`
`002+03:57:45.803 N 1: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 3E (offset 1)`
`002+03:57:46.648 N 2: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 1)`
`002+03:58:49.036 N 3: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 48`
`002+03:59:49.612 N 4: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 4A`
`002+04:00:08.066 N 5: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 51 (offset 1)`
`002+04:00:13.585 N 6: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 4A`
`002+04:00:27.738 N 7: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 39`
`002+04:00:28.583 N 8: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 0D`
`002+04:00:29.758 N 9: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 00`
`002+04:01:33.034 N 10: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 47`
`002+04:01:37.460 N 11: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 51 (offset 1)`
`002+04:02:25.049 N 12: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 4A`
`002+04:02:29.632 N 13: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 51 (offset 1)`
`002+04:02:47.080 N 14: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 3D (offset 1)`
`002+04:02:47.911 N 15: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 1)`
`002+04:02:51.163 N 16: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 52 (offset 1)`
`002+04:03:50.740 N 17: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 52 (offset 1)`
`002+04:04:51.235 N 18: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 52 (offset 1)`
`002+04:05:51.632 N 19: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 52 (offset 1)`
`002+04:06:51.224 N 20: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 52 (offset 1)`
`002+04:07:36.017 N 21: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 3D (offset 1)`
`002+04:07:36.863 N 22: [emsesp] Connect(0x12) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 1)`
`ems-esp:$`

MichaelDvP, 2022-07-22:
Ok, thanks. Seems the module does not have any own value and telegram.
Telegram 0x1A offset 0 is `setFlowTemp`, offset 1 is `setBurnPower` in boiler settings. No need to show it redundant for the module.
For the error there is nothing sent to the module, so it seems the module listen to 0x10 or 0xBF messages to read the error states.

So it's only to show that the module is connected and decide what category we use. "Connect", "Switch", "Gateway" or something new (how to call it? What icon?)? @proddy Any suggestions? 

The 230V output is something like a switch, the 24V relais is for connecting a PLC (connect or gateway fits), the input is also for connecting a PLC and remote control the boiler (also more gateway or connect?). For a new category i dont have any idea how to call it and what icon.

proddy, 2022-07-22:
maybe a Connect or we invent something new. Any idea what the KM200 calls it?

Oktonaut, 2022-07-22:
Since the device is something like an intermediary between two systems, I would call it an "interface" or "brocker".

For the logo maybe something like this:
[image]
[image]
[image]

closed 2022-11-11 by proddy