emsesp/EMS-ESP32 #576: MQTT Home-Assistant Template variable error

makuser, 2022-07-19:
**Bug description**
`~/boiler_data` does not contain boiler data for all boilers, when cascade is used

**Steps to reproduce**

1. Connect ems-esp to an EMS bus with for example: Cascade MC400 and 4x Logamax Plus/GB192/Condens GC9000
2. Enable MQTT
3. Look at for example `homeassistant/sensor/ems-esp/boiler_hs13_burnworkmin/config`, which references `hs13.burnworkmin` in `~/boiler_data`

**Expected behavior**
MQTT Home Assistant Discovery works
`~/boiler_data` contains one dict per boiler

**Logs**
```log
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs1' when rendering '{{value_json.hs1.burnworkmin}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs1' when rendering '{{value_json.hs1.selflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs1' when rendering '{{value_json.hs1.selburnpow}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs1' when rendering '{{value_json.hs1.curflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs1' when rendering '{{value_json.hs1.curburnpow}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs5' when rendering '{{value_json.hs5.burnworkmin}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs5' when rendering '{{value_json.hs5.selflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs5' when rendering '{{value_json.hs5.selburnpow}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs5' when rendering '{{value_json.hs5.curflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs5' when rendering '{{value_json.hs5.curburnpow}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs9' when rendering '{{value_json.hs9.burnworkmin}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs9' when rendering '{{value_json.hs9.selflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs9' when rendering '{{value_json.hs9.selburnpow}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs9' when rendering '{{value_json.hs9.curflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs9' when rendering '{{value_json.hs9.curburnpow}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs13' when rendering '{{value_json.hs13.burnworkmin}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs13' when rendering '{{value_json.hs13.selflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs13' when rendering '{{value_json.hs13.selburnpow}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs13' when rendering '{{value_json.hs13.curflowtemp}}'
2022-07-19 11:34:37 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'hs13' when rendering '{{value_json.hs13.curburnpow}}'
```

**Device information**
```json
{
  "System Status": {
    "version": "3.4.1",
    "uptime": "000+00:21:52.977",
    "freemem": 107,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -82,
    "IPv4 address": "10.60.10.131/255.255.255.0",
    "IPv4 gateway": "10.60.10.1",
    "IPv4 nameserver": "10.60.10.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": true,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "pool.ntp.org",
    "tz label": "Europe/Prague"
  },
  "OTA Status": {
    "enabled": false,
    "port": 8266
[... 166 lines cut ...]
    },
    {
      "type": "Mixer",
      "name": "MM100",
      "device id": "0x20",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D7",
      "handlers ignored": "0x0255 0xBF"
    }
  ]
}
```

**Additional context**
MQTT Data:
`homeassistant/sensor/ems-esp/boiler_hs13_burnworkmin/config`
```json
{"~":"ems-esp","uniq_id":"boiler_hs13_burnworkmin","stat_t":"~/boiler_data","name":"Boiler hs13 total burner operating time","val_tpl":"{{value_json.hs13.burnworkmin}}","unit_of_meas":"minutes","ic":"mdi:clock-outline","state_class":"total_increasing","dev":{"ids":["ems-esp-boiler"],"name":"EMS-ESP Boiler","mf":"","mdl":"Logamax Plus/GB192/Condens GC9000","via_device":"ems-esp"}}
```
`ems-esp/boiler_data`
```json
{"heatingactive":"off","tapwateractive":"off","selflowtemp":0,"selburnpow":0,"heatingpump2mod":0,"outdoortemp":28.4,"curflowtemp":65.2,"syspress":0,"burngas":"off","flamecurr":0,"heatingactivated":"on","heatingtemp":80,"burnminperiod":0,"burnminpower":0,"burnmaxpower":100,"boilhyston":-2,"boilhystoff":0,"curburnpow":0,"lastcode":"A01(1901) @uptime 21 - 2 min","servicecodenumber":0,"emergencyops":"off","emergencytemp":30}
```

MichaelDvP, 2022-07-19:
Could you check with the latest dev build 3.4.2b3.  The mqtt handling of multiple devices of same type have changed in 3.4.2.

proddy, 2022-07-25:
Please re-open if it's still an issue

closed 2022-07-25 by proddy