emsesp/EMS-ESP32 #578: ISM1 wrong collector temperature

FredericMa, 2022-07-19:
**Bug description**
During the hot days here in Belgium our solar boiler is fully heated which causes the pump to stop and a rise in temperature of the collector temperature. Within the range of 140°C until 152,4°C the reported value causes a spike that goes from 237,2°C till 247,8°C which is obviously incorrect. See screenshot below.

**Steps to reproduce**
Heat the ISM1 solar collector/ temperature sensor above 140°C 😄 

**Expected behavior**
I don't expect this spike in the reported values.

**Screenshots**
[image]

**Device information**
```
{
  "System Status": {
    "version": "3.4.1",
    "uptime": "001+00:20:37.700",
    "freemem": 155,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp32",
    "RSSI": -18,
    "IPv4 address": "192.168.50.119/255.255.255.0",
    "IPv4 gateway": "192.168.50.1",
    "IPv4 nameserver": "192.168.50.3",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Brussels"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 45289,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": true,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 111910,
    "bus reads (tx)": 14626,
    "bus writes (tx)": 3,
    "bus incomplete telegrams": 0,
    "bus reads failed": 15,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "NODEMCU",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "23.04",
      "entities": 61,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x26 0x2A",
      "handlers ignored": "0x05"
    },
    {
      "type": "Thermostat",
      "name": "FW200",
      "device id": "0x10",
      "product id": 106,
      "version": "12.14",
      "entities": 27,
      "handlers received": "0x06 0x23",
      "handlers fetched": "0x016F 0x0170 0x0165 0x0166 0xBB",
      "handlers pending": "0xA3 0xA2 0x12 0x0171 0x0172 0x0167 0x0168 0x0123",
      "handlers ignored": "0xBE 0x01D3"
    },
    {
      "type": "Mixer",
      "name": "IPM",
      "device id": "0x20",
      "product id": 102,
      "version": "20.08",
      "entities": 2,
      "handlers received": "0x010C",
      "handlers pending": "0x011E",
      "handlers ignored": "0xBE"
    },
    {
      "type": "Mixer",
      "name": "IPM",
      "device id": "0x21",
      "product id": 102,
      "version": "20.08",
      "entities": 4,
      "handlers received": "0x010C",
      "handlers pending": "0x011E",
      "handlers ignored": "0xBE"
    },
    {
      "type": "Solar",
      "name": "ISM1",
      "device id": "0x30",
      "product id": 101,
      "version": "23.04",
      "entities": 8,
      "handlers fetched": "0x0103 0x0101",
      "handlers pending": "0x0104",
      "handlers ignored": "0xBE"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "23.04",
      "entities": 0
    }
  ]
}
```

MichaelDvP, 2022-07-19:
I have SM10 and don't see this (max temperatures 158°C). The handling of collectortemp is identical for SM10 and ISM. Have you checked that it is only wrong in ems-esp? 
[image]

FredericMa, 2022-07-19:
You're right, the thermostat shows the same (wrong) values. So I guess this is coming from the ISM1 or the temperature sensor.
Do you guys have any idea what could cause the error?

I'll close the issue since it is not EMS-ESP related.

closed 2022-07-19 by FredericMa

MichaelDvP, 2022-07-20:
> Do you guys have any idea what could cause the error?

Most probably the sensor. Next time the measurement goes above 140°C: disconnect the wires from ISM and measure the resistance of the sensor, 140°C is ~400 ohm, 240°C is ~20 ohm. 

But i think it does not really matter, the nonlinearity is above the collector shutdown limit and has no effect. Only spikes in the curve.

FredericMa, 2022-07-20:
I'll give it a try. I might need to make an automation that notifies me when this happens since it is a narrow window when the issue occurs 😄 

You're right, it operates normal. It's just the visual aspect that "bothers" me.