emsesp/EMS-ESP32 #581: Loosing entity boiler_wwcomfort

sweetpants, 2022-07-21:
**Bug description**
Coming from the HA Nefit easy integration i purchased the EMS-ESP32S gateway running version 3.4.1 and am very happy with it except for one thing.

The number of entities discovered varies from 86 to 87. Sometimes the entity wwcomfort is there and the next moment it's gone making the entity unavailable in HA now and then.

**Steps to reproduce**
just let HA MQTT discover the device and add all to HA will show this issue

**Screenshots**
[image]

[image]

**Additional context**

HA log shows the discovery of the device and later it's removed again

2022-07-21 12:02:57 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: select ems-esp boiler_wwcomfort

2022-07-21 12:03:25 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wwcomfort' when rendering '{{value_json.wwcomfort}}'
2022-07-21 12:03:25 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.boiler_dhw_comfort: '' (valid options: ['hot', 'eco', 'intelligent'])

Not sure what is happening here as i am just starting to ute this EMS gateway over the Nefit integration

Regards, Harry

sweetpants, 2022-07-21:
Tested EMS-ESP Version, v3.4.2b3, same problem

proddy, 2022-07-21:
EMS-ESP will remove entities (also from HA) when they drop off the EMS bus feed to avoid errors in HA because the info in boiler_data/thermostat_data is missing. But they will be added back when they get re-discovered.

How often do you lose wwcomfort? does it come back? what does the EMS-ESP logs say?

MichaelDvP, 2022-07-21:
Yes, there is an issue in handling wwComfort when reading parts of telegram 0x33 without this value.
I'll make a fix. 
@proddy  see [here](https://github.com/MichaelDvP/EMS-ESP32/commit/295bbed4ae660eafe8f2ae2334fdae8d2dfa5858)
Could we wait for completing the AM200 and EM10 changes, or do you want add this fix earlier?
@sweetpants You can try with [this dev build](https://github.com/MichaelDvP/EMS-ESP32/releases)

proddy, 2022-07-21:
> Yes, there is an issue in handling wwComfort when reading parts of telegram 0x33 without this value. I'll make a fix. @proddy see [...]

ah great find @MichaelDvP. You can do a PR now or wait, up to you. I'm sure sweetpants (!) can use the build you linked to in the time being

sweetpants, 2022-07-24:
@MichaelDvP, i have installed the Development Build v3.4.2b4, see what happens. At first glance it looks better now.
@proddy , it used to change/disappear every few minutes.

When subscribing to the ems-esp/boiler_data_ww topic i now always get a value for wwcomfort.

Great job guys, thanks a lot

commit 295bbed4 (2022-07-23): fix #581
closed 2022-07-25 by proddy