emsesp/EMS-ESP32 #582: Option to invert RX and TX lines' logic level

minusdreidb, 2022-07-21:
I have designed and built my own somewhat simplified variant of an EMS-bus interface, that required me to invert the logic level of the ESP32's TX-gpio pin. The same functionality could be useful for the RX-gpio pin too with other interface circuits.

 Proposed changes to implement this feature and selecting inverted levels by configuring negative GPIO PIN numbers for RX and TX in the settings interface can be found in my branch https://github.com/minusdreidb/EMS-ESP32/tree/inv_ems_lines. I really only tested the direct patch of `uart_set_line_inverse(EMSUART_UART, inv_mask)` into `src/uart/emsuart_esp32.cpp` and added all the code changes for the WebUI and settings as an afterthought.

Perhaps someone from the core developers could review my changes and hint if I missed something. I would be happy to create a pull request and see my changes merged if deemed worthy, as I think they could be useful for other people and make the configuration and adaption of the firmware more flexible.

MichaelDvP, 2022-07-22:
Nice idea to use negative pin-numbers. But it's a hidden feature and in a few month nobody remembers that we have it. We need some explanations in setting page and/or documentation. Isn't it more obvious to add two checkboxes `[ ] invert tx`, `[ ] invert rx` to the custom board setting? (Also this is a bit more in line with LED handling (checkbox hide led, but in general options)).

Your uart code will only work in hardware tx-mode, the other tx-modes use tx-invert to generate the break. ([see here](https://github.com/minusdreidb/EMS-ESP32/blob/909cfee49ac17c746cf360e811ac224bd9c82b26/src/uart/emsuart_esp32.cpp#L212-L214)). Rx will work in 3.4.1, but not in 3.4.2dev.

For a PR you should use the dev branch. And the current dev uses the new espressive/arduino core with a bit different uart handling. You have to make your inv_mask file-global and use for breaks (e.g. [here](https://github.com/emsesp/EMS-ESP32/blob/32dcd7313bec0c58eb697c5c60f134d2eccf2848/src/uart/emsuart_esp32.cpp#L164-L166)):
```
    uart_set_line_inverse(EMSUART_NUM, inv_mask ^ UART_SIGNAL_TXD_INV);
    delayMicroseconds(EMSUART_TX_BRK_EMS);
    uart_set_line_inverse(EMSUART_NUM, inv_mask);
```

proddy, 2022-07-22:
would also love to see the simplified schematics

minusdreidb, 2022-07-22:
> But it's a hidden feature and in a few month nobody remembers that we have it. We need some explanations in setting page and/or documentation. Isn't it more obvious to add two checkboxes `[ ] invert tx`, `[ ] invert rx` to the custom board setting?

I [added an explanation](https://github.com/minusdreidb/EMS-ESP32/commit/7c58f4e081dfe5a1cd175300d058bf001ccff837) to the text field labels for documentation. As for me, I'd like to keep the solution with the negative GPIOs for inverting line levels as it requires minimal code changes. I'm not that fond of UI programming and also wanted to avoid the hassle of extending the UI.

> Your uart code will only work in hardware tx-mode, the other tx-modes use tx-invert to generate the break. ([see here](https://github.com/minusdreidb/EMS-ESP32/blob/909cfee49ac17c746cf360e811ac224bd9c82b26/src/uart/emsuart_esp32.cpp#L212-L214)). Rx will work in 3.4.1, but not in 3.4.2dev.

> For a PR you should use the dev branch. And the current dev uses the new espressive/arduino core with a bit different uart handling. You have to make your inv_mask file-global and use for breaks [...]

[Done](https://github.com/minusdreidb/EMS-ESP32/blob/inv_ems_lines/src/uart/emsuart_esp32.cpp). I'd really appreciate if you could review my changes again. Please note that they are not confirmed on HW yet. I'm willing to do that if all necessary bases in the uart code are covered.

I also did some code deduplication in `EMSuart::transmit` and could not resist replacing the if-branches with a switch-case 😀.

Is there some way to at least test the UI on my PC instead of uploading to the ESP32?

minusdreidb, 2022-07-22:
> would also love to see the simplified schematics

I'll eventually upload those to github too. Currently they're still work in progress. But basically I didn't like the inverting comparator in front of the TX-Transistor in the typical circuits one finds online (which I believe bbqkees is also using), firstly as it is only really needed for inverting the GPIO's logic level and secondly and more importantly to also get rid of the pullup for the transistor. If the inverter failed the transistor would permanently pull the bus low and possibly ruin communication between the remaining nodes.

MichaelDvP, 2022-07-23:
> I [added an explanation to the text field labels for documentation.

That's also a solution.
> Please note that they are not confirmed on HW yet

Yes, the `<brk>` generation is wrong, it should invert the tx-line by xor and set it back to idle (invert_mask):
```
    uart_set_line_inverse(EMSUART_NUM, invert_mask ^ UART_SIGNAL_TXD_INV);
    delayMicroseconds(length_us);
    uart_set_line_inverse(EMSUART_NUM, invert_mask);
```
> Is there some way to at least test the UI on my PC instead of uploading to the ESP32?

Yes, take a look at the [mock-api](https://github.com/emsesp/EMS-ESP32/tree/dev/mock-api)

minusdreidb, 2022-07-23:
Oops, i was concerned about not changing state of RX and did exactly that, for <brk> generation it would have worked though 🙈. Thanks, will fix.

bbqkees, 2022-07-25:
> I'll eventually upload those to github too. Currently they're still work in progress. But basically I didn't like the inverting comparator in front of the TX-Transistor in the typical circuits one [...]

A working RX/TX circuit with the least amount of components (that I know of) is only like 7 components. It's based around a zener diode. But I never liked that one.
By putting the pretty robust LM393 in between both RX and TX of the flimsy ESP GPIO and the 16V bus (with allowed noisy wire lengths up to 100m) you provide a kind of additional 'shield' around the ESP.
If you fry the TX transistor, the bus can come into direct contact with the GPIO if you omit the LM393.

closed 2023-02-19 by proddy