emsesp/EMS-ESP32 #584: MQTT disconnects in 3.4.2.b4

mtc716, 2022-07-22:
I checked the new beta which has been released and I find the same issue.
MQTT disconnects and does not reconnect automatically.
Disabling and enabling of MQTT solves the problem.

In the logs I do not find any hint for a Wifi loss like it has been in 3.4.2b3 or B2. Is it just disabled in the logs?

mtc716, 2022-07-22:
@proddy : Can I do something to support?

sweetpants, 2022-07-24:
I'm running the same version but do not recognize this issue. How often does this appear?

proddy, 2022-07-24:
I've been trying for months to reproduce this and can't, very frustrating. @mtc716 if the MQTT or WiFi disconnects you'll see a Warning message in the logs. If it's frequent you may find it in the EMS-ESP logs. Alternatively, run a SysLog server to capture the logs. What I do when debugging is plug a micro-usb into the ESP32, connect it to the PC, change pio_local.ini to enable the Serial port and then see the logs running in on the Serial console via ptty/telnet.

sweetpants, 2022-07-25:
I've setup remote syslog (ALL) and will watch my gateway for a while also. At the moment it's running fine

mtc716, 2022-07-25:
@proddy syslog has been setup and I will be watching for the error. Just for your information, meanwhile I jumped back to 3.4.1. and here I clearly see that when MQTT disconnects it automatically reconnects a couple of seconds later. Which let me conclude that the MQTT broker is working well.
Now I am on 3.4.2.b4 and I assume that latest by tomorrow I have a reoccurrence. Should I send you the complete log via github?

mtc716, 2022-07-25:
One thing I directly observed while switching from 3.4.1 to 3.4.2.b4 is that after startup of the system I can see the logs for Wifi starting up, but there is no Startup of MQTT even when its enabled. When I switch if OFF and ON in the WebUI, then its working like expected. Could this be a hint?

proddy, 2022-07-25:
Good spot. I think it's a reporting or timing error, I'll check.  Is this in SysLog, console or the web UI?

mtc716, 2022-07-25:
> Good spot. I think it's a reporting or timing error, I'll check. Is this in SysLog, console or the web UI?

No entries in the syslogs, except the starting up of Wifi, NTP, ....etc.. The "expected" entry MQTT connected is just missing. 
I observed it in the WebUI that MQTT status stayed disconnected after the uploading of 3.4.2.b4 and the normal restart....

proddy, 2022-07-26:
I'm not seeing that. Do you have a screenshot? This is what it looks like on my test system:

[image]

mtc716, 2022-07-26:
On 3.4.1. it looked like the following:

2022-07-25 21:18:38.342 INFO 1809: [mqtt] MQTT disconnected: TCP
2022-07-25 21:19:33.079 INFO 1811: [mqtt] MQTT connected

On 3.4.2.b4 it looks like:

Jul 25 21:41:41 ems-esp-new emsesp 000+00:00:00.000 I 1: Starting EMS-ESP version 3.4.2b4 (hostname: ems-esp-new)
Jul 25 21:41:41 ems-esp-new dallassensor 000+00:00:00.000 I 2: Starting Dallas sensor service
Jul 25 21:41:41 ems-esp-new analogsensor 000+00:00:00.000 I 3: Starting Analog sensor service
Jul 25 21:41:41 ems-esp-new emsesp 000+00:00:00.000 I 4: Loaded EMS device library (102 records)
Jul 25 21:41:41 ems-esp-new emsesp 000+00:00:00.000 I 5: Starting Access Point with captive portal on 192.168.4.1
Jul 25 21:41:41 ems-esp-new emsesp 000+00:00:00.000 D 6: No telegram type handler found for ID 0x2A5 (src 0x10)
Jul 25 21:41:41 ems-esp-new emsesp 000+00:00:00.000 D 7: No telegram type handler found for ID 0x31D (src 0x10)
Jul 25 21:41:41 ems-esp-new emsesp 000+00:00:00.000 D 8: No telegram type handler found for ID 0x1A (src 0x10)
Jul 25 21:41:41 ems-esp-new emsesp 000+00:00:00.000 D 9: No telegram type handler found for ID 0x267 (src 0x10)
Jul 25 21:41:42 ems-esp-new emsesp 000+00:00:02.178 D 10: No telegram type handler found for ID 0x06 (src 0x10)
Jul 25 21:30:41 ems-esp-new emsesp 000+00:00:04.351 I 11: Starting NTP
Jul 25 21:30:41 ems-esp-new emsesp 000+00:00:04.351 I 12: WiFi connected with IP=192.168.178.36, hostname=ems-esp-new
Jul 25 21:30:41 ems-esp-new emsesp 000+00:00:04.384 I 13: mDNS responder service started
Jul 25 21:41:41 ems-esp-new system 000+00:00:04.987 I 14: NTP connected
Jul 25 21:41:42 ems-esp-new telegram 000+00:00:05.869 D 15: Sending read Tx [#0], telegram: 0B 88 07 00 1B
Jul 25 21:41:42 ems-esp-new emsesp 000+00:00:05.909 D 16: Last Tx read successful
Jul 25 21:41:42 ems-esp-new emsesp 000+00:00:05.909 D 17: New EMS device detected with ID 0x08. Requesting version information.
Jul 25 21:41:42 ems-esp-new telegram 000+00:00:05.909 D 18: Tx read request to device 0x08 for type ID 0x02
Jul 25 21:41:42 ems-esp-new emsesp 000+00:00:05.909 D 19: New EMS device detected with ID 0x09. Requesting version information.
Jul 25 21:41:42 ems-esp-new telegram 000+00:00:05.909 D 20: Tx read request to device 0x09 for type ID 0x02
Jul 25 21:41:42 ems-esp-new emsesp 000+00:00:05.909 D 21: New EMS device detected with ID 0x10. Requesting version information.
Jul 25 21:41:42 ems-esp-new telegram 000+00:00:05.909 D 22: Tx read request to device 0x10 for type ID 0x02

mtc716, 2022-07-26:
@proddy: I can directly provoke the error. If everything works well and MQTT is enabled and publishes correct values, I just hit the "RESTART" button in the WebUI and the ems-esp restarts and logging as expected, but MQTT is disconnected and stays like it is.

[image]
[image]
[image]
[image]

mtc716, 2022-07-26:
@proddy : Might it be that I have somehow a corrupted configuration or setting, which prevents a normal automatic startup of MQTT?
If so, I could try a factory reset on 3.4.2b4 and check if it helps?

proddy, 2022-07-26:
it's not corruption, I think the EMS-ESP can't connect to the broker. Which MQTT broker are you running and how? We need to turn on its verbose logging to see why the connection is failing. Perhaps an authentication error. It would be handy if you shared your MQTT settings to see if there is a difference.

mtc716, 2022-07-26:
> it's not corruption, I think the EMS-ESP can't connect to the broker. Which MQTT broker are you running and how? We need to turn on its verbose logging to see why the connection is failing. Perhaps [...]

I will check for the loggings in the broker. Anyhow, whenever I press SAVE in the MQTT Settings Tab, the connection is immediately made to the broker .... Even no enabling is necessary. Just a save disconnects and reconnects

proddy, 2022-07-26:
> I will check for the loggings in the broker. Anyhow, whenever I press SAVE in the MQTT Settings Tab, the connection is immediately made to the broker .... Even no enabling is necessary. Just a save [...]

so it could be a timing issue. Without the logs its hard to trace. Are you able to compile the source code yourself as there is a DEBUG flag which will show more details on the wifi and mqtt connect status. Then you just need to upload the firmware, plug the ESP32 into a PC using a serial USB and watch the logs....

mtc716, 2022-07-26:
Compiler has been setup and the production release seems to work, but DEBUG env always give me failures with maximum memory:

Retrieving maximum program size .pio/build/debug/firmware.elf
Checking size .pio/build/debug/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.1% (used 49524 bytes from 327680 bytes)
Error: The program size (2331861 bytes) is greater than maximum allowed (2162688 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  107.8% (used 2331861 bytes from 2162688 bytes)

sorry but I am not experienced in such things and try my best to support....

proddy, 2022-07-26:
no problem. You need to make a copy of the file `pio_local.ini_example` and name it to `pio_local.ini`. Then uncomment the line
```
; my_build_flags = -DEMSESP_DEBUG -DEMSESP_USE_SERIAL
```
this is now your local config for compiling. You can use pio or VS Code to create the build.

But before doing this it would be easier if you log the MQTT broker, as I mentioned before.

sweetpants, 2022-07-27:
> I've setup remote syslog (ALL) and will watch my gateway for a while also. At the moment it's running fine

Still running fine on my side. Just restarted EMS-ESP32S as @mtc716 suggested to reproduce the issue, no problem there. EMS-ESP still connects to the MQTT broker. I'm using the HA MQTT add-on (version 6.1.2) with a username/password on port 1883 (default)

mtc716, 2022-07-28:
Just to keep you informed. The disconnects of Wifi happen due to auto-channel optimizations. The MQTT broker I use is the same like sweetpants. HA addon 6.1.2. with username/pw at port 1883.
The logging of this broker has been setup to debug and I caught some disconnects. In the HA webui the logs are rather small and I need to check if I can get them complete somewhere.  What I observed is that as long as everything works well the user of ems-esp is visible in the logging as well as the IP address. But as soon as a Wifi disconnects happened, I only see that an unknown client disconnects and even no IP or anything else is visible for some time.

proddy, 2022-07-28:
ok, then that's a good use case I can try out as well. I'll mimic a new Wifi SSID and switch the channel and see if EMS-ESP reconnects to the WiFi correctly and also to the MQTT server on your HA environment. It may be just a timing issue as the ESP32s do store wifi details in RAM for fasting connecting by design. 

Is your HA also on a Wifi network of wired? Which router are you using?

mtc716, 2022-07-28:
HA is running on raspberry via wired connection, but the wired connection is just to a Fritzbox 7490 acting as a repeater which is connected via wifi to the router which is a Fritzbox 7590....

mtc716, 2022-07-29:
I meanwhile have got the logging of an event from my broker. It is a debug log.
I did an extract  shortly before the issue started. You will see around line 293, that before everything seems to be fine and then the connection got lost and a new "unknown" connection was trying to connect.

For your information, user "ha_mqtt" is the MQTT connection from HA and user "ems-esp" is the one coming from the ESP connected to the boiler
[Logging_failure_ems-esp.txt](https://github.com/emsesp/EMS-ESP32/files/9219283/Logging_failure_ems-esp.txt)
.

proddy, 2022-07-29:
I'll take a look. I wonder if you ran your MQTT server on a fixed network (ethernet or a wifi channel that isn't changing every day) would you still see connectivity issues with EMS-ESP? You do have a very particular setup

sweetpants, 2022-08-02:
Just my 2 cents. I'm not a cpp guru, but should there not be a "MQTT disconnected" message with a possible reason after WiFi got disconnected? I only see a "MQTT connected" message in the syslog of mtc716. I assume this is after disable/enable MQTT?

mtc716, 2022-08-02:
> Just my 2 cents. I'm not a cpp guru, but should there not be a "MQTT disconnected" message with a possible reason after WiFi got disconnected? I only see a "MQTT connected" message in the syslog of mtc716. I assume this is after disable/enable MQTT?

your observation is correct. I missed also the "MQTT disconnected" at any log, by which it could be assumed, that something which has never been disconnected, why should it be automatically re-connected. And the "MQTT connected" rises as you describe it, after disabling/enabling of MQTT.

proddy, 2022-08-06:
There is a MQTT disconnected log message (including the reason), but if your network is down then there is no way to write to the log (syslog, webUI or console), which is why you're not seeing it.

mtc716, 2022-09-24:
@proddy : Just to let you that since 3.4.2.b5 no disconnect without an automatic reconnect happened. Which means for me, that the issue is somehow solved or the timing issue is gone.
The hardware remained the same at all time, even the Wifi stayed the same, but there has been a recent update in the Fritzbox firmware (beta) , stating that they fixed some Wifi stability issues.
Let's cross our fingers that it will stay perfect as it is since nearly 1,5 month....

Thanks to you

proddy, 2022-09-24:
that's great, thanks for coming back on this. Fingers crossed indeed!