emsesp/EMS-ESP32 #590: Add Buderus RC100(H) support

MichScha, 2022-08-01:
Hi, I have a combination of
- Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx
- Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II
- KM200
- RC100(H)

The device RC100(H) seems to be recognized as HP module and has a humidity sensor integrated in contrast to the RC100 device
```
Heatpump: HP Module (DeviceID:0x38)
 Received telegram type IDs:
 Fetched telegram type IDs:
 Pending telegram type IDs: 0x42B 0x47B
 Ignored telegram type IDs: 0xF7 0x01 0xBF
```

Captures telegrams
```
000+05:50:21.619 TRACE 16583: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP1(0x042B), data: 00 EA 09 22 01
000+05:50:21.619 DEBUG 16584: [emsesp] No telegram type handler found for ID 0x42B (src 0x38)
000+05:50:21.656 TRACE 16585: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP2(0x047B), data: 10 38 00 9B
000+05:50:21.656 DEBUG 16586: [emsesp] No telegram type handler found for ID 0x47B (src 0x38)

From another day
003+07:44:12.385 N 1610: [emsesp] Heatpump(0x38) -> All(0x00), ErrorMessage(0xBF), data: 38 C8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
003+07:44:16.774 N 1611: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP1(0x042B), data: 00 E9 09 1E 01
003+07:44:16.812 N 1612: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP2(0x047B), data: 0F 37 00 96
```

The identified information in the customization page shows the following two attributes which make perfectly sense
- relative air humidity (airhumidity)
- dew point temperature (dewtemperature)

What can I do to support the device?

Thanks!

MichaelDvP, 2022-08-02:
It seems there is something mixed up with heatpump module and remote thermostat.
The heatpump was introduced in [emsesp1.9](https://github.com/emsesp/EMS-ESP/issues/44) and [emsesp 2.0](https://github.com/emsesp/EMS-ESP/issues/560) and i think the heatpump controller was such a remote thermostate.
Also Norberts1 have listed device_id 0x38 as remote thermostat.

@MichScha Please post the debug information from emsesp-help page, we need the product id of the thermostat.

The remote thermostats do not have much values, the control is done by the main thermostat (RC300) and the remote only send actual temperature and, in this case, humidity and dewtemp (frst two bytes in 0x47B) . Actual room temperature is first 16 bits of telegram 0x42B, this is sent to RC300 and RC300 also publishes this temperature as roomtemp. The other bytes of these telegrams are unknown for now.

MichScha, 2022-08-02:
Jep thats why I am wondering why there is no value for those attributes. Strange is that there is no ProductID present from the device.

Debug
```
{
  "System Status": {
    "version": "3.4.1",
    "uptime": "003+21:52:02.922",
    "freemem": 136,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -86,
    "IPv4 address": "192.168.178.215/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 50558,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 2,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 30,
    "publish time thermostat": 30,
    "publish time solar": 30,
    "publish time mixer": 30,
    "publish time other": 30,
    "publish time sensor": 30,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 1,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 919106,
    "bus reads (tx)": 13747,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 5,
    "bus reads failed": 22894,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 38
  },
  "Settings": {
    "board profile": "CUSTOM",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 16,
    "tx gpio": 17,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 2,
    "bool dashboard": 2,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx",
      "device id": "0x08",
      "product id": 172,
      "version": "01.16",
      "entities": 88,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x0494 0x0495 0x048D 0x048F",
      "handlers fetched": "0x048A",
      "handlers pending": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x049F 0x04A0 0x04A2 0x04A3 0xF9 0x0497 0xF7 0x04B4 0xF6 0xEF 0x0488 0x0491 0x02CC 0x0484 0xC6 0x04 0xBB 0xC7 0x2D 0x0499 0x04A4 0x0485 0x0486 0x048B 0x0492 0x0493 0x049C 0x049D 0x05 0x01 0x03 0x0489 0xEB"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 0,
      "version": "00.00",
      "entities": 38,
      "handlers received": "0x06 0x02BA 0x02B0 0x029C 0x0472 0x02BB 0x02B1 0x029D 0x0473 0x02BC 0x02B2 0x029E 0x0474 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x02A6 0x02A7 0x02A8 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x031B 0x031E",
      "handlers ignored": "0x01 0xEF 0x0269 0xF7 0xF6 0xBF 0xC0 0xC1 0x026A 0x026B 0x026C 0x026D 0x0291 0x0292 0x0293 0x0294 0x02C3 0x02C4 0x02C5 0x02C6 0x02FF 0x0309 0x0449 0x044A 0xF9 0x044B 0x044C"
    },
    {
      "type": "Thermostat",
      "name": "generic thermostat",
      "device id": "0x17",
      "product id": 0,
      "version": "00.00",
      "entities": 0,
      "handlers pending": "0xB1 0xB0"
    },
    {
      "type": "Heatpump",
      "name": "HP Module",
      "device id": "0x38",
      "product id": 0,
      "version": "00.00",
      "entities": 0,
      "handlers pending": "0x042B 0x047B",
      "handlers ignored": "0xF7 0x01 0xBF"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 0,
      "version": "00.00",
      "entities": 0
    }
  ]
}
```

MichaelDvP, 2022-08-02:
Oh, `bus reads failed": 22894` with 38% line quality does not look ok. 
Also there is a thermostat at device 0x17, this should conflict with the master 0x10. 

First you need better tx quality to get the right infos from the devices. Check the cable connection, use thicker cable, try a different tx-mode.

MichScha, 2022-08-02:
Like I wrote before it related to the missing tx cable (just connected the rx before...) which fixed the whole connection. After a complete restart of the heating system / ems esp the device is now recognized correctly but the value of humidity seems to be incorrect. Currently it reports 28% but I know from another sensor that it should be around 60%. Could there be an error while decoding the value?
```
{
  "System Status": {
    "version": "3.4.1",
    "uptime": "000+00:01:38.617",
    "freemem": 142,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -86,
    "IPv4 address": "192.168.178.215/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 33,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 2,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 30,
    "publish time thermostat": 30,
    "publish time solar": 30,
    "publish time mixer": 30,
    "publish time other": 30,
    "publish time sensor": 30,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 433,
    "bus reads (tx)": 90,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "CUSTOM",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 16,
    "tx gpio": 17,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 2,
    "bool dashboard": 2,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx",
      "device id": "0x08",
      "product id": 172,
      "version": "01.16",
      "entities": 87,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x049F 0x04A0 0x04A3 0x04A2 0xF9"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "13.07",
      "entities": 38,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x031B 0x031E",
      "handlers ignored": "0xF9"
    },
    {
      "type": "Heatpump",
      "name": "HP Module",
      "device id": "0x38",
      "product id": 200,
      "version": "40.04",
      "entities": 2,
      "handlers fetched": "0x042B 0x047B"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.07",
      "entities": 0
    }
  ]
}
```

MichScha, 2022-08-04:
Got the first batches with smal changes in the values. On what part do I have to look now in specific? You mentioned frst two bytes in 0x047B would be the interesting one. I will try to catch more also for the dewpoint where no other values are changing.
```
Hum/dew 57%/16
001+05:23:08.197 N 26: [emsesp] Me(0x0B) <- Heatpump(0x38), HP1(0x042B), data: 19
001+05:23:08.232 N 27: [emsesp] Heatpump(0x38) -> Me(0x0B), HP1(0x042B), data: 00 F3 09 7E 01
001+05:23:08.548 N 28: [emsesp] Me(0x0B) <- Heatpump(0x38), HP2(0x047B), data: 19
001+05:23:08.574 N 29: [emsesp] Heatpump(0x38) -> Me(0x0B),d HP2(0x047B), data: 10 39 00 A2

Hum/dew 57%/16 - Roomtemp 24
000+01:36:04.684 N 2: [emsesp] Heatpump(0x38) -> Me(0x0B), HP1(0x042B), data: 00 F2 09 70 01
000+01:36:04.917 N 4: [emsesp] Heatpump(0x38) -> Me(0x0B), HP2(0x047B), data: 10 39 00 A2

Hum/dew 55%/14 - Roomtemp 22,5
000+10:45:31.963 N 5: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP1(0x042B), data: 00 E1 08 CE 01
000+10:45:32.001 N 6: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP2(0x047B), data: 0E 37 00 8D

Hum/dew 56%/14 - Roomtemp 22,5
000+10:59:30.891 N 4: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP1(0x042B), data: 00 E0 08 BD 01
000+10:59:30.929 N 5: [emsesp] Heatpump(0x38) -> Thermostat(0x10), HP2(0x047B), data: 0E 38 00 8D

Hum/dew 56%/15 - Roomtemp 24
000+14:11:05.647 N 3: [emsesp] Heatpump(0x38) -> Me(0x0B), HP1(0x042B), data: 00 EE 09 48 01
000+14:11:05.982 N 5: [emsesp] Heatpump(0x38) -> Me(0x0B), HP2(0x047B), data: 0F 38 00 9A
```

MichScha, 2022-08-04:
@MichaelDvP with your development build I get the following error
`Problem fetching device data (offset is outside the bounds of the DataView)`
and the available attributes of the devices are sometimes there sometimes away in the overview and not all values are there like before.

MichaelDvP, 2022-08-04:
Please refresh browser cache (reload page).

MichScha, 2022-08-08:
That was the problem ;-)

Here a bunch of value changes. The concrete value has been read from the display with the corresponding message for the value below.
- Hum/dew xx%/xx- Roomtemp xx
- MESSAGE FOR THAT VALUE

```
Hum/dew 57%/15 - Roomtemp 23
000+00:47:56.294 N 112: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 07 01
000+00:47:56.332 N 113: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0F 39 00 98

Hum/dew 54%/15 - Roomtemp 23
000+00:48:56.377 N 114: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 07 01
000+00:48:56.415 N 115: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0F 36 00 93
000+00:49:01.923 N 116: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 00 8F (offset 2)

Hum/dew 53%/14 - Roomtemp 23
000+00:49:56.220 N 117: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 07 01
000+00:49:56.258 N 118: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 35 00 8F

Hum/dew 52%/14 - Roomtemp 23
000+00:50:56.319 N 119: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 07 01
000+00:50:56.357 N 120: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 34 00 8B
000+00:51:56.082 N 121: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 07 01
000+00:51:56.120 N 122: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 34 00 8B

000+00:52:28.458 N 123: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 00 9D (offset 2)
000+00:52:29.377 N 124: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 00 B0 (offset 2)
000+00:52:30.999 N 125: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 00 C4 (offset 2)
000+00:52:51.564 N 126: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 00 B1 (offset 2)

Hum/dew 65%/18 - Roomtemp 23
000+00:52:56.118 N 127: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E8 09 15 01
000+00:52:56.155 N 128: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 12 41 00 B1
000+00:53:10.964 N 129: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 00 A3 (offset 2)

Hum/dew 55%/15 - Roomtemp 23
000+00:53:38.313 N 130: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 00 97 (offset 2)
000+00:53:55.968 N 131: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 0A 01
000+00:53:56.006 N 132: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0F 37 00 93

000+00:54:08.877 N 133: [emsesp] Thermostat(0x38) -> All(0x00), ErrorMessage(0xBF), data: 38 C8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Hum/dew 53%/14 - Roomtemp 23
000+00:54:55.804 N 134: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 0A 01
000+00:54:55.842 N 135: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 35 00 8E
```

How can we interpret those values or how can those bytes be converted to "normal" text?

MichaelDvP, 2022-08-08:
> How can we interpret those values or how can those bytes be converted to "normal" text?
> Hum/dew 53%/14 - Roomtemp 23
> Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E7 09 0A 01

00E7 is the temperature in 1/10°C -> 23.1°C, 09 0A 01 are unknown, check if there is a change if you change operation mode or whatever can be changed on the thermostat.
> Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 35 00 8E

0E is 14°C dewtemp, 0x35 the humidity 53%, 00 E8 unknown

I don't think we have lost values, may they needs a few minutes to show up after an update?

MichScha, 2022-08-08:
Hm maybe with your experience you can identify something here. It is only possible to
- set the heatingcircuit by number to which the module is bound to (probably the last value (01) in remoteTemp as it is for my setup the heating circuit 1) - I won´t change that because it was a huge effort to get this not that much intelligent HP in the mood of cooling
- set the correction value for roomtemperatur as temperature in 1/10°C
- read the Softwareversion
- Reset the device

```
Startpoint - The targetroomtemperatur is been defined by HP. When a targetroomtemperatur is been set on the remote thermostat it will last till the heating system change the value on itself
000+00:02:43.926 N 0: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E3 08 E1 01
000+00:02:43.963 N 1: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 33 00 86
->change of target roomtemperatur from 22 to 21,5
000+00:03:05.665 N 2: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RC300Set(0x02B9), data: 2B (offset 17)
000+00:03:44.072 N 3: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E3 08 E3 01
000+00:03:44.110 N 4: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 33 00 86
000+00:04:43.739 N 5: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E4 08 EB 01
000+00:04:43.777 N 6: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 33 00 86
->change of target roomtemperatur from 21,5 to 22
000+00:05:19.253 N 7: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RC300Set(0x02B9), data: 2C (offset 17)
00+00:05:43.926 N 9: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E5 08 F6 01
000+00:05:43.964 N 10: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 33 00 86
000+00:06:44.017 N 11: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E6 09 00 01
000+00:06:44.054 N 12: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 33 00 86
->change of target roomtemperatur from 22 to 21
000+00:07:16.278 N 14: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RC300Set(0x02B9), data: 2A (offset 17)
000+00:07:44.299 N 15: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E6 09 00 01
000+00:07:44.337 N 16: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 33 00 86

Correctionvalue - Startvalue 1.0
000+00:09:43.874 N 20: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E8 09 15 01
000+00:09:43.912 N 21: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 34 00 8F
change from 1.0 to 1.1
000+00:10:19.739 N 23: [emsesp] Thermostat(0x38) -> All(0x00), ?(0x0273), data: 0B
000+00:10:43.414 N 24: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 ED 09 3E 01
000+00:10:43.451 N 25: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 34 00 8F
000+00:11:43.417 N 26: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 EF 09 52 01
000+00:11:43.455 N 27: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 33 00 8B
000+00:11:43.417 N 26: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 EF 09 52 01
000+00:11:43.455 N 27: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 33 00 8B

change from 1.1 to 1.0
000+00:13:20.030 N 30: [emsesp] Thermostat(0x38) -> All(0x00), ?(0x0273), data: 0A
000+00:13:43.161 N 31: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 EF 09 53 01
000+00:13:43.199 N 32: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0E 33 00 8B

Softwareversion nf40.04
```

Regarding the lost values I will wait some time but it doesn´t seem to come up. Reflashing the old stable version it is present after a few seconds.

MichaelDvP, 2022-08-08:
Setting roomtemp is done in master thermostat 0x10 in hc settings: 
`Thermostat(0x38) -> Thermostat(0x10), RC300Set(0x02B9), data: 2B (offset 17)`
Funny: all known thermostats send this telegram only with 16 bytes, offset 17 is new.
How do we call this value?
The correction value is stored in the remote thermostat in new telegram 0x273. 
`Thermostat(0x38) -> All(0x00), ?(0x0273), data: 0B`
I'll add the value and command.
Version info is telegram 0x2, it's shown in the device info (i) on the right side in dashboard, or use telnet with `read 38 2`

I'll take a second look to the missing values.

MichaelDvP, 2022-08-09:
Please test with a new build on my GH. I've added the correction to remote thermostat and the remote setpoint to master. 
wwstarts2 was tagged wrong, now with dhw prefix again. hpoperatingmode was changed in #550 and should show as summermode without cooling and operatingmode with cooling.  
> set the heatingcircuit by number to which the module is bound to (probably the last value (01) in remoteTemp as it is for my setup the heating circuit 1) - I won´t change that because it was a huge effort to get this not that much intelligent HP in the mood of cooling

Could you explain exactly what you mean? Maybe with pictures. I don't have a heatpump.
>Reset the device

Which device? Emsesp an be set to factory defaults in menu, telnet (format) or via button. Boiler reset command is implemented an can reset blocking errors and mainenance messages.

MichScha, 2022-08-10:
Just flashed with the dev-build and the device is correctly identified but currently without values. It is also not appearing under customization.
[image]
I have watched the related telegrams and they have been send out so I would expect emsesp would also see them.

>wwstarts2 was tagged wrong, now with dhw prefix again. hpoperatingmode was changed in https://github.com/emsesp/EMS-ESP32/issues/550 and should show as summermode without cooling and operatingmode with cooling.

Thanks for clarification, works!

>    set the heatingcircuit by number to which the module is bound to (probably the last value (01) in remoteTemp as it is for my setup the heating circuit 1) - I won´t change that because it was a huge effort to get this not that much intelligent HP in the mood of cooling
Could you explain exactly what you mean? Maybe with pictures. I don't have a heatpump.

The RC100H does have a humidity sensor (the H) included and is able to provide temperatur and humidity for the heatpump. You need both values when you want to enable the cooling mode in the heatpump. The device has to be placed in a reference room from where it measures those to values and the system is calculation the dew-point. This is very important to get no moisture along the pipes and thats the minimal temperatur you should put in your flow temperatur. (normally you are +3 C over it to be sure). The RC100H can only be used for exactly 1 heating circuit (if you have more you need an additional one) and you have to tell the RC100H for which it is measuring the data. Thats what you need to do on the RC100H side. You also have to tell the heatpump that for heating circuit 1 there is a modul that is able to remote control and provides temperatur and humidity.
Here you can see what has to be selected on the heatpump to enable the remote device. ->RC100H has to be selected.
[image]
>RC100H installed as a remote control for the selected heating/cooling circuit. This enables the influence of the the room temperature on the outside temperature guided control.
Hopefully that makes it a little bit clearer?

> Reset the device

That was just the last option you can select on the RC100H - reset the device itself - for the sake of completeness

MichaelDvP, 2022-08-10:
> Just flashed with the dev-build and the device is correctly identified but currently without values. 

Oh, that's bad. Values and commands should be mapped to heatingcircuit. Whats the output of [http://ems-esp.local/api/thermostat](http://ems-esp.local/api/thermostat) and [http://ems-esp.local/api/system](http://ems-esp.local/api/system).
I've added some checks to hc detection, please update.
> That was just the last option you can select on the RC100H - reset the device itself - for the sake of completeness

For all you can select on the RC100H you have to log the telegrams and set this option on the thermostat to see what telegram/value is sent. But an internal function in the thermostat is maybe not accessible via ems-bus. Just try.

MichScha, 2022-08-10:
Your GHA have been skipped so no release available.
http://ems-esp.local/api/thermostat
```
{
  "datetime": "10.08.2022 11:50",
  "floordry": "off",
  "dampedoutdoortemp": 20.8,
  "floordrytemp": 0,
  "building": "medium",
  "minexttemp": -14,
  "damping": "ON",
  "wwmode": "own_prog",
  "wwcircmode": "on",
  "wwchargeduration": 60,
  "wwcharge": "OFF",
  "wwextra1": 0,
  "wwdisinfecting": "OFF",
  "wwdisinfectday": "tu",
  "wwdisinfecttime": 120,
  "hc1": {
    "seltemp": 22,
    "currtemp": 22.6,
    "mode": "manual",
    "modetype": "comfort",
    "ecotemp": 21,
    "manualtemp": 21,
    "comforttemp": 23.5,
    "summertemp": 10,
    "designtemp": 34,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 43,
    "roominfluence": 0,
    "roominflfactor": 4,
    "curroominfl": 0,
    "nofrostmode": "outdoor",
    "nofrosttemp": 5,
    "targetflowtemp": 17,
    "heatingtype": "floor",
    "summermode": "summer",
    "controlmode": "weather-compensated",
    "program": "prog_2",
    "tempautotemp": -1,
    "remoteseltemp": 22,
    "fastheatup": 0,
    "switchonoptimization": "OFF",
    "reducemode": "reduce",
    "noreducetemp": -31,
    "reducetemp": 5,
    "wwprio": "OFF"
  }
}
```
http://ems-esp.local/api/system
```
    {
      "type": "Thermostat",
      "name": "RC100H",
      "device id": "0x38",
      "product id": 200,
      "version": "40.04",
      "entities": 0,
      "handlers fetched": "0x0273",
      "handlers pending": "0x042B 0x047B",
      "handlers ignored": "0xBF"
    },
```

MichaelDvP, 2022-08-10:
> Your GHA have been skipped so no release available.

No, the last dev is one hour old.
[https://github.com/MichaelDvP/EMS-ESP32/releases](https://github.com/MichaelDvP/EMS-ESP32/releases)

MichScha, 2022-08-10:
My fault...

Now it seems to be a little bit better. One value present
[image]

MichaelDvP, 2022-08-10:
The handlers 42B and 47B are pending, are they send periodically or only on change. Should we fetch these handlers every minute?

MichScha, 2022-08-10:
They are send periodically
```
000+00:29:05.953 N 0: [emsesp] Me(0x0B) <- Thermostat(0x38), RemoteCorrection(0x0273), length: 0x19
000+00:29:05.983 N 1: [emsesp] Thermostat(0x38) -> Me(0x0B), RemoteCorrection(0x0273), data: 0A 00
000+00:29:20.427 N 2: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E5 08 F6 01
000+00:29:20.464 N 3: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 32 00 80
000+00:30:06.228 N 4: [emsesp] Me(0x0B) <- Thermostat(0x38), RemoteCorrection(0x0273), length: 0x19
000+00:30:06.257 N 5: [emsesp] Thermostat(0x38) -> Me(0x0B), RemoteCorrection(0x0273), data: 0A 00
000+00:30:20.741 N 6: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteTemp(0x042B), data: 00 E5 08 F6 01
000+00:30:21.266 N 7: [emsesp] Thermostat(0x38) -> Thermostat(0x10), RemoteHumidity(0x047B), data: 0D 32 00 80
```

MichaelDvP, 2022-08-10:
After first send of these telegrams they should be listed as `handlers received` not `handlers pending` and the values are read.

MichScha, 2022-08-10:
Hm still pending here. Should I restart EMSESP?

MichaelDvP, 2022-08-10:
Ahh, it was wrongly mapped to the dest. thermostat 0x10. Please update again.

MichScha, 2022-08-10:
Looks good! Just checked the values they are also correct
[image]

MichaelDvP, 2022-08-11:
Please check a new update, i've cleaned up the code, making it more consistent. There should be no change in displaying data.
BTW: have you checked changing the values for temp offset and temporary set temp from remote?

MichScha, 2022-08-13:
> Please check a new update, i've cleaned up the code, making it more consistent. There should be no change in displaying data.

Looks good still all values are present and correct.

> BTW: have you checked changing the values for temp offset and temporary set temp from remote?

Not yet as I never used the writing mode, will have tomorrow time to check.

MichScha, 2022-08-14:
Do we have now a doubling in attributes? I am talking about the temperatur values.

- currtemp = Current room temperature
- remotetemp

MichaelDvP, 2022-08-14:
currtemp is reported from master thermostat 0x10, remotetemp is what 0x38 sends to 0x10, it's the same value it RC100H is correctly configured as remote.

MichScha, 2022-08-16:
Just a question regarding writing commands. When changing the correction value for temperature I only can verify that value on the remote thermostat. Is it also recognizing that value because what I have understood is that the command will be sent to the master thermostat.

MichaelDvP, 2022-08-16:
The temperatursensor correction is a telegram stored in remote thermostat, we have to read and write to the RC100H.

proddy, 2022-08-21:
@MichaelDvP has some of this gone into the latest 3.4.2 dev release?

MichScha, 2022-08-21:
> The temperatursensor correction is a telegram stored in remote thermostat, we have to read and write to the RC100H.

When I change the value in ems-esp the correction value should be updated in the remote thermostat right? I will test it this afternoon. If thats working we can close this issue - works like charm.

proddy, 2022-08-24:
Does it work?

MichScha, 2022-09-01:
Okay just tested now and there is no telegram like the one you are describing. "READ Mode" is deactivated in settings and I have changed the value / watched for the telegram. There is just nothing different from the normal "update" telegrams.

MichaelDvP, 2022-09-01:
Please make sure you use the latest dev build, in doubt reinstall. Select log all and change the value. Post the log and your system info here.

MichScha, 2022-09-02:
Latest dev did the trick. Imho everything works now like expected. Closing this issue. Great work!

closed 2022-09-02 by MichScha