emsesp/EMS-ESP32 #591: Add a tip/warning on the screen when the bus stays disconnected

bbqkees, 2022-08-04:
If you have an E32 and you reset it or upload new firmware (via USB), then the flash is erased and all settings are gone.
Most users will pick the default binary from the repository, which has a default board profile 'custom'.

So the firmware update goes well, and the E32 reboots. But now it cannot find the EMS bus anymore.

Now both EMS-ESP and the Gateway hardware development have matured over the years I hardly get emails with support requests.
If so, 70% is with this issue (the other 30% either MQTT or HA config issues).

Can we add a note on the Status page pinpointing people to possible solutions?
Like 'Bus disconnected. Check wiring and board profile'.

MichaelDvP, 2022-08-04:
If the bus isn't connected the dashboard permanently shows a warning banner "Scanning for EMS devices...". 
Should we change this (in case of bus disconnect) to an error-banner like this:
[image]
Or do you think of something different?

bbqkees, 2022-08-05:
Yes like that but maybe say something like 'check settings and board profile'.
For non-technical users it is not trivial at all that the board profile may be the cause of the issue. Better point them directly to it.

closed 2022-08-21 by proddy