emsesp/EMS-ESP32 #592: [telegram] Incomplete Rx

stmb-100, 2022-08-05:
Hello
I saw some ([telegram] Incomplete Rx) in the console
I think they come from the connected ServiceKey, are they important telegrams?

Regards
Stefan

ems-esp-3.3:$ show devices
These EMS devices are currently active:

Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:04.09)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A
 Fetched telegram type IDs: 0x14 0x16 0x33 0x26
 Pending telegram type IDs: 0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA
 Ignored telegram type IDs: 0x25 0x27 0x04 0x28 0x01

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.18)
 Received telegram type IDs: 0xA3 0x06 0xA2 0x12 0x42 0x48 0x47 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60 0x38 0x39
 Fetched telegram type IDs: 0x3E 0x3D 0x3F 0xA5 0x37
 Pending telegram type IDs:
 Ignored telegram type IDs: 0x13 0xA4 0x01

Solar: SM10 (DeviceID:0x30, ProductID:73, Version:02.02)
 Received telegram type IDs: 0x97
 Fetched telegram type IDs: 0x96
 Pending telegram type IDs:
 Ignored telegram type IDs:

Gateway: KM200/MB LAN 2 (DeviceID:0x0D, ProductID:189, Version:01.10)

Controller: BC25 (DeviceID:0x09, ProductID:125, Version:01.06)

Connect: service key (DeviceID:0x0B)

000+14:40:13.712 W 727: [telegram] Incomplete Rx: 0B 08 00 18 00 05 02 42 00 00 00 00 40 40 02 60 01 A3 01 7A 00 00 FF 30 48 00 CB 00 00 00
000+14:40:25.228 W 728: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:40:57.607 W 729: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:41:17.143 W 730: [telegram] Incomplete Rx: 0B 08 00 07 00 23 01 00 00 00 00 00 00 00 00 00 00 00
000+14:41:44.065 W 731: [telegram] Incomplete Rx: 0B 08 00 34 00 3C 01 A3 01 A3 01 00 01 01 00 05 89 44 02 36 EC 00 80 00
000+14:41:45.522 W 732: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:42:19.518 W 733: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:42:43.250 W 734: [telegram] Incomplete Rx: 0B 08 00 2A 00 00 00 00 00 00 00 00 01 20 00 00 80 00 00 01 D8 80 00 02 F8 04
000+14:42:46.247 W 735: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:43:23.054 W 736: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:43:55.856 W 737: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:44:53.959 W 738: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:45:12.756 W 739: [telegram] Incomplete Rx: 0B 08 00 07 00 2B 05 00 00 00 00 00 00 00 00 00 00 00
000+14:45:15.770 W 740: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 00 00 00 00 00 00 00 00
000+14:45:35.868 W 741: [telegram] Incomplete Rx: 0B 08 00 07 00 2B 05 00 00 00 00 00 00 00 00 00 00 00
000+14:45:45.558 W 742: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 00 00 00 00 00 00 00 00
000+14:45:51.279 W 743: [telegram] Incomplete Rx: 0B 08 00 18 00 05 02 3D 00 00 00 00 40 40 02 5C 01 A2 01 72 00 00 FF 30 48 00 CB 00 00 00
000+14:45:54.061 W 744: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:46:00.830 W 745: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:46:31.664 W 746: [telegram] Incomplete Rx: 0B 08 00 2A 00 00 00 00 00 00 00 00 01 20 00 00 80 00 00 01 D3 80 00 02 F8 04
000+14:46:42.077 W 747: [telegram] Incomplete Rx: 0B 08 00 2A 00 00 00 00 00 00 00 00 01 20 00 00 80 00 00 01 D3 80 00 02 F8 04
000+14:46:51.818 W 748: [telegram] Incomplete Rx: 0B 08 00 34 00 3C 01 A2 01 A2 01 00 01 01 00 05 89 44 02 36 EC 00 80 00
000+14:47:16.694 W 749: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:47:51.484 W 750: [telegram] Incomplete Rx: 0B 08 00 34 00 3C 01 A2 01 A2 01 00 01 01 00 05 89 44 02 36 EC 00 80 00
ems-esp-3.3:$
000+14:48:27.539 W 751: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
000+14:48:50.020 W 752: [telegram] Incomplete Rx: 0B 08 00 07 00 23 05 00 00 00 01 00 00 00 00 00 00 00
ems-esp-3.3:$

MichaelDvP, 2022-08-05:
These are boiler telgrams, but with an extra 0B in front. I think this is a conflict between service key and emsesp working also as service key. Change the device-id of emsesp.

stmb-100, 2022-08-05:
the device-id of emsesp ist 0x12
[image]

MichaelDvP, 2022-08-05:
Ok, this explains the device-id 0x12 in the 07 telegram of the boiler and not in emsesp, but the boiler does not recognize the 0x0B service-key. (data of 07: 23 05 00 00 00 01, -> recognized devices 08, 09, 0D, 12, 30) 
Is this a normal service-key? It looks like 0x0B answers a poll without break.
Could you attach a full log of about a minute?

stmb-100, 2022-08-05:
watch raw in the console ?

stmb-100, 2022-08-05:
[EMSESPLog.txt](https://github.com/emsesp/EMS-ESP32/files/9267012/EMSESPLog.txt)

MichaelDvP, 2022-08-05:
> watch raw in the console ?

In webpage, system->system log, level all, after a minute klick `export`, drop file here (as attachment). 
log all is better than watch to see also the debug messages.

stmb-100, 2022-08-05:
[EMSESPSystemLog.txt](https://github.com/emsesp/EMS-ESP32/files/9267055/EMSESPSystemLog.txt)

MichaelDvP, 2022-08-05:
Seems the service key is sending nothing on the bus, except sometimes it's id (poll reply) without a break. and after master-timeout. This gives the extra 0x0B in front of other telegrams and the boiler does not recognize the service key. The broken telegrams are repeated, so it does not matter.
The thermostat is repeatly trying to get version info from alarm module 0x12 (emsesp) to configure it. But this also does not matter. I think with id 0x0A for emsesp it will be less traffic.

stmb-100, 2022-08-05:
i set emsesp = 0x0A

[EMSESPSystemLog_2.txt](https://github.com/emsesp/EMS-ESP32/files/9267708/EMSESPSystemLog_2.txt)

MichaelDvP, 2022-08-05:
Yes, as expected the version requests are gone. 
To remove the rx errors starting with 0B i think you have to live with it, or remove the service key.

stmb-100, 2022-08-05:
OK thank you
R
Stefan

suncarver, 2022-08-06:
Hi, 

it worked in the past for > 1 year with earlier versions on ESP32 with my Buderus. 
Then "suddenly" I got no MQTT messages any more and I found Incomplete Rx in the log.
Starting new with EMS-ESP 3.4.1 update on ESP32, I also get Incomplete Rx.

Now initial scanning already hangs with endless "Loading.." and a lot of Incomplete Rx: 

000+00:01:07.514 W 524: [telegram] Incomplete Rx: 89 00 09
000+00:01:07.545 W 525: [telegram] Incomplete Rx: 95 38 15
000+00:01:07.614 W 526: [telegram] Incomplete Rx: DD 00 A9 D5
000+00:01:07.718 W 527: [telegram] Incomplete Rx: 8C 55 89 00 09 55
000+00:01:07.789 W 528: [telegram] Incomplete Rx: 5F 00 89
000+00:01:08.039 W 529: [telegram] Incomplete Rx: 0D 86 D0 00 B5
000+00:01:08.087 W 530: [telegram] Incomplete Rx: AF 02 AB
000+00:01:08.216 W 531: [telegram] Incomplete Rx: BD 00 29 44 9D 00 0D C3 FD 00 DD
000+00:01:08.295 W 532: [telegram] Incomplete Rx: 92 00 CD 85 10 55
000+00:01:08.389 W 533: [telegram] Incomplete Rx: 09 2A DD 00 09 14 D7 3E
000+00:01:08.490 W 534: [telegram] Incomplete Rx: ED F5 95 00 89 E3 09 00
000+00:01:08.538 W 535: [telegram] Incomplete Rx: 19 10 55
000+00:01:08.738 W 536: [telegram] Incomplete Rx: 89 15 09 00 9D 3E 09 00 B5 E4 3F 50 90 15 3E 00 85 FC 09 55
000+00:01:08.839 W 537: [telegram] Incomplete Rx: 40 C9 55 7D A0 9B
000+00:01:08.935 W 538: [telegram] Incomplete Rx: 84 00 97
000+00:01:09.009 W 539: [telegram] Incomplete Rx: DC 32 15 40 9E 55
000+00:01:09.058 W 540: [telegram] Incomplete Rx: 89 00 AD
000+00:01:09.109 W 541: [telegram] Incomplete Rx: 0D 3A E3 C0
000+00:01:09.260 W 542: [telegram] Incomplete Rx: 90 00 CF 55 FF FD 55 09 00
000+00:01:09.336 W 543: [telegram] Incomplete Rx: E0 C2 00 A6 AA
000+00:01:09.387 W 544: [telegram] Incomplete Rx: 09 00 FF DF
000+00:01:09.460 W 545: [telegram] Incomplete Rx: DD 00 5D 68 89 00
000+00:01:09.607 W 546: [telegram] Incomplete Rx: 90 3E C8 05 B6 00 9D 54
000+00:01:09.681 W 547: [telegram] Incomplete Rx: DF C7 89 01 D5 FC 89
000+00:01:09.809 W 548: [telegram] Incomplete Rx: 84 00 D8 38 F0 A8 89 00 8B 00 8D E3 09 00
000+00:01:09.910 W 549: [telegram] Incomplete Rx: 09 8D 41 DD 00 09 00 D7 C6 89 60
000+00:01:09.957 W 550: [telegram] Incomplete Rx: 03 E6 15
000+00:01:10.010 W 551: [telegram] Incomplete Rx: AA 50 89 10 E1 00
000+00:01:10.129 W 552: [telegram] Incomplete Rx: D5 FC D6 55 FF 89 10
000+00:01:10.281 W 553: [telegram] Incomplete Rx: 09 00 1D E0 B8 01 D9 00 19 CD
000+00:01:10.406 W 554: [telegram] Incomplete Rx: 55 09 00 EB 05 09 00 90 55 FF
000+00:01:10.478 W 555: [telegram] Incomplete Rx: BD 55 09 00 FE F5
000+00:01:10.510 W 556: [telegram] Incomplete Rx: 45 FE 00
000+00:01:10.559 W 557: [telegram] Incomplete Rx: BE 09 00
000+00:01:10.679 W 558: [telegram] Incomplete Rx: 4B C7 89 50 09 02
000+00:01:10.878 W 559: [telegram] Incomplete Rx: E3 90 00 D5 55 AB 64 A9 C5 05 00 A9 A5 3F 00 C9 18 FF F3 46
000+00:01:11.129 W 560: [telegram] Incomplete Rx: D8 A0 B8 00 35 40 D7 00 89 52 5D 55 CC 80 8D 00 09 15 8D 55 FF 99 40 9B F5 DD

Any idea or recommendation what I could try?

Thanks in advance, Michael

MichaelDvP, 2022-08-06:
Looks like garbage and not connected rx  line. Check settings, board profile and bus cables. If you use service jack, check the plugs pushed in completly.

suncarver, 2022-08-06:
Thanks Michael for the quick response.

Checked bus cable. I don't use service jack. I can't check settings and board profile in the UI, because I also see only permanent "Loading ..." in settings.
When I select menu Dashboard or Settings, a red info box is shown in the lower left corner: "Problem fetching: Unexpected token < in JSON at position 0". The box disapperas after about 5 seconds, but still permanently "Loading .."

MichaelDvP, 2022-08-06:
Have you tried Ctrl-R (reload page to update browser-cache)?

suncarver, 2022-08-06:
Tried new Ctrl-R, F5, new browser window..

MichaelDvP, 2022-08-06:
I think you should check all parts from start, disconnect all, than:
- check powersupply
- check esp32 powered, but not connected to interface: if not working, reflash, if working check all settings.
- connect interface, but not ems-bus, check power on interface board
- connect bus.

suncarver, 2022-08-06:
Now I have pure ESP32 directly connected to  PCs USB - neither EMS-ESP nor 1 wire connected.
VSC - platformio: loaded EMSP-ESP 3.4.1, compiled and uploaded to ESP32.
Started with access point, set my WiFi settings.
Rebooted and connected with my WiFi in new Browser Window
Log so far: 

000+00:00:00.000 I 0: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
000+00:00:00.000 I 1: [emsesp] Loading board profile CUSTOM
000+00:00:00.000 I 2: [emsesp] Starting EMS-ESP version 3.4.1 (hostname: ems-esp)
000+00:00:00.000 I 3: [dallassensor] Starting Dallas sensor service
000+00:00:00.000 I 4: [analogsensor] Starting Analog sensor service
000+00:00:00.000 I 5: [emsesp] Loaded EMS device library (100 records)
000+00:00:00.000 I 6: [emsesp] Starting Access Point with captive portal on 192.168.4.1
000+00:00:04.544 I 7: [emsesp] WiFi connected with IP=192.168.178.54, hostname=ems-esp
000+00:00:04.554 I 8: [emsesp] mDNS responder service started
000+00:00:10.000 I 9: [emsesp] Stopping Access Point
==> no more logs so far - now without "Incomplete Rx" in the log any more.

I also tried a factory reset and restart.
-> still endless "loading" when I try to open Settings menu or Dashboard.

Any idea?

MichaelDvP, 2022-08-06:
After factory reset you are in AP mode only again. Connect to AP and reconfigure.

suncarver, 2022-08-07:
Yes, I'm aware of this, thanks for your patience and hints MIchael. I rebuilt ESP32 EMS-ESP V3.4.1 from scratch with PlatformIO.
Now I can see Settings Menu with ESP32 while EMS is not connected ("endless loading" problem is gone.)
But again I see only Incomplete Rx in the log when I connect EMS cables - no scan results, no devices detected. EMS Disconnected.
 
2022-08-07 20:44:26.200 W 5475: [telegram] Incomplete Rx: BF 40 D9 00 59 81 89 15 49 00 B2
2022-08-07 20:44:26.251 W 5476: [telegram] Incomplete Rx: 10 F1 9D 00
2022-08-07 20:44:26.349 W 5477: [telegram] Incomplete Rx: 99 05 09 00 89 54 FE 00 89
2022-08-07 20:44:26.849 W 5478: [telegram] Incomplete Rx: 0D F0 89 00 49 EC DF 22 FF 92 1D 18 FC E4 CC 44 8D 10 0D 1A FF 99 9B
2022-08-07 20:44:27.072 W 5479: [telegram] Incomplete Rx: CF 95 09 00 D8 92 92 12 FE 49 E9 1C EF 5F AB 01 09 05 F7 82 CC 5F
2022-08-07 20:44:27.220 W 5480: [telegram] Incomplete Rx: C0 00 89 00 AB E3 BB 92 1B 12 BB 82 D4
2022-08-07 20:44:27.323 W 5481: [telegram] Incomplete Rx: FF 44 8B 00 E6 8F 12 49 89 31 79 80
2022-08-07 20:44:27.469 W 5482: [telegram] Incomplete Rx: 55 7F C2 00 C9 92 9B 26 FF CE E6 92 12 9D
2022-08-07 20:44:27.681 W 5483: [telegram] Incomplete Rx: E4 FF DD 00 AB 6E F6 22 AD C4 F2
2022-08-07 20:44:27.843 W 5484: [telegram] Incomplete Rx: 54 FF 10 00 CE E6 AB 48 92 66 FF D7 24 89
2022-08-07 20:44:27.969 W 5485: [telegram] Incomplete Rx: 2B 22 D9 00 85 86 F2 24 56 C6 AB 44
2022-08-07 20:44:28.092 W 5486: [telegram] Incomplete Rx: 20 FF 8B 00 00 40 FF FF 92 9B 24 FF FD
2022-08-07 20:44:28.292 W 5487: [telegram] Incomplete Rx: 91 44 91 00 FD 32 FF 22 1E FC E2 B2 12 33 38 46 FA 11
2022-08-07 20:44:28.541 W 5488: [telegram] Incomplete Rx: E1 00 89 00 49 3E 93 92 FE C6 19 02 D5 E0 89 AA 09 80 90 54 B5 

I started Buderus again - still Incomplete Rx.

Are there known issues with Buderus/Bosch EMS+ Service key blocking?

MichaelDvP, 2022-08-08:
Still garbage from a floating rx line. I don't know how youre system is connected and configured, so it's all just a shot in the dark.
Is GND, 3.3V, rx, tx connected proper with low impedance? rx, tx not exchanged? rx set to the right and a valid GPIO?
Do you get the rx flooding only when connecting the ems-bus or also ith ems disconnected but esp-GW connected?
...

suncarver, 2022-08-08:
"Is GND, 3.3V, rx, tx connected proper with low impedance?" Yes, cross checked several times. I can't judge on impedance.
I changed rx,tx for testing: "EMS disconnected" - no incomplete Rx logs, ESP shows one periodical LED signal.
GPIOs exactly as shown in BBQs V3 picture to ESP32 Mini.
EMS disconnected but esp-GW connected: no rx flooding in log.
I think I'll check how I can test the communication on the 2 EMS wires. Can I verify EMS in listening mode? I need to find out  how I can do it with EMS 15V.

MichaelDvP, 2022-08-09:
I'm confused now. Blinking led means bus disconnected, e.g no or only garbage telegrams are received.
If it wasn't blinking before, there must be some legal telegrams from the bus. But your logs show only garbage?

You have to desctribe more precise your complete system, your complete configuration, complete logs, ...

A working bus with some or many garbage mixed in points to a defective power supply, But this as already my first advice.

suncarver, 2022-08-09:
OK Michael, I have to apologize. I tried 2 power supplies. Obviously both too weak. Using a better one now brings me back to a working solution.  So thanks a lot! I expected other effects due to a weak power supply, but not the effect of insufficient power for the serial modul. 
Thank you so much for your help and patience - we can close the issue.

proddy, 2022-08-09:
@MichaelDvP @BBQKees what do you think should be the min power required for the circuit, in mV, so I can add to https://emsesp.github.io/docs/#/Troubleshooting?id=many-rx-errors

MichaelDvP, 2022-08-09:
It's not the power, esp32 takes in peak less 200mA, cheapest usb loaders deliver more than 500mA. The esp32 takes 20mA in normal operation with wifi-tx peaks of 180mA. The usb chargers can not regulate the power that fast and react with a voltage drop, that causes a uart startbit. For charging a good regulater is not required. Raspberrys have the same issue, a good power supply is that from pi3.
Reducing tx-power in esp32 also reduces current peaks and could be first aid for increasing rx issued. 
in this case i misunderstood, thought it was no valid rx at all.

proddy, 2022-08-09:
yes, sorry I did mean mA. What kind of text shall we add to the wiki?

MichaelDvP, 2022-08-10:
Maybe only a simple advice: "Try different usb power supplies"

bbqkees, 2022-08-10:
The interface board itself draws like 3mA. That's not the problem. The problem is indeed usually in USB plug phone chargers.
They are designed to slowly charge your phone not to provide instantaneous power to an ESP board. The cheap ones also have little to no filtering so they output signal is very noisy.
Most of those chargers will work fine though but sometimes you have a crappy one that can't handle the current surges well.

suncarver, 2022-08-10:
very good insights from you experts, I appreciate this. I think a Wiki hint on this topic as a guidance for non hardware experts like me would help a lot to avoid efforts on wrong root cause analysis. Point is - what are the power supply capability to be considered when buying a power supply. Obviously ist not just 5V, relying USB standard and a mA value on the label. Its more and its usually not written in the sales information. Also its hard to "measure" and verify with limited laboratory equipment. So, as an idea, maybe a forum member maintained  list with power supply links which work?  Maybe hosted on Kees sites with link from EMS-ESP?

bbqkees, 2022-08-10:
Poor power supply is usually not an issue. You are the first in quite some time.
Usually those that come with a decent brand phone will do fine.

closed 2023-02-19 by proddy