emsesp/EMS-ESP32 #593: missing shower_data

stmb-100, 2022-08-05:
Hi 
I get no Shower_Data in my IO-Broker. This is how I configured it.
[image]

That's all I see in the log file : [mqtt] Publishing topic ems-esp-33/shower_active (#62442, retain=1, retry=1, size=5, pid=1)
[image]
This value has not changed for a few days, even though people shower several times a day. I can't say if it's ever been different.
How to narrow down the problem

Regards
Stefan

These EMS devices are currently active:

Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:04.09)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A
 Fetched telegram type IDs: 0x14 0x16 0x33 0x26
 Pending telegram type IDs: 0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA
 Ignored telegram type IDs: 0x25 0x27 0x04 0x28 0x01

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.18)
 Received telegram type IDs: 0xA3 0x06 0xA2 0x12 0x42 0x48 0x47 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60 0x38 0x39
 Fetched telegram type IDs: 0x3E 0x3D 0x3F 0xA5 0x37
 Pending telegram type IDs:
 Ignored telegram type IDs: 0x13 0xA4 0x01

Solar: SM10 (DeviceID:0x30, ProductID:73, Version:02.02)
 Received telegram type IDs: 0x97
 Fetched telegram type IDs: 0x96
 Pending telegram type IDs:
 Ignored telegram type IDs:

Gateway: KM200/MB LAN 2 (DeviceID:0x0D, ProductID:189, Version:01.10)
Controller: BC25 (DeviceID:0x09, ProductID:125, Version:01.06)
Connect: service key (DeviceID:0x0B)

MichaelDvP, 2022-08-05:
Shower detection only works with flow-systems, dhw buffer-systems does not have a flow sensor. Check in boiler data what type your system is: My boiler shows:
[image]

stmb-100, 2022-08-05:
Hi my dhwType is flow

[image]

MichaelDvP, 2022-08-05:
Do you see nonzero values in `dhw current tap water flow (wwcurflow)` when tap water is used?
What is `tapwater_active` showing?

stmb-100, 2022-08-05:
[image]
[image]

[image]

stmb-100, 2022-08-05:
[image]
[image]

MichaelDvP, 2022-08-05:
Is it really a flow system? How do you manage flow with solar? Solar needs a buffer and in case of no sun the buffer is charged by boiler? Or do the solar heated water flow though a boiler heatingblock and burner starts if it is colder than required? `tapwateractive` and `shower-active` are ony triggered when burner is started in dhw mode.

stmb-100, 2022-08-05:
Is it really a flow system? I don't know
in case of no sun the buffer is charged by boiler? I hope not
Or do the solar heated water flow though a boiler heatingblock and burner starts if it is colder than required?  It could be but I'm not sure how to find out

stmb-100, 2022-08-05:
Hi
GERMAN : (Könnte hier ein Zusammenhang mit dem Fehler A01/811 bestehen, den ich bei aktiver thermischer desinfektion bekomme)

Could there be a connection with error A01/811, which I get when thermal disinfection is active ?

GERMAN:
Diese Meldung wird erzeugt, wenn 3 Stunde nach Start der thermischen Desinfektion die Solltemperatur für die thermische Desinfektion nicht erreicht wurde.
AUSWIRKUNG : Nach der Fehlermeldung wird die thermische Desinfektion abgebrochen

This message is generated if the target temperature for thermal disinfection has not been reached 3 hours after the start of thermal disinfection.
EFFECT: After the error message, the thermal disinfection is aborted

MichaelDvP, 2022-08-05:
I really don't know. 
The dhw-system is fixed in boiler, so flow type is right. I have no experience with flow systems, but imo it needs a flow sensor and `wwcurrflow` should have a value if water is running. 
Maybe @proddy can help, afaik his system is also flow type.

proddy, 2022-08-06:
yes I have a flow type too. When the hot water is running see if the "current tap water flow" value changes.

stmb-100, 2022-08-09:
[image]
Hi
I don't think that the value changes during hot water flow, why it changes on 3.8. I don't know.

proddy, 2022-08-10:
3.8?

stmb-100, 2022-08-10:
2022-08-03 09:33:20 last Status change

proddy, 2022-11-11:
is this still an issue?

closed 2023-02-19 by proddy