emsesp/EMS-ESP32 #594: Make log "Operation type" more self-explanatory.

mvjt, 2022-08-07:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
A more self-explanatory log output.

**Describe alternatives you've considered**
In the log output, read and write operations are indicated using -> and <-. I would suggest using something more self-explanatory or at least add an initial line to the log output explaining this. One suggestion is to use [R] for read and [W] for write. It would also make it possible to support more types of operation than two, for example [C] for Command.

proddy, 2022-08-07:
we've had this discussion before. See https://github.com/emsesp/EMS-ESP32/issues/491

I agree it would be good to show if it's a read or write request. 

Note you only see the direction of the EMS traffic with watch. Maybe a good option to add this to the Web UI?

MichaelDvP, 2022-08-07:
Yes it is continuing #491, but it seems there are more people not intuitive getting the logic of the arrows. 
It's only a text change [here](https://github.com/emsesp/EMS-ESP32/blob/8d3bcde8d6723d1ab75dfb4504ccef38c0ffa0bb/src/emsesp.cpp#L686-L692). For me it's important to see the read/write on first look, it does not matter if this is an arrow, a char or text. 

@mvjt There are only read and write operations, in raw telegram read is msb of destination set, write msb in dst cleared.
A raw read request is `0B 88 33 00 20` as `Me(0x0b) <- Boiler(0x08) UBAParameterWW(0x33) length 32 (offset 0)`
a write  `0B 08 33 00 20` as `Me(0x0b) -> Boiler(0x08) UBAParameterWW(0x33) data 32 (offset 0)`

@proddy I think the watch is mostly usefull in combination with other commands in the telnet. Adding it to web system log will be confusion for users, a `loglevel all` is more to read but easier to explain to people not familiar with the telnet console. Keep the system log simple.

mvjt, 2022-08-07:
Thanks for the explanation. I understand. For me the arrow between two "addresses" meant direction of comms. But if you prefer to keep it, you can also add an explanatory row when enabling watch. If I was the second person pointing this out maybe the issue is with me :-)

proddy, 2022-08-07:
Usability is always high on the list so happy to change it to make it more readable. The syntax goes back years to version 2 and it hasn't changed since.

proddy, 2022-08-24:
Read = -<
Write = ->

how should we handle broadcast messages like 
`Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 01 06 00 00 80 00 00 80 00 80 00 80 00 00`

I think this needs more thinking

mvjt, 2022-08-24:
Why not

-> and <- are directions
(R) and (W) are read and write

`->` (broadcast from Boiler to all)
`-> (R) `(Read request from Thermostat to Boiler
`-> (W)` (Write command from Thermostat to Boiler)

proddy, 2022-08-24:
I like it. I'll add to 3.5.0

mvjt, 2022-08-25:
Sounds good. I had one more thought. If you would import the log file into lets say Excel to filter, we might need to add (B) for broadcast so we have equal amount of columns.  Or simply if one want to find all broadcast messages they can grep `-> (B)`

MichaelDvP, 2022-08-25:
Just a remark: The notation is always src .. dest...  giving the direction, a direction arrow is redundant and always points to the right. But it's pretty telegram notation and if a few extra chars make it better readable.

MichaelDvP, 2022-09-12:
I've tested a bit and think two separators `-> (B)` (takes two coulums in spreadsheets) are to much, can we merge character and arrow in a single separator `-B->`: 
`Boiler(0x08) -B-> All(0x000) ...`
`Me(0x0B) -R-> Boiler(0x08) ...`
`Boiler(0x08) -W-> Me(0x0B) ...`
or with small read/write/broadcast:
`Boiler(0x08) -b-> All(0x000) ...`
`Me(0x0B) -r-> Boiler(0x08) ...`
`Boiler(0x08) -w-> Me(0x0B) ...`

If this is ok, i'll add to v3.5..0

proddy, 2022-09-12:
fine with me!

mvjt, 2022-09-17:
If it is good with you It is good with me :-)

MichaelDvP, 2022-09-17:
Oh, now i see that i've added it only to my GH, i'll make a PR now.

proddy, 2022-09-19:
This is in 3.5.0

closed 2022-09-19 by proddy