emsesp/EMS-ESP32 #598: Heatpump: HP Module - relative air humidity is half the value it should be

beSt0815, 2022-08-08:
Hi,
the EMS shows the relative humidity of my CR 10H Thermostat.
Unfortunately it shows exactly half the value it should be.
The HPC400 shows the correct value.
[image]
[image]
Thank you.

MichaelDvP, 2022-08-08:
So this is also a thermostat (RC10H) and not a hp-module, please attach the debug info from help page of ems-esp. Then i can see the telegram-ids, product id and device id to map the values right.

beSt0815, 2022-08-08:
Device Details

    Type:    Heatpump
    Name:    HP Module
    Brand
    Device ID:    0x38
    Product ID:    200
    Version:    40.07

MichScha, 2022-08-08:
@MichaelDvP maybe same calculation like for rc100h #590 
I have appended a few minutes ago new debug values.

MichaelDvP, 2022-08-08:
Yes, it's the same product_id and same device_id working as remote thermostat and should work in my [dev-branch](https://github.com/MichaelDvP/EMS-ESP32/releases) as thermostat, i've already made a PR.

proddy, 2022-08-21:
@beSt0815 did Michael's firmware work?

beSt0815, 2022-08-21:
Yes, the humidity is correct now.

proddy, 2022-08-21:
Great. Can you close this issue?

closed 2022-08-21 by beSt0815