emsesp/EMS-ESP32 #599: Low quality of EMS Reads (Tx), Junkers Cerapur ZSB 14

0Bu, 2022-08-10:
**Bug description**
The quality of the EMS Reads (Tx) is about 85% with Tx mode HT3, EMS and EMS+ even lower.
EMS-ESP version: v3.4.1

**Expected behavior**
The quality of 100% 🤓

**Device information**
Junkers Cerapur ZSB 14-3 A 23 S0000
```
"System Status": {
    "version": "3.4.1",
    "uptime": "000+01:35:48.138",
    "freemem": 155,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
…
  "Bus Status": {
    "bus status": "connected, tx issues - try a different Tx Mode",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 3580,
    "bus reads (tx)": 419,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 77,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 85
  },
  "Settings": {
    "board profile": "S32",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": false
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "18.01",
      "entities": 60,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "FW100",
      "device id": "0x10",
      "product id": 105,
      "version": "13.03",
      "entities": 4,
      "handlers received": "0x06",
      "handlers fetched": "0x016F",
      "handlers pending": "0xA3 0xA2 0x12 0x0170 0x0171 0x0172 0x0165 0x0166 0x0167 0x0168 0xBB 0x23 0x0123",
      "handlers ignored": "0xBE"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "18.01",
      "entities": 0
    }
  ]
```

**Additional context**
[log.txt](https://github.com/emsesp/EMS-ESP32/files/9303740/log.txt)

proddy, 2022-08-11:
most likely it's not a software bug. Can you give details of the hardware/gateway/circuit you are using, how and where it's installed. Also did you experience similar drop outs in previous EMS-ESP versions?

MichaelDvP, 2022-08-11:
Looks like the thermostat is not answering to 0x16F requests. Also there was a 0xBE received from thermostat, which is an actual errorstate. The other devices seems to work proper, there is something wrong with the thermostat.
Please send a full log (loglevel all) of more than a minute.

This thermostat is implemented since a long time [https://github.com/emsesp/EMS-ESP/pull/128](https://github.com/emsesp/EMS-ESP/pull/128), the telegrams should be right.

0Bu, 2022-08-11:
@proddy thank you for updating the issue to "help wanted" 👍
- Hardware
  - Boiler: Junkers Cerapur ZSB 14-3 A 23 S0000
  - Thermostat: Junkers FW100 (wired directly to the boiler)
- Gateway
  - [Gateway S32 (Standard WiFi Edition)](https://bbqkees-electronics.nl/wiki/gateway/connecting-a-gateway-s32.html#connecting-a-gateway-s32-esp32-standard-edition)
- Installation
  - The gateway is wired over the [BB Therminal](https://bbqkees-electronics.nl/wiki/gateway/connecting-to-specific-boilers-and-heatpumps.html#cerapur) by cable to the boiler
- Previous EMS-ESP versions?
  - This is my first gateway, what a wonderful project!

@MichaelDvP
[log all.txt](https://github.com/emsesp/EMS-ESP32/files/9307130/log.all.txt)

MichaelDvP, 2022-08-11:
Ok, seems the thermostat does not reply to the read of higher offsets. This causes the tx-read errors and emsesp creates a dummy telegram with wrong offset. Because of this the set-telegram 0x165 is not fetched.

In first step i've fixed the offset. [This dev-build](https://github.com/MichaelDvP/EMS-ESP32/releases) should show up the setpoints and fetch the settings. But will not fix the counting of tx-read-errors. I'm not sure how to handle this strange behaviour of that thermostat. 
@proddy Any idea? Ignore missing replys if read with offset? Maybe also no retries for higher offsets? Fetch higher offsets with nebyte overlap to first, than there should be a single byte reply?

proddy, 2022-08-11:
@MichaelDvP all good options. A retry with higher offset would be nice, but that's quite a bit of extra code. Personally I prefer just ignoring it and mentioning it in the trace or debug log, only because I'm too busy with other stuff at the moment

0Bu, 2022-08-11:
I have updated the FW to [v3.4.2b5](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest), it looks like more read tx fails are coming now
[log.txt](https://github.com/emsesp/EMS-ESP32/files/9307904/log.txt)

MichaelDvP, 2022-08-11:
It's only few more errors at startup, when emsesp tries to fetch the other telegrams (which is workking now).
I'm supprised that there is no 0x165 setting. Could you check in terminal with the command `read 10 179` if there is an answer, maybe this thermostat uses the old junkers command.
I've pushed a update, not counting errors for higher offsets, log as debug-message, and not sending a dummy for this.
@proddy retry is already implemented.

0Bu, 2022-08-11:
Thank you for the explanation! `read 10 179` returns this

[image]

MichaelDvP, 2022-08-11:
Sad, so this thermostat has no, or a complete unknown, setting telegram. If you are missing a setting in emsesp for the thermostat, log all or use `watch 10` and change the setting on the thermostat. Then we can check if there is a setable value somewhere.
Most of the tx-read-errors should be gone now, some after startup, but not contious increasing later, right?

MichaelDvP, 2022-08-11:
Oh, i see we had this thermostat before in #468 with software 13.05 and telegram 0x165 implemented. Seems your thermostat is a bit older (Ver 13.03) and does not have this telegram.

0Bu, 2022-08-11:
here what i got from `watch 10` - the thermostat is the old one, it's from 2007 or early 2008
```
┌──────────────────────────────────────┐
│ EMS-ESP version 3.4.2b5              │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└──────────────────────────────────────┘

ems-esp:$ watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x10
000+02:39:00.824 N 0: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x19
000+02:39:00.905 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 01 A8 F3 34 04 F5 9E 00 5E 1D 00 71 00 00 00 00 00 00 20 9E 00
000+02:39:01.073 N 2: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:39:01.442 N 3: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:39:01.822 N 4: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:39:02.191 N 5: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:39:02.455 E 6: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 19 20 00 6F
000+02:39:02.455 N 7: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: <empty> (offset 25)
000+02:39:03.114 N 8: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 08 0E 0B 2E 0B 04 01
000+02:39:03.475 N 9: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), length: 0x0C
000+02:39:03.503 N 10: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: 00 00 64 00 00 F6 03 01 03 64 0A 04
000+02:39:03.653 N 11: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 01 A8 F3 34 0A
000+02:39:34.458 N 12: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), length: 0x0A
000+02:39:34.487 N 13: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 01 00 00 00 00 00 00 00
000+02:40:00.919 N 14: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x19
000+02:40:00.998 N 15: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 01 A8 F3 34 04 F5 9E 00 5E 1D 00 C9 00 00 00 00 00 00 20 41 00
000+02:40:01.137 N 16: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:40:01.543 N 17: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:40:01.917 N 18: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:40:02.323 N 19: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:40:02.620 E 20: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 19 20 00 6F
000+02:40:02.620 N 21: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: <empty> (offset 25)
000+02:40:03.092 N 22: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 08 0E 0B 2F 0B 04 01
000+02:40:03.420 N 23: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), length: 0x0C
000+02:40:03.447 N 24: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: 00 00 64 00 00 F6 03 01 03 64 0A 04
000+02:40:03.631 N 25: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x016F), data: 03 02 00 D2 01 A8 F3 34 0A
000+02:40:31.698 N 26: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 2B (offset 1)
000+02:40:31.701 N 27: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00
000+02:40:31.751 N 28: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+02:40:31.788 N 29: [emsesp] Thermostat(0x10) -> Boiler(0x08), JunkersSetMixer(0x23), data: 00 00 00
000+02:40:31.816 N 30: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
000+02:40:31.858 N 31: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xBE), data: 10 00 00 00 00
000+02:40:35.689 N 32: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), length: 0x0A
000+02:40:35.710 N 33: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 01 00 00 00 00 00 00 00
000+02:41:00.738 N 34: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x19
000+02:41:00.820 N 35: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: 03 02 00 D2 01 A8 F3 34 04 F5 9E 00 5E 1D 00 C9 00 00 00 00 00 00 20 C9 00
000+02:41:00.957 N 36: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:41:01.362 N 37: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:41:01.737 N 38: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:41:02.111 N 39: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x016F), length: 0x20 (offset 25)
000+02:41:02.413 E 40: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 19 20 00 6F
000+02:41:02.413 N 41: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: <empty> (offset 25)
000+02:41:02.915 N 42: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 08 0E 0B 30 0B 04 01
000+02:41:03.267 N 43: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), length: 0x0C
[... cut ...]
000+02:45:03.056 E 106: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 19 20 00 6F
000+02:45:03.056 N 107: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x016F), data: <empty> (offset 25)
000+02:45:03.089 N 108: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 16 08 0E 0B 34 0B 04 01
000+02:45:03.450 N 109: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), length: 0x0C
000+02:45:03.480 N 110: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: 00 00 64 00 00 F6 03 01 03 64 0A 04
```

0Bu, 2022-08-11:
there are no errors in the log, that is true, the quality is about 80%

[image]