emsesp/EMS-ESP32 #600: read and write inputs of heatpump

beSt0815, 2022-08-11:
To control the heatpumps functions (e.g. Block electrical heater, block DHW, only cooling with PV power...) it would be nice to set the inputs via EMS.
I logged the traffic and switched them by using a wire:

i3 on: 
`Boiler(0x08) -> All(0x00), ?(0x04A2), data: 02 01 01 00 01 00`
i3 off:
`Boiler(0x08) -> All(0x00), ?(0x04A2), data: 02 01 01 00 00 00`

i2 and i3 off: `Boiler(0x08) -> All(0x00), ?(0x0488), data: 08 00 00 00 00 00 02 00`
i2 on, i3 off: `Boiler(0x08) -> All(0x00), ?(0x0488), data: 08 40 00 00 00 00 02 00`
i2 off, i3 on: `Boiler(0x08) -> All(0x00), ?(0x0488), data: A8 40 00 00 00 00 02 00`

I didn't find a telegram for I1 and I4 at 0x0488 or 0x04A2.

I changed logic in my Controller for input 4 from high to low and vice versa:

Logic of input 4 set to low (means true)
```
Thermostat(0x10) -> Boiler(0x08), ?(0x0486), data: 01 (offset 42)
Boiler(0x08) -> All(0x00), ?(0x0486), data: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Boiler(0x08) -> All(0x00), ?(0x0486), data: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 01 00 00 00 00 00 (offset 25)
Boiler(0x08) -> All(0x00), ?(0x0486), data: 00 00 (offset 51)
```
Logic of input 4 set to high (means false)
```
Thermostat(0x10) -> Boiler(0x08), ?(0x0486), data: 00 (offset 42)
Boiler(0x08) -> All(0x00), ?(0x0486), data: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Boiler(0x08) -> All(0x00), ?(0x0486), data: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 (offset 25)
Boiler(0x08) -> All(0x00), ?(0x0486), data: 00 00 (offset 51)
```

By changing logic in the controller you don't need to open the HP and connect wires. 
You can change from high to low logic, so its inverted.

I think you cannot set the inputs via EMS but you can change the logic.
I didn't find a telegram for input 1 yet.

Inputs 1-3:
[image]
Input 4:
[image]

Changing logic in controller:
[image]

Thank you!

MichaelDvP, 2022-08-11:
0x4A2 looks like status of the inputs, You should check if the other bytes represents the status of all inputs, please check  by toggeling or inverting each input and watch 04a2.
0x488 looks like bitfields, but i think not the input state, but a reaction of the inputs, maybe the block states.
0x486 is really long, i expect that there are also the configs of all inputs. Watch this telegram and toggle the settings for logic level, and each blocking mode for each input, to get the positions.
Are the options for each input identical?

beSt0815, 2022-08-11:
Ok, I do.
Yes, options are identical.

beSt0815, 2022-08-19:
I deactivated all actions for inputs in the HP. Set the logic to high and tested all inputs.
The only input witch generates a single telegram is input 1:

```
On:	Boiler(0x08) -> All(0x00), ?(0x04A2), data: 00 01 01 00 00 00
Off:	Boiler(0x08) -> All(0x00), ?(0x04A2), data: 02 01 01 00 00 00
```
Seems that the telegrams are for the actions e.g. block DHW, block cooling ... except input 1.

Not that easy as it seems to be.

mvjt, 2022-08-21:
> I deactivated all actions for inputs in the HP. Set the logic to high and tested all inputs. The only input witch generates a single telegram is input 1:
>
> ```
> On:	Boiler(0x08) -> All(0x00), [...]

Hi, thanks for your time and help. I'm a noob but I was wondering if you have a function configured for the inputs when you trigger them? On the screenshot above they are all off (aus)...

The competitor, Husdata, only allows triggering EXT1 and EXT2 software wise so there might be e limitation for EXT3 and above...

beSt0815, 2022-08-21:
There are 10 functions in total.
[image]
[image]
[image]

I tried a few but 100 combinations (10^2 ) in total for each input is a lot.
The telegrams for ext1 without function are above. @MichaelDvP ist that enough to try it at least for this input?

I ordered a 4ch relais module an I will connect it to my smart home system. It allows me to debug from my computer.
A binary input gives me the status of my electrical heater.

MichaelDvP, 2022-08-25:
>  I tried a few but 100 combinations (10^2 ) in total for each input is a lot.

You don't need to try any combination. Just each of the single settings for one input and one setting for each inputs.
I think the settings are all in 0x486, we know only input 4 logic as offset 42 and 10 options per input. 
Try to set the logic level for first input and read 0x486 and 0x4A2 (this coud be state of input), set the level back and set level of next input.
Next choose one input and set first option, read 0x486, reset first option and set second, and so on. I think each option is a sngle offset in 0x486, but we need to know the order of options and of inputs. Bosch not always orders the telegrams/offsets in the way they are numbered in manual 1-2-3-4, it could also be 3-2-4-1 or something different.

If we implement the inputs with each setting we have 48 new entities, blowing the mqtt and webpage buffers. Any idea how to reduce this? Could we make two entities per input as: `in1state: on/off` and `in1option: 00000000000` with options as string with a 0/1 for each option?

proddy, 2022-08-25:
is it more important to show these settings in EMS-ESP or be able to modify them? And how often? And via the web, API or MQTT? If we know what the usage patterns will be like it will help define the best way to implement it. For example if its a one-off then we could consider a setting in the web where when you 'change value' you upload a file with the new settings.

bbqkees, 2022-08-25:
I think the most useful use case is that once you set the parameters via the control panel, you can use EMS-ESP to set these relays to on of off. 
For instance if you have selected one contact for PV, you can use this to signal the heat pump to heat up the boiler from HA when the PV system has enough generated solar power.
Or you can block the auxiliary heating element from operating at times when electricity is too expensive.

mvjt, 2022-08-28:
> I think the most useful use case is that once you set the parameters via the control panel, you can use EMS-ESP to set these relays to on of off.
> For instance if you have selected one contact for [...]

+1

MichaelDvP, 2022-08-31:
> I think the most useful use case is that once you set the parameters via the control panel, you can use EMS-ESP to set these relays to on of off.

I think you don't need a relais if you can switch the logic. Keep input as it is and change the logic will set/clear the input.

The question is about setting all other parameters. Publish/set all as single entitity in mqtt will blow the buffer. 

This discussion was also for switchtimes in program and holiday-modes. These are also settings made not often and normaly dont change. We dont have a solution for this yet.

For a quick implementation my suggestion is to publish 4 states (on(off, probably from 0x4A2) and  4 options as string `00000000000` with one char per option from 0x486. Sending a string to the option command will change all settings, sending only a bool/single character only change the logic level. But still we need the offset posiition of the single settings.

proddy, 2022-09-08:
> For a quick implementation my suggestion is to publish 4 states (on(off, probably from 0x4A2) and 4 options as string 00000000000 with one char per option from 0x486. Sending a string to the option [...]

I think this is best working solution for now.

MichaelDvP, 2022-09-09:
I've prepared a test-build with this logic, only need to insert the right offset positions of the parameters.

mvjt, 2022-09-17:
Would be awesome if we could get all the details needed to close this one...

Cali1205, 2022-09-25:
Hi all,
I just stumbeled over this discussion because I'm also working on optimizing my heatpump related to the production of my PV system.
So I did the work to find out the offsets of the register 0x486. There are also some Offset values where I cannot see any impact.

Description | Offset
-- | --
Input High/Low I1 | 0
Input High/Low I2 | 1
Input High/Low I3 | 2
EVU 1 - I1 | 3
EVU 1 - I2 | 4
EVU 1 - I3 | 5
EVU 2 - I1 | 6
EVU 2 - I2 | 7
EVU 2 - I3 | 8
EVU 3 - I1 | 9
EVU 3 - I2 | 10
EVU 3 - I3 | 11
Compressor block - I1 | 12
Compressor block - I2 | 13
Compressor block - I3 | 14
aux. Heater block - I1 | 15
aux. Heater block - I2 | 16
aux. Heater block - I3 | 17
Cooling block - I1 | 18
Cooling block - I2 | 19
Cooling block - I3 | 20
Heating block - I1 | 21
Heating block - I2 | 22
Heating block - I3 | 23
Hot water block - I1 | 24
Hot water block - I2 | 25
Hot water block - I3 | 26
PV - I1 | 39
PV - I2 | 40
PV - I3 | 41
Input High/Low I4 | 42
Compressor block - I4 | 43
aux. Heater block - I4 | 44
Cooling block - I4 | 45
Heating block - I4 | 46
Hot water block - I4 | 47
PV - I4 | 52

MichaelDvP, 2022-09-25:
Wow, great!
Nasty that I4 follows a different logic, but we know this from Bosch. 
Have you also tried to find the input state if you switch an input or invert it's logiic. 
I'll implement a setting strng for each input as mentioned before as 0/1 for each setting. `0000000000` for I1-I3, 
Sending a single 0/1 char inverts, otherwise need to send all settings.
Should we make I4 only 7 bytes long, or also 10 bytes and ignore EVU positions?

MichaelDvP, 2022-09-29:
@Cali1205 Have you tested the dev build? This should show the config.

mvjt, 2022-09-29:
> Hi all, I just stumbeled over this discussion because I'm also working on optimizing my heatpump related to the production of my PV system. So I did the work to find out the offsets of the register [...]

Praise the lords! Thanks @Cali1205

mvjt, 2022-10-01:
I can confirm the output I see from my HP matches the summary from @Cali1205

MichaelDvP, 2022-10-01:
> The big question is now how we can simulate activation of the inputs themselves...

Toggle the logic level and the input will switch. Just set `hpin1opt`to 0/1, on/off or true/false

proddy, 2022-10-01:
yes, Michael had already implemented this with the commands hpin1opt....hpin4opt

mvjt, 2022-10-01:
Fantastic work. I just need a few days to build courage to test it. If it works, I'll reach Nirvana...

mvjt, 2022-10-01:
Tested this in RO. Exposing my lack of understanding but isn't offset supposed to be 00 for Input 1?

`001+15:51:13.940 I 1: [command] [readonly] Calling command 'boiler/hpin1opt' (Inställningar Ingång 1) with value on
001+15:51:14.057 I 2: [telegram] [readonly] Sending write Tx telegram: 0B 08 FF 01 03 86 01`

MichaelDvP, 2022-10-01:
You're right, typo in the handling of bool input. Full string with all options should use correct offsets. I'll fix it.

mvjt, 2022-10-10:
Has any daredevil tested controlling the inputs using hpin1opt? I would be happier if someone else was willing to put heating/varmwater at risk for the winter :-)

Bingo2023, 2023-04-26:
sorry for opening this again.
I have a new Buderus Heat Pump Model: "Buderus Logatherm WSW196i.2-6".
Buderus did a SW update in Summer 2022 (HP and display), so I guess it is one of the latest Versions.

Attached you see one picture of the display setting where you can activate the options by hand.
In addition I texted all 4 inputs via Excel in a list.

[image]
[image]

Bingo2023, 2023-04-26:
I'm using **EMS-ESP Version v3.6.0-dev.9**
when using the new string input I get strange results (also note that input 1 has now 12 instead of 11 bits)
[image]

I also tried to add a custom command using the above offset list. It seems I can read the PV state by EMS but not write. So if I activate it via the touch screen I see it on the below custom variable changing from 0 to 1.
[image]

Also note that the second "bit" is now a percentage input instead of a bit.

Basicaly I'm only interested in writing the PV bit - so any help regarding that would be nice.

MichaelDvP, 2023-04-27:
> when using the new string input I get strange results (also note that input 1 has now 12 instead of 11 bits)

Yes there is a bug in reading the positions. I'll fix it. The string should be 10 char long with order described in the helptext below the input field. Could you make a log while changing the settings for brime pump percent and brime low pressure, than i can add these values.
> I also tried to add a custom command using the above offset lis

This should work, but use Value_Type `UINT` and set 0/1. BOOL is not clear because Bosch uses mostly 0x00/0xFF, but sometimes (like here) 0x00/0x01 for bools.  Now writing custom bool ON is 0xFF.
(I suspect that ems telegrams with type-id 0-7F uses 0/FF for bool and ems+ uses 0/1, I'll implement this to the custom entitiies)

proddy, 2023-04-27:
@MichaelDvP I was going to refactor `DeviceData.tsx` today to split out the Dialogs for performance, so you may want to wait before applying your Input field fix.

MichaelDvP, 2023-04-27:
@proddy I'll update my dev and wait with PR for your changes/current PR merged. I'll check for conflicts before making PR.

MichaelDvP, 2023-04-27:
@Bingo2023 Please test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
This should fix string in/output and the custom bool entity. The brine pump mod/pressure is not included in this.

Bingo2023, 2023-04-28:
> @Bingo2023 Please test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest This should fix string in/output and the custom bool entity. The brine pump mod/pressure is not included in this.
Thank you so much for the intermediate release of this SW. I flashed it an testet it. All following result are based on this Version.
Setting all 10 bits works now for Input 1 - 3 (Input 4 has some issue - not analysed further).

1000000001		turn PV on @Input 1/3
1000000000		turn PV off @Input 1/3
That is all I wanted for now and it is working - thx for that. Just need to figure out how to set it via MQQT.

Bingo2023, 2023-04-28:
[image]
[Input1_33prz+LowPressure.txt](https://github.com/emsesp/EMS-ESP32/files/11351476/Input1_33prz%2BLowPressure.txt)
here are two settings withe logfile.
How done: off -> 33% -> 10sec wait -> low pressure on -> 10sec; same to turn off

[image]
[image]
here some pics from the settings, you have turn it on -> 33% -> press OK

Bingo2023, 2023-04-28:
by the way "custom entitiies" are not working ether on bool or uint (yesterday release).
[image]
But I don't mind as the string setting is now working :)

MichaelDvP, 2023-04-28:
> by the way "custom entitiies" are not working ether on bool or uint (yesterday release).

The NaN for the bool setting is a bug, i'll fix it. 39/40 should show up, but it takes a minute and you have to reload the page to update values. But on this setting page the values are only for info, the values should be correctly shown on the dashboard, device: custom, there you can also set the value (if writeable).

The brine low-pressure is offset 30, brine pump modulation offset 30. Also you have mentioned "overheating protection hc1", this seems to be offset 36. So only the setting in offset 33 is unknown. 

I think i'll rework the setting and add the brine settings to the end, modulation as 3 digit number 000-100. Also for setting new values allow `x` for unchanged value, e.g. `xxxxxx1` to switch on PV setting.

MichaelDvP, 2023-04-28:
Please update to my new build and test. I don't have a heatpump, but added the entity This seems to work (i dont get values or a reply from my ems-oil-boiler, but see the command sent):
```
ems-esp32-s3:# call boiler hpin1opt "01xxx10xxxxx100" 1
000+00:05:30.273 N 33: [emsesp] Me(0x48) -W-> boiler(0x08), ?(0x0486), data: 64 (offset 27)
000+00:05:31.391 N 34: [emsesp] Me(0x48) -W-> boiler(0x08), ?(0x0486), data: 00 (offset 18)
000+00:05:31.651 N 35: [emsesp] Me(0x48) -W-> boiler(0x08), ?(0x0486), data: 01 (offset 15)
000+00:05:31.950 N 36: [emsesp] Me(0x48) -W-> boiler(0x08), ?(0x0486), data: 01 (offset 3)
000+00:05:32.265 N 37: [emsesp] Me(0x48) -W-> boiler(0x08), ?(0x0486), data: 00
```
Also for a custom entity for pv:
```
ems-esp32-s3:# call custom pv 1
000+00:38:32.247 N 2: [emsesp] Me(0x48) -W-> boiler(0x08), ?(0x0486), data: 01 (offset 39)
```

Bingo2023, 2023-05-04:
[image]

with your latest release is something wrong.
See picture, all setting are off and therefore should be 000000000000000

MichaelDvP, 2023-05-05:
Ups, try again with corrected build.

Bingo2023, 2023-05-05:
now it works.
I can confirm:
`100000000000000		PV off`
`100000000100000		PV on`
`xxxxxxxxx1xxxxx		PV on`
`xxxxxxxxx0xxxxx		PV off`

bbqkees, 2023-10-06:
My own heat pump still did not arrive unfortunately.
@Bingo2023 Could you perhaps make a screenshot of the entities and how this now looks like in the web interface?

I often get questions about this feature, I want to add an article to the wiki.

masterx1981, 2023-10-29:
Hi! I'm sorry to ask in this discussion, but i'm new on the ems-esp firmware. I've all setup correctly, connected with the heatpump and to the ha via mqtt quite easily, but now i need to set/unset the i4 input based on the battery/solar energy info from the inverter via ha. All the info's are here, but i'm a bit cofused how to publish the command for change only the correct bit via mqtt.
Someone can, please, link some pages where i can start to "get the grip"? Thanks!

proddy, 2023-10-30:
We need @Bingo2023 to help us write the wiki article

MichaelDvP, 2023-10-31:
> but now i need to set/unset the i4 input based on the battery/solar energy info from the inverter via ha.

I can not help with HA, 
> how to publish the command for change only the correct bit via mqtt.

I you only want to change the state (e.g. the invert bit to toggle the input read), you can send a single `0` or `1` to the mqtt command. 
If you want to configure the input to PV, you have to set the right bit and send a string  `xxxxxxxxx1xxxxx`

masterx1981, 2023-10-31:
The problem is that the hpin1opt...hpin4opt from home assistant seem that are seen as read only entities...

proddy, 2023-10-31:
can you share a screenshot?

masterx1981, 2023-10-31:
From what i know, a "sensor." entity is readonly. A writeable entity is, for example, "number."

[image]

proddy, 2023-10-31:
thanks and yes. it should be a string. I'll check

proddy, 2023-10-31:
I found the reason. Since it's a string it's excluded as a writeable entity in HA. I think using https://www.home-assistant.io/integrations/text.mqtt/ we can add it, and this will also be useful for set_datetime, set_holiday, set_wwswitchtime etc

I'll create a new issue

proddy, 2023-10-31:
will be addressed in https://github.com/emsesp/EMS-ESP32/issues/1373

proddy, 2023-10-31:
@masterx1981 can you try https://github.com/emsesp/EMS-ESP32/releases/tag/test

masterx1981, 2023-10-31:
Seem ok, now the date on the hmc310 thermostat and the binary fields are text fields, so i can change using string operators the values. So i can, for example, update co stantly the date time via ntp, nice! Now i need to figure out why the binary fields not change when i1 or i4 are changed, but i think that my wpls8.2 isn't fully supported (is recognized as Enviline/Compress 6000AW/etc.)
Some controls works ok, but other are way off. But this is another topic 😅

proddy, 2023-10-31:
great, thanks for the feedback.

masterx1981, 2023-10-31:
Plase note, i see the vinary fields both as "text." and "sensor.", it's correct?

proddy, 2023-11-01:
> Plase note, i see the vinary fields both as "text." and "sensor.", it's correct?

This is left over in your MQTT broker. You will need to use MQTT Explorer and remove them manually (under homeassistant/sensor/ems-esp).

masterx1981, 2023-11-01:
Strange, before installing the new firmware i've deleted the mqtt devices in home assistant, in the hope yo not have any leftover.
If i go under home assistant/sensor/ems-esp-pdc/boiler_hpin1opt i have a message with the date of this morning (and i've done the job yesterday evening)

proddy, 2023-11-01:
> Strange, before installing the new firmware i've deleted the mqtt devices in home assistant, in the hope to not have any leftover. If i go under home assistant/sensor/ems-esp-pdc/boiler_hpin1opt i have a message with the date of this morning (and i've done the job yesterday evening)

oh - let me check too, I may have missed something.

proddy, 2023-11-01:
> oh - let me check too, I may have missed something.

its definitely removed. you need to remove from the MQTT broker as those old topics are persisted.

masterx1981, 2023-11-01:
Yes, had to disable mqtt from ems-esp, delete devices in ha, delete leftovers from mqtt explorer, reenable mqtt from ems-esp, and now is all ok, sorry but i'm pretty new also on ha.
Thanks!

MichaelDvP, 2023-11-01:
EMS-ESP has a function to clean the HA states in mqtt broker: 
leave mqtt enabled, but disable HA-discovery, then ems-esp starts to remove all retained HA entries, after some minutes you can enable HA discovery again and the new states are published.

masterx1981, 2023-11-01:
Wow, good to know for the future, thanks!

Bingo2023, 2023-11-01:
> We need @Bingo2023 to help us write the wiki article

sure I help, what is needed?

[20 more replies]

commit 5f87718d (2023-05-27): heatpump inputs, fix #600
commit c9ab6b23 (2023-05-27): fix #600,  read telegram 486
closed 2024-02-03 by proddy