emsesp/EMS-ESP32 #601: DHW modes on BOSCH and IVT (Geo500series) heat pumps

henrik-github, 2022-08-15:
Bosch 7001i/IVT geo 500-series have three preset DHW modes. Eco, Normal and Comfort.

When setting DHW mode to Comfort using the Bosch app, DHW mode is set to "high" in the EMS interface.
When setting DHW mode to Normal using the Bosch app, DHW mode is set to "low" in the EMS interface.
When setting DHW mode to Eco using the Bosch app, nothing happends in the EMS interface.

The following is posted in the log when sniffing over Telnet:
Eco:
Gateway(0x48) -> Thermostat(0x10), RC300WWmode(0x02F5), data: 05 (offset 2)

Normal
Gateway(0x48) -> Thermostat(0x10), RC300WWmode(0x02F5), data: 01 (offset 2)

Comfort
Gateway(0x48) -> Thermostat(0x10), RC300WWmode(0x02F5), data: 02 (offset 2)

Would it be possible to add Eco (data:05) so it could be set from the EMS UI?
Would it also be possible to change the names of the DHW modes?

proddy, 2022-08-15:
thanks for reporting. Yes, this should be possible.

proddy, 2022-08-21:
@henrik-github please check the latest dev build and close this issue if its resolved

henrik-github, 2022-08-22:
Great job! Works as expected.

closed 2022-08-22 by henrik-github