emsesp/EMS-ESP32 #602: Add simulated EMS output in log when using "Read Only Mode"

mvjt, 2022-08-16:
When using the Settings->Read Only mode, there is no output of outgoing EMS messages in the logs (telnet). I recommend we add the output to the log, but marked with [SIM] or similar to highlight it is a simulated message not actually being sent to the boiler/HP. 

This would give us the possibility to validate that the FW will send is expected without any risk of messing up the HP/boiler operation.

proddy, 2022-08-16:
I think it's a good idea and relatively simple to add.

MichaelDvP, 2022-08-17:
I think this is already implemented. 
[https://github.com/emsesp/EMS-ESP32/blob/8d3bcde8d6723d1ab75dfb4504ccef38c0ffa0bb/src/telegram.cpp#L364-L369](https://github.com/emsesp/EMS-ESP32/blob/8d3bcde8d6723d1ab75dfb4504ccef38c0ffa0bb/src/telegram.cpp#L364-L369)
Or do you mean something different?

mvjt, 2022-08-17:
This is a little bit embarrassing but I guess I missed this because it is logged in raw and not pretty print. But raw is good enough for me so I will close this.

closed 2022-08-17 by mvjt