emsesp/EMS-ESP32 #604: Trendline outdoor temperature sensor

sweetpants, 2022-08-23:
Does anybody know if the 'outdoor temperature' sensor of the Trendline boiler is exposed on the EMS/MQTT bus?

regards, Harry

sweetpants, 2022-08-23:
Hmmm, searching a little further it seems a function of the Thermostat? Mine is a Moduline Easy, sw version 2.22.0 hw version 3, Boiler Trendline HRC30/CW5. Can someone confirm?
Regards, Harry

proddy, 2022-08-23:
do you see the outdoor temp anywhere, like on the Easy thermostat or in the boiler's panel? If it's there, we should be able to grab it

sweetpants, 2022-08-24:
Here you go, screenshot on easy app on Android
[image]

proddy, 2022-08-24:
ok, EMS-ESP does read the outdoor temp from the boiler, from telegram 0x19. Are you not seeing this?

bbqkees, 2022-08-25:
Do you really have an outdoor temperature sensor connected to the boiler or is the app getting it from elsewhere?

sweetpants, 2022-08-26:
@bbqkees , i think you are right. I have opened the ceiling and could not find an external temperature sensor wires so the app is getting it from somewhere else. Because I want to be independent of external (internet) sensors i'm going to add a separate dallas sensor to the EMS gateway. Case closed

closed 2023-02-19 by proddy