emsesp/EMS-ESP32 #609: Set cooling temperature Nefit Enviline HMC300 and switch of relative humidity (dewpoint) minimum

jankees17031961, 2022-08-27:
We use the HP to cool our house. I remove the RV (rel. humidity) sensor on the tube and changed the thermostat 1010H for a 1010 to have no RV limitations. Next I set the target cooling temp in the service menu via the minimal temperature for cooling in Heating Circuit 1, the only connected circuit.

it is helpful to manage this with an EMS write command to switch off/on the RV limitation ( like I believe the German version has) and to be able to set the minimum flow temperature when cooling.

I would appreciate your help and please let me know if you need more info.

Thanks, Jan Kees

My system details are:

{
  "System Status": {
    "version": "3.4.2b5",
    "uptime": "000+10:35:17.427",
    "freemem": 110,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -25,
    "IPv4 address": "192.168.1.98/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.1",
    "IPv6 address": "fe80:0000:0000:0000:0edc:7eff:fe5d:9648",
    "static ip config": true,
    "enable IPv6": true,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 12443,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 115612,
    "bus reads (tx)": 10858,
    "bus writes (tx)": 2,
    "bus incomplete telegrams": 3,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 5,
    "tx gpio": 17,
    "dallas gpio": 4,
    "pbutton gpio": 33,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx",
      "device id": "0x08",
      "product id": 172,
      "version": "01.18",
      "entities": 105,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x049F 0x04A0 0x04A3 0xF9 0x04A2 0x04AC 0x17 0x36"
    },
    {
      "type": "Thermostat",
      "name": "RC100/Moduline 1000/1010",
      "device id": "0x38",
      "product id": 165,
      "version": "40.04",
      "entities": 2,
      "handlers received": "0x042B",
      "handlers fetched": "0x0273",
      "handlers pending": "0x047B",
      "handlers ignored": "0xBF"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "13.07",
      "entities": 47,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x02D0 0x02D2 0x031B 0x031E",
      "handlers ignored": "0xF9 0x0291 0xE7 0x02E0 0x02EA 0x061E 0xBF"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0
    }
  ]

jankees17031961, 2022-09-02:
> Is it possible to cool without rel. humidity send by a remote thermostat? In my case it is the RC100H device and in every manual you need that device connected.

You can change the 1010H with an 1010 and change in the service menu HC1 the device form RC100H to RC100. Then it works without rel. humidity. Make sure you program the new thermostat as ‘Fb’

proddy, 2023-07-03:
@jankees17031961 can this issue be closed?

jankees17031961, 2023-07-03:
Hi Proddy,

Is ok to close, would be helpful if resolved as now it is not possible to lower the temp below the humidity in the outside air. In my case, the pipes are isolated with Armaflex so they can not condense at low temperature.

Best,

Jan Kees

closed 2023-07-05 by proddy