emsesp/EMS-ESP32 #610: Support for more telegrams for FW100

nemesis03, 2022-08-30:
At the moment only a few telegrams are supported by ems-esp. It would be great to add more telegrams to see (and edit) the schedule for a heating circuit and for warm water.
I just did some testing and found some telegrams that work.

**instant warm water**
I go a FB100 for testing purposes to see how the command for instant warm water is transmitted from the FB100 to the FW100.
The FB100 sends the following command:
Thermostat(0x19) -> Thermostat(0x10), ?(0x0115), data: FF (offset 1) (in hex: "98 10 FF 00 00 15 FF CD")

I can send the same command from the ems-esp by doing: call system send "8B 10 FF 00 00 15 FF"
After issuing this command the FW sends a command to the boiler and another command to all
000+00:01:14.026 N 6: [emsesp] Me(0x0B) -> Thermostat(0x10), ?(0x0115), data: FF
000+00:01:16.560 N 10: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 00 3C (offset 1)
000+00:01:16.589 N 11: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x01D3), data: 08 00 00
The "8" in "08 00 00" indicates that instant warm water is active.
0x01D3 can be queried by "read 90 01d3" and gives the result "08 00 00" when instant warm water is active or "00 00 00" when in automatic mode.
I did not manage to query 0x0115, the result is always empty.

**warm water schedule**
When I edit the warm water schedule in the FW100 the current schedule is broadcasted to all
000+23:05:42.034 N 325: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x01FB), data: 32 14 0F 5C FF FF FF FF FF FF FF FF 32 14 0F 5C FF FF FF FF FF FF FF FF
000+23:05:42.902 N 326: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x01FB), data: 32 14 0F 5C FF FF FF FF FF FF FF FF 32 14 0F 5C FF FF FF FF FF FF FF FF (offset 24)
000+23:05:45.864 N 327: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x01FB), data: 32 14 0F 5C FF FF FF FF FF FF FF FF 3C 18 0F 5C FF FF FF FF FF FF FF FF (offset 48)
000+23:05:46.958 N 328: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x01FB), data: 3C 1C 0F 5C FF FF FF FF FF FF FF FF (offset 72)

There are 6 time/temperature pairs per day. So in thies example  "32 14" means 32(hex)=>50(°C) at 14(hex)=>20 quarterhours past midnight=>5h, "0F 5C" means 0F=>15(°C) at 5C=>92 quarterhours past midnight=>23h
The following 4 pairs of "FF FF" indicate that those 4 time/temperature pairs are not set. After that comes the next day.

The schedule can be queried with "read 10 01FB".

It is also possible to chance values, for example
call system send "0B 10 FF 00 00 FB 3c"
will change the first datapoint to "3c" (60°)
and
call system send "0B 10 FF 01 00 FB 15"
will change the time to 15 (5:25h)

**heating circuit schedule**
Like for warm water there is a schedule for the heating circuit.
When I edit the schedule A n the FW100 the current schledule is broadcasted
001+02:19:52.007 N 927: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x022D), data: 02 16 01 4E FF FF FF FF FF FF FF FF 02 16 01 50 FF FF FF FF FF FF FF FF
001+02:19:52.440 N 928: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x022D), data: 02 16 01 50 FF FF FF FF FF FF FF FF 02 16 01 50 FF FF FF FF FF FF FF FF (offset 24)
001+02:19:52.882 N 929: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x022D), data: 02 16 01 58 FF FF FF FF FF FF FF FF 02 1C 01 58 FF FF FF FF FF FF FF FF (offset 48)
001+02:19:53.289 N 930: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x022D), data: 02 1C 01 58 FF FF FF FF FF FF FF FF (offset 72)

I think this is 02=heat, 01=eco and I would guess 00=nofrost
The times work like the warm water schedule and again 6 time/mode pairs per day.

When I try to query the schedule with "read 10 022d" I get an error when the offset is 75.
000+12:21:42.421 I 16: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x022D), data: 02 16 01 4E FF FF FF FF FF FF FF FF 02 16 01 50 FF FF FF FF FF FF FF FF 02
000+12:21:42.703 I 17: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x022D), data: 16 01 50 FF FF FF FF FF FF FF FF 02 16 01 50 FF FF FF FF FF FF FF FF 02 16 (offset 25)
000+12:21:42.986 I 18: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x022D), data: 01 58 FF FF FF FF FF FF FF FF 02 1C 01 58 FF FF FF FF FF FF FF FF 02 1C 01 (offset 50)
000+12:21:45.533 E 19: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 4B 20 01 2D

This only happens when querying 20 bytes from offset 75. If I query 19 bytes it works
call system send "8B 90 FF 4B 19 01 2D"
000+23:00:42.263 I 1631: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x022D), data: 58 FF FF FF FF FF FF FF FF 50 72 6F 67 72 61 6D 6D 20 41 20 20 20 20 20 20 (offset 75)
000+23:00:42.489 I 1632: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x022D), data: 20 20 (offset 100)
Any ideas why it is a problem to quer 20 bytes from offset 75?

I did not try editing yet, but I guess it will work like with warm water. Also I did not try editing the other schedules (B and C).

I would like to know how to send a command to edit the time for the next change. On the FW100 and the FB100 I can press the heating button and turn the knob to change the time for the next mode change, but unfortunately I cannot see that this information is passed from FB100 to FW100.
If I set up the FB100 communicates directly with the boiler. Does anyone know whether there is a possibility to make a setup that the FB100 is communicating with the FW100 (like a remote) and then the FW100 is executing the commands towards the boiler?

I hope my findings help other to make their FW100 smarter. Maybe some of it can be implemented into ems-esp. I think the "instant warm water" might be helpful, for example to have warm water in the morning on weekends that one has to get up early (for work).

Im am open to doing more tests and experimenting with the FB100.

MichaelDvP, 2022-08-31:
Nice work. Could you please also send your configuration (from help-page). Looks like you have FW100 as master thermostat id 0x10 and FB100 as remote on hc2 (0x19). Do you have other hcs active?
>Thermostat(0x19) -> Thermostat(0x10), ?(0x0115), data: FF (offset 1) (in hex: "98 10 FF 00 00 15 FF CD")

This is is offset 0 in the hex, Sending offset 0 seems to work, does the log really reports offset 1?
This could easly added, is the telegram 0x1D3 frequently broadcasted, or do we need to fetch it? Have you seen other values for this?

The circuit schedule is difficult. We had discussions about that for RC35, RC30, RC300. Publishing the lot of time/temp values will blow the mqtt buffer and also the web buffer. For RC20/RC30/RC35 we have now a single value with no, day, temp. Sending only number reads back the new data, sending number and data will change this switchpoint. This can be adapted for the Junkers schedule, but there are discussions about other methods for switchpoints, holiday-modes, heatpump inputs,....

Can you test what telegrams are used for the other heating-circuits. Just read telegrams before and ahead of 0x22D to check which are scheduls. In #400 we have for Junkers FR100 0x230 for hc1/A, 0x231 for prog B, 0x236 for hc2/A, etc. Maybe the FW100 is similar organized. 
> but unfortunately I cannot see that this information is passed from FB100 to FW100.

Than this is only stored in the remote and not writable from the bus.
> Does anyone know whether there is a possibility to make a setup that the FB100 is communicating with the FW100 (like a remote) and then the FW100 is executing the commands towards the boiler?

I think  if FW100 is the master thermostat, this is normally the only thermostat communicating to the boiler. The remote FB100 only takes to the master. You have to remove the FW100 and set FB100 as master for the hc, working only for a single hc.

nemesis03, 2022-08-31:
You can find the configuration in the attached file. [emsesp_info.json.txt](https://github.com/emsesp/EMS-ESP32/files/9460088/emsesp_info.json.txt)
FW100 is the master thermostat with id 0x10. The FB100 is the remote for hc1 with id 0x18 or when set to hc2 the id is 0x19. We only have one hc so the remote is set to hc1. I told the FW100 that there is no remote, so that the FW100 continues to control the boiler.
If I activate the FB100 as remote for hc1 the hc1 is controlled by the FB100 and the FW100 only controls warm water. In that configuration I cannot set any values/schehules for the heating in the FW100. 
In the current configuration I have a schedule for the heating in the FW100 and a different schedule for the heating in the FB100. I am not entirely sure which device controls the boiler right now.
As I set I did not yet manage a setup where the FB100 remote controls the FB100. I cannot remove the FW100, because the FB100 is not capable of controlling warm water (the FB100 can only send the instant warm water command to the FW100).

> This is is offset 0 in the hex, Sending offset 0 seems to work, does the log really reports offset 1?
I tested the command two time from the FB100, the first time with normal "watch" where the offset 1 is reported, the second time with "watch raw". The command definitely works with the offset 0. 

I think the telegram 0x1D3 is not frequently broadcasted. Unfortunately it is of limited use, because after triggering instant warm water it stays on "08 00 00" even after instant warm water is finished. Maybe it is reset to "00 00 00" when the next warm water timepoint is reached. I just a received a 0x1D3 broadcast and currently waiting for the next one. OK, it is broadcasted every 10 minutes.

There also are telegrams for 0x22D and 0x22E that work for the FW100 and the FB100.

For the circulation pump the schedule is stored in 0x20F.
The circulation pump mode is not stored in the FW100, but directly in the boiler. When accessing the setting in the FW100 the current value is read from the boiler. Setting the circulation pump mode via ems-esp works and is not overwritten by the FW100. When I select the "off" value in ems-esp the value is written to the boiler, but the boiler does not know the off value ("00") and send back a "01" (one per hour). When I select "continuous" ems-esp also send the value "00" to the boiler. Manually sending the value "07" works and results in continuous circulation pump activity (but labeled as "7 times per hour" in the FW 100). It look like the sending of "00" when "continuous" is selected is an error in ems-esp.

Is there a script somewhere to query the FW100 for all kind of telegrams and see which one get a response?

RoooNY87, 2023-11-20:
I'm also interested in more functions for FW100.
I tried the demo for ems esp Webinterface with RC35, and thought these functions are also available for fw 100.
For me the time scheduling and hc2 heating type + party + vacation is very important.
The Functions are all available in manual mode in FW100

proddy, 2023-12-02:
The easiest way is you guys with an FE100 to use custom entities to find the data - see https://emsesp.github.io/docs/Troubleshooting/#im-missing-certain-data-from-an-ems-device

it's very time consuming and back-and-forth communications for us developers to add new entities

closed 2024-03-24 by proddy