emsesp/EMS-ESP32 #612: Change name of entity within WebUI

proddy, 2022-09-06:
Extending the Customization page in the WebUI so when clicking on an entity you can manually change it's name, or reset it back to the original. The name will be stored in the customization file so we may need to increase the SPIFFS partition later.

proddy, 2022-09-09:
I was working on this and checked in the work so far. It's not tested and will probably crash. I'll get round to finishing it off later

MichaelDvP, 2022-09-09:
Working, but test input is very slow.

proddy, 2022-09-09:
> Working, but test input is very slow.

it's because of the onChange event forcing a repaint. I'm not how to fix it easily.

MichaelDvP, 2022-09-10:
Not rendering the content works, but looks a bit odd:
```
  const content = () => {
    if (deviceEntity) {
      return;
    }
...
```
[image]

Some other remarks:
- entering empty name does not clear the custom name, also entering the default name.
- the dialog shows actual name and new name, but not the default.
- for my system with only one hc (rc5) and a simple oil-burner the ram-entities take 20kB heap. For people with more hcs and heatpump this could be critical, especially for ETH variant. We have seen that below 90k heap there are issues with json-buffers for api/web. Maybe @bbqkees shoud look for a PSRAM ESP32 (WROVER) for next generation gateway ;-).

proddy, 2022-09-10:
Thanks for checking Michael.

> entering empty name does not clear the custom name, also entering the default name.

That's odd. I'll look into it.

> the dialog shows actual name and new name, but not the default.

It shows the translated default (in orange text). This is intended, or is it not intuitive? 

> heap

I was worried about this. I had to make the dv.custom_fullname a std::string so you can rename on the fly. You can't put things into Flash dynamically. So now all the command descriptions are fullnames are duplicated in RAM which isn't good.  I need to see how to put things into PROGMEM during runtime. Any ideas?

MichaelDvP, 2022-09-10:
> It shows the translated default (in orange text). This is intended, or is it not intuitive?

Ah, so [this line](https://github.com/emsesp/EMS-ESP32/blob/4cd655fb36af656d08b2466084ec457579bc909a/src/emsdevice.cpp#L964) should be be the translated_word(dv.fullname), that's what i expected.

proddy, 2022-09-10:
> > It shows the translated default (in orange text). This is intended, or is it not intuitive?
> 
> Ah, so [this line](https://github.com/emsesp/EMS-ESP32/blob/4cd655fb36af656d08b2466084ec457579bc909a/src/emsdevice.cpp#L964) should be be the translated_word(dv.fullname), that's what i expected.

oops, yes, correct

MichaelDvP, 2022-09-10:
Empty name works and sets back to flash-strings, don't know why my first try haven't worked.

proddy, 2022-09-10:
Also interesting is that it seems on the ESP32 you don't need to use flashstrings. Anything that is declared `const` goes into Flash PROGMEM. So those macros like F() F_() etc can be reduced which will save code. I still need to find how to put things into PROGMEM at runtime. See https://esp32.com/viewtopic.php?t=20595

proddy, 2022-09-10:
I did # 3 and rolled back to use the flash strings for command descriptions instead of strings. It saves about 3.6KB

proddy, 2022-09-10:
> Ok, i can reproduce now. Setting empty name sets web back to n string. But saving to emsesp does not work, the cn-string is not cleared in customization file. So next visit of customization page shows the old string again.

ok. I'm out for the next 2 weeks so have limited time (and no ESP device with me) but can try and fix while travelling. Otherwise if you know where to change it please go ahead

MichaelDvP, 2022-09-10:
Till now i havn't found why the (empty) name is not updated in FS. I'm searching..

proddy, 2022-09-10:
the bug is in `EMSdevice::setCustomEntity()`. When you remove the custom fullname, `has_custom_name` becomes  `false` so that line `dv.custom_fullname = entity_id.substr(custom_name_pos + 1);` is not called so it's not updated. I think best fix is just remove the `if..` statement

MichaelDvP, 2022-09-10:
Thx, i think better `else dv.custom_fullname = ""`, to avoid the undefined substring.

proddy, 2022-09-10:
yes that's better

closed 2022-09-19 by proddy