emsesp/EMS-ESP32 #614: Firmware reset after repowering

christoph-ng, 2022-09-09:
I've falshed the firmware to a new esp32, after setting the wlan SID and Password it gets connected to my wlan, but forgets the whole configuration after repowering the device.

MichaelDvP, 2022-09-09:
Probably a issue with partitioning. Make sure you have flashed the partition table with 64k SPIFFS. Use esptool for flashing.

proddy, 2022-09-09:
There's also a rare bug I fixed in v3.5.0 that creates the FS folder structure as I ran into this issue once before. I can back port that to 3.4.2 dev branch?

MichaelDvP, 2022-09-10:
> I can back port that to 3.4.2 dev branch?

Yes, please do. What was the issue, what the fix?

proddy, 2022-09-10:
> > I can back port that to 3.4.2 dev branch?
> 
> Yes, please do. What was the issue, what the fix?

On a fresh ESP32, whn uploading using pio it would not persist the configuration files. Connecting to Serial I could see the LittleFS creation errors when trying to write the file. It may be by chance, but I added this code (lines 71-79) to create the directories first. I think it was a change in LittleFS. I keep forgetting to do a stupid 'pio pkg update' to get the latest libs

https://github.com/emsesp/EMS-ESP32/blob/cc44bc9d7fa085ea03d85277a89c8d4246aa465b/lib/framework/FSPersistence.h#L71

christoph-ng, 2022-09-11:
I'Ve solved the problem for me, I've flashed to 3.4.2 and the upgraded to 3.4.2b5, This works for me now. 
How can I get a built firmware for dev or 3.5.0 ?

proddy, 2022-09-11:
Just stick with the dev version (from https://github.com/emsesp/EMS-ESP32/releases/tag/latest) for now. It gets updated frequently and you'll always have the latest working version. v3.5.0 is pre-dev and needs to be built manually using PlatformIO. When the translations are finished it'll be moved to dev, probably in a few weeks time.

closed 2023-02-19 by proddy