emsesp/EMS-ESP32 #619: Device resets automatically multiple times a day.

nurk, 2022-09-14:
**Question**
My device does a reset multiple times a day without any obvious reason
The device is the E32 connected with a LAN cable and connected to a Junkers boiler

**Screenshots**
Not really applicable

**Device information**
```
{
  "type": "settings",
  "System": {
    "version": "3.4.1"
  },
  "Network": {
    "ssid": "",
    "password": "",
    "hostname": "ems-esp",
    "static_ip_config": false,
    "enableIPv6": false,
    "bandwidth20": false,
    "tx_power": 20,
    "nosleep": false,
    "enableMDNS": true
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "password": "****",
    "channel": 1,
    "ssid_hidden": false,
    "max_clients": 4,
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": true,
    "host": "***",
    "port": 1883,
    "base": "ems-esp",
    "username": "***",
    "password": "***",
    "client_id": "ems-esp",
    "keep_alive": 60,
    "clean_session": true,
    "max_topic_length": 128,
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_qos": 0,
    "mqtt_retain": false,
    "ha_enabled": true,
    "nested_format": 1,
    "discovery_prefix": "homeassistant",
    "publish_single": false,
    "publish_single2cmd": false,
    "send_response": false
  },
  "NTP": {
    "enabled": false,
    "server": "time.google.com",
    "tz_label": "Europe/Brussels",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": true,
    "port": 8266,
    "password": "***"
  },
  "Security": {
    "jwt_secret": "***",
    "users": [
      {
        "username": "admin",
        "password": "***",
        "admin": true
      },
      {
        "username": "***",
        "password": "***",
        "admin": true
      }
    ]
  },
  "Settings": {
    "tx_mode": 3,
    "ems_bus_id": 11,
    "syslog_enabled": false,
    "syslog_level": 3,
    "trace_raw": false,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "syslog_port": 514,
    "master_thermostat": 0,
    "shower_timer": false,
    "shower_alert": false,
    "shower_alert_coldshot": 10,
    "shower_alert_trigger": 7,
    "rx_gpio": 5,
    "tx_gpio": 17,
    "dallas_gpio": 4,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": false,
    "low_clock": true,
    "telnet_enabled": true,
    "notoken_api": false,
    "readonly_mode": false,
    "analog_enabled": false,
    "pbutton_gpio": 33,
    "solar_maxflow": 30,
    "board_profile": "E32",
    "fahrenheit": false,
    "bool_format": 1,
    "bool_dashboard": 1,
    "enum_format": 1,
    "weblog_level": 4,
    "weblog_buffer": 25,
    "weblog_compact": true,
    "phy_type": 1,
    "eth_power": 16,
    "eth_phy_addr": 1,
    "eth_clock_mode": 0
  }
}
```
And this is what the log output says:
```
000+00:00:00.000 INFO 0: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
000+00:00:00.000 INFO 1: [emsesp] Loading board profile E32
000+00:00:00.000 INFO 2: [mqtt] [DEBUG] Adding to queue: (Subscribe) topic='system/#'
000+00:00:00.000 INFO 3: [emsesp] Starting EMS-ESP version 3.4.1 (hostname: ems-esp)
000+00:00:00.000 INFO 4: [dallassensor] Starting Dallas sensor service
000+00:00:00.000 INFO 5: [mqtt] [DEBUG] Adding to queue: (Subscribe) topic='dallasssensor/#'
000+00:00:00.000 INFO 6: [analogsensor] Starting Analog sensor service
000+00:00:00.000 INFO 7: [mqtt] [DEBUG] Adding to queue: (Subscribe) topic='analogsensor/#'
000+00:00:08.822 ERROR 8: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 02 00 1B
000+00:00:12.067 ERROR 9: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 9A FF 00 19 00 71
000+00:00:12.442 ERROR 10: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 9B FF 00 19 00 72
000+00:00:13.253 ERROR 11: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 7B
000+01:53:02.166 ERROR 12: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 79
000+01:54:01.990 ERROR 13: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 79
000+01:57:01.835 ERROR 14: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 6F
000+02:02:01.922 ERROR 15: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 6F
000+02:02:02.483 ERROR 16: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 79
000+03:06:01.356 ERROR 17: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 6F
000+03:06:01.917 ERROR 18: [telegram] Last Tx Read operation failed after 0 retries. Ignoring request: 8B 90 FF 00 19 00 79
```

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2022-09-14:
Difficult to analyse, we had something like ths before: [In this case](https://github.com/emsesp/EMS-ESP32/issues/244#issuecomment-1010875734) a reflash and factory reset helped,[ in this case](https://github.com/emsesp/EMS-ESP32/issues/156#issuecomment-965709486) it was a faulty dc-dc converter (older type than eth-gateway), in some other the power supply was not stable. 

Try new software download, reflashing, factor reset and a different power supply.

What is your free heap? ETH needs more ram and the log indicates thermostats at 0x10 and 0x1A, so there are more heatingcircuits.

nurk, 2022-09-14:
Free heap is
[image]

I do not really know about more heating circuits.  I just have the one.

I reuploaded the 3.4.1 firmware and did a factory reset but now the blue led blinks 3 times, pauzes and repeats.
I cannot get into the web interface anymore.

Currently I have no option to change the power supply.  It is the one I got when buying it from bbqkees

nurk, 2022-09-14:
After reflashing with a usb cable the output says:
```
Wrote 1816928 bytes (1153714 compressed) at 0x00010000 in 29.4 seconds (effective 493.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[11:05:20]
[11:05:20]┌──────────────────────────────────────┐
[11:05:20]│ EMS-ESP version 3.4.1                │
[11:05:20]│ https://github.com/emsesp/EMS-ESP32  │
[11:05:20]│                                      │
[11:05:20]│ type help to show available commands │
[11:05:20]└──────────────────────────────────────┘
[11:05:20]
[11:05:20]ems-esp:$ lib/LITTLEFS/src/lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
[11:05:20]E (49) esp_littlefs: mount failed,  (-84)
[11:05:20]E (52) esp_littlefs: Failed to initialize LittleFS
[11:05:22][11:05:22]
```

MichaelDvP, 2022-09-14:
This seems to be something like #614, @proddy knows better how to manage this.

nurk, 2022-09-14:
I am so sorry for the trouble.  I did recover the device.
I connected to the gateway wifi and change the board type.
After connecting the lan cable and rebooting everything is back up.
I do find the LittleFS error on boot a bit weird.

bbqkees, 2022-09-14:
In the past weeks I mistakenly uploaded 3.4 instead of 3.4.1 to some Gateways.
Wasn't there an issue with 3.4 that the config was not persistent?

MichaelDvP, 2022-09-14:
In the OP the system info shows 3.4.1. But it's also worth a try with v3.4.2, there are a lot of fixes and improvements and it's pretty stable. I think proddy will release it soon.
BTW: if board-profile is correct after reboot, the LittleFS is working ok. Maybe this message was due to empty filesystem after factory reset.

nurk, 2022-09-14:
I will keep monitoring it and when 3.4.2 is out I will upgrade.

I am not going to leave this ticket open unneeded.  I can alway open a new one later if the new firmware has the same issues.

Thanks for the help.

MichaelDvP, 2022-09-15:
I don't think it's a software bug. v3.4.1 and v3.4.2 are running stable on many emsesp. 
There is also a reset-reason shown in system-status of v3.4.2. If this shows a exception we have to search in software and need to know what commands are sent to your system.,

nurk, 2022-09-16:
After the reinstall and reset it has been running for 44 hours which is a new record for me.

closed 2023-02-19 by proddy