emsesp/EMS-ESP32 #621: Bug in dhw circulation pump mode on RC35

miege2k, 2022-09-16:
Hey, at first my setup:
- Buderus GB125
- Buderus MC10 
- Buderus BC10
- Buderus RC35
- EMS-ESP E32

If I change the the value "dhw circulation pump mode" on my RC35 the following values appears on the telnet interface:
- off   : Thermostat(0x10) -> All(0x00), WWSettings(0x37), data: 00 (offset 3)
- on    : Thermostat(0x10) -> All(0x00), WWSettings(0x37), data: 01 (offset 3)
- auto: Thermostat(0x10) -> All(0x00), WWSettings(0x37), data: 02 (offset 3)

if I change the value in the ems-esp web interface 
- on    : Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 01 01

Is it possible to change this behavior? 
Thanks Christopher

proddy, 2022-09-16:
Hi, what do you see in Telnet when you change the 'dhw circ pump mode' from the Console or WebUI?

MichaelDvP, 2022-09-17:
What you see is the reaction of the thermostat, sending the `start pump` commant to the boiler. The setting appears eon the log as `Me(0x0B) -> Thermostat(0x10), WWSettings(0x37), data: 01 (offset 3)`
And i'm sure that's working, i have same heating GB125, RC35, BC10, MC10, MM10, SM10 and use the circulation mode every day by starting the ciculation pump when a aquara-doorcontact idicates that someone enters the bathroom.

miege2k, 2022-09-17:
I'm a little confused, I was sure it wouldn't work, but maybe I didn't wait long enough or I change the wrong value. sorry for that.

MichaelDvP, 2022-09-17:
It's a bit confusing, because boiler controlls the circualtion pump, but thermostat have an extra mode.
Boiler wwcircmode sets the intervall per hour (telegram 0x33, offset 7), wwcirc starts a single 3 min pumpcycle (telegram 0x35, offset 1, bits 2,5), This is the same as pressing the water button on the RC35. 
Setting thermostat circmode to on, sets in thermostat 0x37, offset 3 and thermostat starts pump on boiler via 0x35 offset1, bit 4.
I have set boiler wwcircmode to continous and control the pump with thermostat wwcircmode on/off. I have a dallas sensor attached to the circulation pipe and stop the pump if temperature increases.

miege2k, 2022-09-29:
@MichaelDvP thanks for your help, now it works. I have two other problems, can you help me with that too?

MichaelDvP, 2022-09-29:
If you explain the probems, i'll try.

miege2k, 2022-09-29:
nice thanks.
1. My goal is to charge the wwstore from my phone when it is necessary. but when i change value boiler/wwonetime via mqtt sometimes it works and sometimes it does not.

2. i have strong fluctuations by the hc1 flow temperature. Setting: 
- outdoor temperature=13°C 
- hc1 target flow temperature=24°C
- hc1 design temperature=50°C
- hc1 day temperature =20°C
- minimal external temperature =-25°C
- hc1 offset temperature=-1°K
so now the behaviour that i see. When the boiler is starting the flow temp increases to 55°C but the target is only 24°. when the flow temp decreases to 22°C the boiler starts again. Did you have the same behaviour? if not can you send my your setting? I have tried to change many values but the behaviour is the same.

in the image you see the current flow temp:
[image]

MichaelDvP, 2022-09-29:
1. wwonetime only works if the dhw temperature is 5 degree below setpoint. In auto-mode this the temperature the charging is startet, so it only works if dhw is not automatic recharged.
2. You have a unmodulated oil boiler and no mixer. The thermostat calculates the required flowtemp (hc1/targetflowtemp), sends it to the boiler and the boiler set it as selflowtemp. With boiler parameters boilhyston/off the burner is started (selflowtemp+boilhyston) and stopped (selflowtemp+boilhystoff). This gives you the variation in real flowtemp. The high peaks to 80°C are from dhw charging. With a mixer (MM10) the boiler temperature will modulate the same, but flowtemp is mixed down and nearly constant at selflowtemp and room thermostats have less work.

Where do you live? minexttemp with -25 is very low. For germany this is in the [range of -11 to -16°C](https://www.haustechnikdialog.de/SHKwissen/Images/Klimazonen.jpg). But if the heatingcurve (minexttemp/designtemp) fit's your house, the setting is ok.

miege2k, 2022-09-29:
I will observe the information on point 1.

2. i changed the value of minexttemp to -25 to pull down the heating curve. Your description match with my imagination. But when the target flow is 25°C and the boilhystoff is +6 why the boiler stops at 55°C? 
I have the impression without a mixer the flow temp does strange thinks or there is a other problem. What is a modulated oil boiler in contrast to unmodulated?

miege2k, 2022-09-29:
Little example:

[image]

miege2k, 2022-09-29:
These are the self-measured values on the flow and reverse temp, they should not be so close to another or?

[image]

MichaelDvP, 2022-09-30:
> boilhystoff is +6 why the boiler stops at 55°C?

There is also a fixed minimum burner time. If heat demand gets higher you'll see that upper temp will get nearer to the +6.
> What is a modulated oil boiler in contrast to unmodulated?

Unmodulated burner is on/off at 100% power, a modulated can reduce burning power, ideal to the heat demand. Then burner stays on at a level that flowtemp stays at required temperature. For gas burners that works good, for oi burners only in a small range and more problems with dirty nozel.
>  flow and reverse temp, they should not be so close to another or?

This is due to the thermostat-valves. When flowtemp becomes low the valves open, resulting in high flow. When flowtemp increases the valves close later when roomtemp is reached. In this time the backflow goes up. With nearly closed valves the flow is small hot water in backflow pipe stays there and cools down slowly.

miege2k, 2022-10-05:
Ok thanks that is realy helpful. 

The only way to solve this problem is a mixer, right? 

I change my burnminperiod i think it was not a good idea, but i'm not sure about the old value. Can you tell my the value of your boiler to fix that?

miege2k, 2022-11-07:
@MichaelDvP the current measurements looks like another problem the boiler runs 30min to reach the target temperatur of 55°C. Did you have another idea?
[image]

miege2k, 2022-11-07:
The Problem is the same at my message: https://github.com/emsesp/EMS-ESP32/issues/621#issuecomment-1262415973
I have no influence on the flow temperature the boiler always stops at 55 °C regardless of design temperature and minimum external temperature

miege2k, 2022-11-07:
ok i see, that makes sense of course, thank you. 

That is, if I want to avoid these temperature fluctuations at low outdoor temperatures, I need a mixer and a buffer tank, right?

MichaelDvP, 2022-11-07:
This is with mixer, without buffer, showing boiler, flowtemp and return temp. To reduce burner starts, the flowtempoffset is 10 degrees (setpoint flowtemp as thin line with 1 degrees steps) and boiler hyst -6/+15.