emsesp/EMS-ESP32 #624: Bosch rf room temperature sensor not identified

tfriberg, 2022-09-18:
**Is your feature request related to a problem? Please describe.**
rf room temperature sensor not identified as sensor

**Describe the solution you'd like**
Identify rf room temperature sensor as a Sensor

**Describe alternatives you've considered**
Checking to see if I can do it myself somewhere in the EMS_ESP UI/dashboard

**Additional context**
```
Type
Unknown

Name
rf room temperature sensor

Brand
Device ID
0x40

Product ID
0

Version
00.00
```
[image]

Label on room sensor reads (sensor module):
Bosch
TR2 RF 8737709446
SW-Ver 2.4.004
TRX2-TIF

MichaelDvP, 2022-09-18:
Seems like the same sensor as in #103.
It only sends the remote temperature in telegram 0x435. This temperature is shown in master thermostat as roomtemperature, there is no real need to read thisextra from sensor. But since we have now support for multiple thermostats we can register it as thermostat (like the RF100H) and show the value.

tfriberg, 2022-09-20:
Will check. Thanks.

tfriberg, 2022-09-20:
@MichaelDvP 
Trying to upload 3.5.0b2 and, updating from 3.4.2. No joy!

[image]

[image]

MichaelDvP, 2022-09-21:
Seems to be a common issue, see #629.

closed 2022-12-16 by proddy